MarlinFirmware/Marlin #11612: [help] Coding own gcode command

Alissonverd, 2018-08-22:
Hello,
I'am coding my own gcode command and need extrude 20mm filament. 
So, with a search in commands that using extruder motion (G0_G1.cpp, M125.cpp and M600.cpp) I see many codes to do it.. 
But, what is the corrent way? And how do it?

G0_G1.cpp
`            current_position[E_AXIS] = destination[E_AXIS]; // Hide a G1-based retract/recover from calculations
            sync_plan_position_e();                         // AND from the planner`

M600.cpp
`  // Initial retract before move to filament change position
  const float retract = -ABS(parser.seen('E') ? parser.value_axis_units(E_AXIS) : 0
    #ifdef PAUSE_PARK_RETRACT_LENGTH
      + (PAUSE_PARK_RETRACT_LENGTH)
    #endif
  );`

M125.cpp (same as M600)
`  // Initial retract before move to filament change position
  const float retract = -ABS(parser.seen('L') ? parser.value_axis_units(E_AXIS) : 0
    #ifdef PAUSE_PARK_RETRACT_LENGTH
      + (PAUSE_PARK_RETRACT_LENGTH)
    #endif
  );`

fwretract.cpp
`    destination[E_AXIS] -= base_retract * renormalize;
    prepare_move_to_destination();                        // set_current_to_destination`
and
  current_position[E_AXIS] = old_e;
  SYNC_PLAN_POSITION_KINEMATIC();                         // As if the move never took place

So, with this search, I think I should use it.
`current_position[E_AXIS] = 0; // Reset E position
SYNC_PLAN_POSITION_KINEMATIC(); 

destination[E_AXIS] = 20; // set extrude amount
prepare_move_to_destination();

current_position[E_AXIS] = 0; // Reset E position
SYNC_PLAN_POSITION_KINEMATIC(); 
`
I need this procedure:
1 - Reset E positon to Zero;
2 - Extrude 20mm of filament;
3 - Reset E position again.

What is the safe way to do it?
Thank you

marcio-ao, 2018-08-28:
@Alissonverd : I feel your pain. I've struggled with this sort of thing myself :) I think your code needs to call `set_destination_from_current()`, otherwise the value of `destination[E_AXIS]` may be undefined so adding 20 to it will give you garbage.

As far as I can make it out, `current_position` corresponds the position the print head would be in once all queued (i.e. planned) moves are complete. This may not correspond to the actual print head position *right now*, but for most GCODE operations, this does not matter since all moves are placed at the end of the queue anyway. If you want to move to a new position, you can call `set_destination_from_current()` to copy the value of `current_position` into the scratch buffer, and then call `prepare_move_to_destination()` to instruct Marlin to move to that position.

You generally not write to `current_position` unless you want to change Marlin's notion of where it is in space without executing a move. If you do so, I think that needs to be followed by `SYNC_PLAN_POSITION_KINEMATIC()`. I believe the planner represents the position in step units, rather than mm units, and I think SYNC_PLAN_POSITION_KINEMATIC() updates those counters with new values corresponding to current_position.

So, with this in mind, I would do this:

```
set_destination_from_current(); // Copy current_position into scratch buffer
const float old_e_position_mygcode = destination[E_AXIS];
destination[E_AXIS] += 20; // set extrude amount
prepare_move_to_destination();  // perform the move to destination

current_position[E_AXIS] = old_e_position_mygcode;  // Restore the old position
SYNC_PLAN_POSITION_KINEMATIC(); // Make sure the planner is aware of the change
```

There may be other (better) ways to do this. This is just the pattern I have been using.

If you were doing motion rather than extrusion, `do_blocking_move_to`, `do_blocking_move_to_xy` and `do_blocking_move_to_z` seem to be useful functions as well. I'm not sure what is meant by "blocking" however...

-- Marcio

boelle, 2019-02-20:
@Alissonverd still having problems?

Alissonverd, 2019-03-02:
@boelle no, all working fine now!
Thank you

boelle, 2019-03-02:
remember to click the close button below :-D

closed 2019-03-02 by Alissonverd