MarlinFirmware/Marlin #11614: Grinding sound from extruder motor when using LIN_ADVANCE and TMC2130.

larsnordstrom, 2018-08-22:
### Description

When I set my K-value in my start script my extruder makes a grinding sound, when I set the K-value to 0 (disable lin_advance) there is no grinding. I have tried different configurations with acceleration, jerk, s_curve etc but have found that the only thing causing this problem is when lin_advanced is avtivated.

I use mostly Simplify3d but tested this with Cura too and the issue still persists.

### Steps to Reproduce

1. Install Marlin 1.1.9, 2.0 or any bugfix.
2. Use the config from the zipfile or run TMC2130 driver with LIN_ADVANCED enabled.
3. Print any object with the K value set in your start script.

**Expected behavior:** 
Extruder motor should run smooth and quiet

**Actual behavior:** 
Extruder motor make a grinding sound.

#### Additional Information
[Video](https://www.facebook.com/lano78/videos/10156479593178476/)
[Config-1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/2311365/Config-1.1.9.zip)

Lefuneste83, 2018-08-26:
Hello. I have a similar problem when feedrate value on at least one motor gets close to HYBRID_THRESHOLD value set for this axis. I am using TMC2208 which has a different library but are quite similar drivers (the libraries for both are being developed by the same courageous person).

https://github.com/teemuatlut/TMC2208Stepper/issues/21#issue-353887143

If I set HYBRID_THRESHOLD to a very high value (ie 150 for X/Y, 10 for Z and 60 for E0) the noise disappears completely and I am able to print smoothly in STEALTHCHOP up to at least 100mm/s speeds. Notice that commenting out #define HYBRID_THRESHOLD line in configuration_adv.h does not seem to disable it properly as set values for axes still bring this grinding noise when feedrate are getting closer to the set values. Maybe commenting the line only disable the activation of SPREADCYCLE only, but still uses the axis values. This is why in order to successfully disable this grinding noise behaviour I've set HYBRID_THRESHOLD to a value way above my typical print feedrates. I've included the conf files in the issue if you want to have a look.

Give it a try, maybe linear advance changes your E0 feedrate in a way that triggers the same behaviour on your printer, or maybe it is a completely different issue. I haven't set linear advance myself (yet).

larsnordstrom, 2018-08-28:
Increasing hybridthreshold for extruder made the grinding go away.
I had to increase all the way up to 190. Works fine👍

larsnordstrom, 2018-08-29:
Every setup is probably different but my advice is to first tune the currents for each axis according to what motors/drivers can handle and what temperatures can be tolerated by the drivers and motors.

When the currents are set, tune the feedrates, max_accel and default_accel and start pretty low. I personally started at 400 and increased until happy.

After the above step I tuned jerk until I got smooth transitional changes in directions and speed.

I have my printers in my office along with co-workers so my printer needs to be quiet and must stay in stealthchop during workhours so next step is the HYBRID_THRESHOLD settings. 
Sometimes the motors have sort of a whining/vibrating sound and that is usually caused by the printspeed reaching just u to the Hybrid limits but still not fast enough to switch over. In that case either lower the values or increase if you want faster or quiet printing. Higher value means it will stay stealthy, lower means it transitions into Hybrid (louder).

Incase of grinding sounds from extruder when LIN_ADVANCED is enabled, you can either set M900 K0
in your start script to disable the feature or increase the E0_HYBRID_THRESHOLD in increments of 10 until the grinding stops.

Current config for Einsy Rambo 1.1:

X/Y/E0 CURRENT          636
Z  CURRENT                   900
X/Y/Z/E0 MICROSTEPS   16

X/Y HYBRID_THRESHOLD     130
Z     HYBRID_THRESHOLD       10
E0   HYBRID_THRESHOLD     200

X_HOMING_SENSITIVITY   4
Y_HOMING_SENSITIVITY   6
 Z_HOMING_SENSITIVITY  6

DEFAULT_MAX_FEEDRATE 300, 300, 12, 120
DEFAULT_MAX_ACCELERATION 1000, 1000, 200, 5000
DEFAULT ACCELERATION/RETRACT/TRAVEL 1000

X/Y JERK 9.0
Z JERK 0.4
E JERK 5.0

// Enabled
S_CURVE_ACCELERATION
LIN_ADVANCE
ADAPTIVE_STEP_SMOOTHING
UBL with BLTouch
Bowden with Bondtech BMG

thinkyhead, 2018-09-09:
`LIN_ADVANCE` was recently patched. Please try the current bugfix branches and see if they're any better.

larsnordstrom, 2018-09-09:
The issue is still there, however if I run E0_MICROSTEPS 32 I am able to lower the the E0 HYBRID_THRESHOLD  to 115 from previous 200.

Lyr3x, 2018-09-17:
I can confirm that a E0_HYBRID_THRESHOLD of 180 removes grinding noises.

larsnordstrom, 2018-09-18:
I have found that when JUNCTION_DEVIATION is enabled the E0_HYBRID_THRESHOLD  have no effect on the grinding sound. Only way is to set K value to 0 or not use JUNCTION_DEVIATION at all.

Lyr3x, 2018-10-11:
Tried 49d4ed4d3f4fcee43364827fe671bd7183ef1fd5 today. Now the threshold of 200 istn woprking anymore. I'm getting retrtaction problems(more retraction then extruding) with weird sounds but a threshold of about 30 is working okay.

thinkyhead, 2018-10-26:
Rumor has it that lowering the jerk setting is helpful for Linear Advance.

larsnordstrom, 2019-01-13:
From what I understand, jerk is disabled when junction deviation is enabled so how could it have anything to do with it?

larsnordstrom, 2019-01-14:
Looking at the 2.0 branch there’s conditionals around XY jerk, E jerk does not. I’ll play around with that and see if there’s a “sweetspot” value.

Lyr3x, 2019-01-14:
In my experience a E-Jerk value of 15-20 eliminates the grinding noise (without stealthchop)

boelle, 2019-02-20:
@larsnordstrom problem solved with latest bugfix 2.0?

larsnordstrom, 2019-02-20:
@boelle you know what, I haven´t tested with the version I just updated to about a week ago.  
I made some changes to my printer too, switched back from bowden to directdrive with the new Bondtech MK3 and I'm also using LDO 0.9 pancake steppermotor. I'll give it a try in the morning and I'll get back.

larsnordstrom, 2019-02-21:
@boelle I did run a couple of prints with LIN_ADVANCE today and I did not notice the same kind of issues I had before. Since last, there have been updates to both Marlin and the TMC library and also there is an option to use Prusa MK3 chopper timings.
This is pretty much where it's at right now;

M92 X100.00 Y100.00 Z400.00 E830.00
M203 X200.00 Y200.00 Z12.00 E120.00
M201 X1000.00 Y1000.00 Z200.00 E5000.00
M204 P1250.00 R1250.00 T1250.00
M205 B20000.00 S0.00 T0.00 J0.03
M906 X636 Y636 Z900 E500 
M913 X120 Y120 Z10 E30
M914 X8 Y8 Z8
M900 K0.22

boelle, 2019-02-21:
so this one is solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

sergeysi779, 2019-03-26:
Please dont't close the issue.
I'm experiencing this right now on Ender-3 with MKS Gen L and TMC2130 in SPI mode. Strangely enough I have CR-10S wiith similar setup (TMC2130 drivers are from other vendor) and there is no such grinding noise when using Linear Advance. Both printers are on 24V.
I'll investigate further and report whether the issue persists.

sergeysi779, 2019-03-30:
Well, the issue is there. I've tested it on two boards:

1. Board: MKS Gen L. Drivers: TMC2130 in SPI mode. 
With Linear Advance K-factor greater than 0 extruder motor makes grinding noise. It doesn't matter whether driver is configured as SPI or standalone mode. It doesn't make grinding noise when it's in spreadCycle mode. Issue appears both in Marlin 1.1.9 and latest (2019-03-30) Marlin 2.0. I tried Prusa Mk3 and standard 24V chopper settings, it didn't help.

2. Board: SKR v1.3. Drivers: TMC2130 in SPI mode.
With Linear Advance feature enabled extruder motor doesn't spin at all. However there is some holding current going through motor. It doesn't matter whether K-factor is 0 or not. It doesn't matter whether driver is in stealthChop or spreadCycle mode. After flashing firmware without Linear Advance feature extruder motor works normally. Looks like I'm not the only one: https://github.com/MarlinFirmware/Marlin/issues/13463

Can someone else confirm either of those bugs? Maybe it's better to create separate issue?

harleycarr, 2019-05-20:
It is not solved, there is another issue about 2208's and it's the same result and reproduce. https://github.com/MarlinFirmware/Marlin/issues/13535

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead