MarlinFirmware/Marlin #11615: [1.1.9] DUAL_X_CARRIAGE where to define HOTEND_OFFSET_Z

PrintXDreams, 2018-08-22:
Recently I would like to update my DIY Dual X Carriage 3D Printer firmware from 1.1.6 to 1.1.9.
When I try to modify the configuration.h file, I found "HOTEND_OFFSET_Z" setting is removed.

Hotend offset Z is working good at 1.1.6 on my 3D printer.

Can someone help advice where to define the "HOTEND_OFFSET_Z" in 1.1.9 firmware?

Following is the setting in 1.1.6 Configuration.h
// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
#define HOTEND_OFFSET_X {0.0, 293.36} // (in mm) for each extruder, offset of the hotend on the X axis
#define HOTEND_OFFSET_Y {0.0, -1.48}  // (in mm) for each extruder, offset of the hotend on the Y axis
**#define HOTEND_OFFSET_Z {0.0, -0.55}**
// @section machine

Thanks in advanced!

RudyBenner, 2018-08-23:
actually, there are several hits on HOTEND_OFFSET.

PrintXDreams, 2018-08-23:
I look into the 1.1.9 code and found the Hotend_Offset_Z is actually applied during the T command in Marlin_main.cpp.

` // Apply Y & Z extruder offset (X offset is used as home pos with Dual X)
    current_position[Y_AXIS] -= hotend_offset[Y_AXIS][active_extruder] - hotend_offset[Y_AXIS][tmp_extruder];
    current_position[Z_AXIS] -= hotend_offset[Z_AXIS][active_extruder] - hotend_offset[Z_AXIS][tmp_extruder];`

Conditionals_post.h also define the default HOTEND_OFFSET_Z as "0"

`#define HAS_HOTEND_OFFSET_Z (HOTENDS > 1 && (ENABLED(DUAL_X_CARRIAGE) || ENABLED(SWITCHING_NOZZLE) || ENABLED(PARKING_EXTRUDER)))
#if HOTENDS > 1
  #ifndef HOTEND_OFFSET_X
    #define HOTEND_OFFSET_X { 0 } // X offsets for each extruder
  #endif
  #ifndef HOTEND_OFFSET_Y
    #define HOTEND_OFFSET_Y { 0 } // Y offsets for each extruder
  #endif
  #if HAS_HOTEND_OFFSET_Z && !defined(HOTEND_OFFSET_Z)
    #define HOTEND_OFFSET_Z { 0 }
  #endif
#endif`

Configuration_store.cpp save HOTEND_OFFSET in EEPROM.

Marlin_main.cpp to get the HOTEND_OFFSET through setting.load from EEPROM at the beginning.

I search the Hotend_Offset_Z and there are following 2 ways to modify.

1.  Configuration.h
Enable (PARKING_EXTRUDER) then define HOTEND_OFFSET_Z
But I don't use Parking Extruder feature.

2.  M218 to modify the HOTEND_OFFSET 
But this is not the way I want.

I also try to add the HOTEND_OFFSET_Z define following X and Y in Configuration.h.
It can be complied without warning or error.
So I will try it on my 3D printer to see.
I guess the removing of HOTEND_OFFSET_Z is one concern for Duplication Mode printing, am I right?

`// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
#define HOTEND_OFFSET_X {0.0, 292.00} // (in mm) for each extruder, offset of the hotend on the X axis
#define HOTEND_OFFSET_Y {0.0, 0.00}  // (in mm) for each extruder, offset of the hotend on the Y axis
#define HOTEND_OFFSET_Z {0.0, 0.2}  //where is this parameter?`

thinkyhead, 2019-02-20:
The latest Marlin has `HOTEND_OFFSET_Z` always available when `HOTENDS > 1`.

closed 2019-02-20 by thinkyhead