MarlinFirmware/Marlin #11618: 1.1.9 + TMC2130 problems maybe

HoeckDK, 2018-08-22:
I have tried to update to 1.1.9 unsuccessfully.

I run a Ramps1.4 and TMC2130 on X, Y, Z.

One issue i see, sometimes X and Y homing is not possible. Motors seem stuck. I have found that re-initializing EEP helps. But it does not seem to be a permanent fix. I think I have seen similar reports elsewhere. 

Another one, that I have not found a work-around for, is that Z-probing seems faulty. If i poll the endstops, the inductive probe reports correctly, if I put metal in front of it or not. If i send G28 Z, Z is raised a bit, then lowered a bit (where is should lower until probing). Then it raises some again. If I continue to send the command, the Z axis will hit max limits eventually. The probe is somehow connected, because if I keep metal in front of the probe, then the Z-axis only raises, with a few pauses. 

For debugging I kept the config file default, except step/mm, Has TMC2130, custom CS pins, ATX PSU control, termistor types, invert endstop signals, axis speeds, invert E0.

teemuatlut, 2018-08-23:
Try connecting your PSU power on connector to ground to effectively turn it on full time. This is to make sure the drivers have 12V power when booting up.

HoeckDK, 2018-08-23:
Power supply is not the issue. Fans are running. I think it is related to TMC2130

oakdesign, 2018-08-24:
How is your power Setup? TMC2130 should have VMOT before VIO. If your board has USB and ATX Connection and USB is plugged before ATX is turned on than your TMC2130 power up cycle VMOT before VIO is not ensured

Lefuneste83, 2018-08-26:
I confirm this behaviour since I switched to 24V on RAMPS (now powered through USB). In order to make sure I don't have this issue I do a RAMPS hardware reset (using the physical switch) once everything is powered. Also make sure you do a M502 followed by M500 everytime you reflash your firmware. I've had similar homing issues by not doing so as some previous values would still be active in the EEPROM.

HoeckDK, 2018-08-27:
I will consider your comments about the power supply 

But did you also the the inability to home related to psu issues?

Lefuneste83, 2018-08-27:
Yes homing would be heratic. Sometimes quite violent, going in reverse or crashing z into the bed.

boelle, 2019-02-20:
@HoeckDK stil having issues when using bugfix 2.0?

HoeckDK, 2019-02-20:
I have now tried to use the reset button on the ramps board. The problem persists. 

@boelle I downloaded the latest 1.1.9 yesterday. Same issue.

boelle, 2019-02-20:
use bugfix 2.0 not 1.1.19

HoeckDK, 2019-02-20:
OK.. I may try that. 

Just to clarify the steppers does move both ways. But they never get down to homing level.

AnHardt, 2019-02-20:
Smells for me as if
#define ENDSTOP_NOISE_FILTER
could help with the homing/probing problems.

boelle, 2019-02-20:
so you cant home? or cant the stepper go slow enough for homing?

boelle, 2019-02-20:
also it might help if you posted both your config files in a single zip file

HoeckDK, 2019-02-20:
[1.1.8.zip](https://github.com/MarlinFirmware/Marlin/files/2886571/1.1.8.zip)
[1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/2886573/1.1.9.zip)

HoeckDK, 2019-02-20:
@AnHardt  Why does it work in 1.1.8 then?

AnHardt, 2019-02-20:
Because a very small amount of default filtering was removed then.

HoeckDK, 2019-03-04:
@AnHardt The endstop noise filter did the trick. Thanks.

boelle, 2019-03-04:
so all god now?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead