MarlinFirmware/Marlin #11620: Compiling Error with exit code 1 for LPC1768

DyonisosHunter, 2018-08-22:
Hi, :)
just ordered an Azteeg X5 GT so for testing purposes I tried to compile Marlin 2.0 bugfix for it. Except copy/pasting the board name from pins.h to the #define in line 136 I kept everything else untouched.
Running PlatformIO: Build (LPC1768) gives me the output below.

Further PIO gives me many (many) errors telling me *function call is not allowed in a constant expression*.
Since - at least to me - there is no clear hint about the actual problem, I have no clue whether or not this has to do with the build failing.

### Description

Cannot compile for lpc1768

### Steps to Reproduce

1. Grab latest Marlin 2.0.x-Bugfix (by today).
2. Edit configuration.h line 136: to #define MOTHERBOARD BOARD_AZTEEG_X5_GT
3. Try to compile using PlatformIO within VSC with LPC1768 as enviroment.

**Expected behavior:** Compiling

**Actual behavior:** Compiling Error

#### Additional Information

```
Error: 
Processing LPC1768 (platform: nxplpc@<3.4.0; board: lpc1768)

Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 120MHz 32KB RAM (512KB Flash)
DEBUG: CURRENT(custom) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> 0.0.0
|-- <U8glib-HAL> 0.4
|-- <TMC2130Stepper> 2.4.3
|-- <TMC2208Stepper> 0.2.1

*** Multiple ways to build the same target were specified for: Z:\Projekte\SqueezeBot\Firmware\Azteeg X5 GT\Marlin-bugfix-2.0.x\.pioenvs\LPC1768\lib99d\U8glib-HAL_ID1932\clib\u8g_delay.o  (from ['Z:\\Projekte\\SqueezeBot\\Firmware\\Azteeg X5 GT\\Marlin-bugfix-2.0.x\\.piolibdeps\\U8glib-HAL_ID1932\\src\\clib\\u8g_delay.c'] and from ['Z:\\Projekte\\SqueezeBot\\Firmware\\Azteeg X5 GT\\Marlin-bugfix-2.0.x\\.piolibdeps\\U8glib-HAL_ID1932\\src\\clib\\u8g_delay.cpp'])
File "C:\users\hunter\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 324, in BuildLibrary
 [ERROR] Took 1.60 seconds 

 [SUMMARY] 
Environment megaatmega2560      	[SKIP]
Environment megaatmega1280      	[SKIP]
Environment at90usb1286_cdc     	[SKIP]
Environment at90usb1286_dfu     	[SKIP]
Environment DUE                 	[SKIP]
Environment DUE_USB             	[SKIP]
Environment DUE_debug           	[SKIP]
Environment LPC1768             	[ERROR]
Environment melzi               	[SKIP]
Environment melzi_optiboot      	[SKIP]
Environment rambo               	[SKIP]
Environment sanguino_atmega644p 	[SKIP]
Environment sanguino_atmega1284p	[SKIP]
Environment STM32F1             	[SKIP]
Environment STM32F4             	[SKIP]
Environment teensy35            	[SKIP]
Environment malyanm200          	[SKIP]
Environment esp32               	[SKIP]
 [ERROR] Took 1.60 seconds 
```

Thanks.

p3p, 2018-08-22:
There was an update (and reversion) of the our custom U8glib-HAL library, so for a while yesterday the build was broken, it may eventual update itself but if you delete `.piolibdeps` from the Marlin directory it should force it to redownload.

Not related but the Azteeg X5 GT does have an example config that may be useful to you.

DyonisosHunter, 2018-08-23:
Sweet, that did the job :D

Out of curiosity - out of the terminal output, how am I able to figure out the actual issue myself?

closed 2018-08-23 by DyonisosHunter

p3p, 2018-08-23:
This was a special case, 2 files conflicting, but you can see from the filenames that it happened inside a library rather than Marlin itself, you would have to know a bit about programming to understand that the error message meant 2 files would compile to object files named the same thing (which can't happen), then go and investigate the suspect libraries for any recent updates that messed it up.

In general without understanding C++ (and sometimes even when you do) the error messages will be a bit cryptic, it mostly comes down to looking through the error log for the first error (all subsequent errors are probably caused by the first) and seeing what line it happened on then investigate.

DyonisosHunter, 2018-08-23:
Huh, really need to get a little more into this ;)
Thanks a lot!

TimotF, 2019-04-16:
Hi, I cloned the project (v1.1.9) and built it successfully with platformio (vscode), however I have the same error message` function call is not allowed in a constant expression` that appears when I open the configuration.h file.
It seems to me that intellisense is unable to find the declaration for ENABLE() and DISABLE(), and I am unable to fix the issue. What did you do to fix it?

thinkyhead, 2019-04-16:
It sounds like you have corrupted configuration files. Start with clean ones and move your settings over with care.

TimotF, 2019-04-16:
I don't think so since I didn't change anything in them. I have the issue with the 'raw' files from the cloned repository...

thinkyhead, 2019-04-16:
We know that LPC1768 compiles. You'll need to provide more information.

TimotF, 2019-04-17:
For anyone having the same issue, it is a bug with intellisense. Just open the file macro.h so that intellisense can process this file and the errors vanishes.

Kraken3d, 2020-02-17:
Where is this macro.h file you speak of. Obviously I am a noob!

DrJohnM61, 2020-02-21:
I believe the file is called macros.h (note macros not macro) and can be found in Marlin > src > sd > usb_flashdrive > lib-uhs2 > macros.h

I opened this file in visual studio code and added a space within a comment line and the deleted it. Then closed the edit window and was prompted to save the file. After saving the file, I went into the Marlin Firmware Auto Build and hit the trash can to 'Clean' everything and then did a 'Build'

All the errors went away.

I assume by faking a modification to the file, the file is correctly re-loaded.

Also a noob here, so assuming a lot...

[image]

DrJohnM61, 2020-02-21:
I may have the wrong directory path. Should be src > core

However, error keeps coming back, so not sure what to do next.

DrJohnM61, 2020-02-22:
OK - now I understand. The comment is to literally have the macros.h file open when doing the build (have the file open in another tab). The file is Marlin > src > sd > usb_flashdrive > lib-uhs2 > macros.h

Kraken3d, 2020-02-22:
That did work for me , also a few times I had to open the board section to
populate. It would always compile but I hated seeing all the errors.

thinkyhead, 2020-02-23:
CTRL-T (or Command-T) pops up a text field to open any file by name. We use it to open up to the exact G-code we want to edit.