MarlinFirmware/Marlin #11622: Bad printing of circles from PC

Blisk, 2018-08-23:
### Description

When I print from SD card printer prints nice round smooth circles. But when I print the same from PC, printing is slow and printing circles have many short pauses after that circles are really bad, robust.

### Steps to Reproduce

comps, 2018-08-23:
It's a known limitation for "higher-resolution" circles (too short segments) on 8bit hardware and Marlin, especially when printing via USB. The situation is better with an SD card (I never figured out why, the serial line is hardly taxed), but only slightly.

Using 32bit boards might be better, but other issue threads here seem to indicate Marlin in general has issues with many small segments regarding acceleration/jerk.

It seems we're still pretty far from an ideal single-step segment resolution.

(TL;DR - use lower resolution STL)

Blisk, 2018-08-23:
Yes but this use to work, that start to happend suddenly I think it is not because updating merlin

MiddleMan5, 2018-08-26:
Can you post the following information:

- Slicing Software, Version, Slicing Configuration
- Marlin Version and Config (or output of M503)
- Printer Hardware Version and Motherboard

Blisk, 2018-08-27:
Slicing software is Repetier 2.1.3
Slicing configuration is CuraEngine
Marlin version is bugfix-1.1.x 31.7.2018
http://zerobin.povej.net/?025fc5b0329fc006#BZSAiAN/l1QZ31cO4eZT1JCCSRDco26enH4Nn3PuSI0=
Printer is Geeetech I3 pro C board is GT2560 rev.A

I also have a problem with shifting layer when I print from SD card with this version of marlin.
Sometimes it shift to X, sometimes to Y for a mm or less. Belt are tighten, but not too much and all screws are tightened. Before that din't happend. I updated marlin because of auto bed leveling and filament out switch.

gloomyandy, 2018-08-27:
In terms of missed steps, what stepper drivers does that board have and do you have them correctly defined in your configuration file?

MiddleMan5, 2018-08-28:
This is a completely unofficial, unrecommended test, but could you add the following:
```
#ifdef STUTTER_FIX
OCR1A = (OCR1A < (TCNT1 +16)) ? (TCNT1 + 16) : OCR1A;
#endif
```

To line 1408 (or thereabouts) of stepper.cpp in the marlin repo, then recompile and flash for your board.
Add the code directly after the line that looks like:
```
// If there is a current block
  if (current_block) {
```

then, in Configuration.h add:
`#define STUTTER_FIX`
so that you can remove it and add it to test.

Blisk, 2018-08-28:
I have regular Nema 17 stepper motors, which works ok with original marlin, but not with latest, where I have constant layer shifting

Blisk, 2018-08-28:
MiddleMan5
I did that but still when I print, circles are not nice round but it is the same as before.

Blisk, 2018-08-29:
Here are results of printing.
No.1 is printed by SD card with original stock firmware.
No.2 is printed with PC with original stock firmware
No.3 and No.4 is printed with PC with Marlin bugfix-1.1.x 31.7.2018
[image]

henry42, 2018-09-01:
Hi, I am facing the same problem.
when I print from pc, the circle is changed to lots of points and printed really slow. but when using sd everything is ok.
It happened just in the version 1.1.9, it was fine in 1.1.8.

Blisk, 2018-09-03:
Can it be a problem settings on PC for port?

henry42, 2018-10-09:
> Can it be a problem settings on PC for port?

I found that it happened only if when I use cura.

If I export to the gcode, and use Printrun to print the model, everything seems fine.
I think maybe the problem is cura.
Have a try.

closed 2018-10-16 by thinkyhead