MarlinFirmware/Marlin #11625: How to I calibrate my marlin for my nema 17 with gearbox.

fatihaslan23, 2018-08-23:
### Description

I did build delta model clay printer and I have a clay extruder. When I started printing, my extruder motor running with vibratory and noisy. My extruder motor is nema 17 with gearbox. I think marlin and gearbox nema 17 not compitable.
How to I calibrate my marlin for my nema 17 with gearbox.

Motor Description
Brand name:HANPOSE
Certification:ROHS,CCC,CE,ISO,UL
Type:Hybrid
Phase:2
Step Angle(degrees):1.8
Model Number:17hs4401S-PG518
Holding Torque:230NM
Current / Phase:1.68A
Size::42*70

Marlin Settings
******Configuration.h******
FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://www.mendel-parts.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:FA3D EXTRUDER_COUNT:1

#define DEFAULT_AXIS_STEPS_PER_UNIT   {100, 100, 100, 200}
#define DEFAULT_MAX_FEEDRATE          {200, 200, 200, 200}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {8000,8000,8000 ,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          9000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000   // X, Y, Z and E m ax acceleration in mm/s^2 for retracts

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
// #define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
// #define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                5.0    // (mm/sec)
#define DEFAULT_ZJERK                 5.0    // (mm/sec)
#define DEFAULT_EJERK                 4.0    // (mm/sec)

******Configuration_adv.h******
#define MICROSTEP1 LOW,LOW
#define MICROSTEP2 HIGH,LOW
#define MICROSTEP4 LOW,HIGH
#define MICROSTEP8 HIGH,HIGH
#define MICROSTEP16 HIGH,HIGH

// Microstep setting (Only functional when stepper driver microstep pins are connected to MCU.
#define MICROSTEP_MODES {16,16,16,16,16} // [1,2,4,8,16]

// Motor Current setting (Only functional when motor driver current ref pins are connected to a digital trimpot on supported boards)
#define DIGIPOT_MOTOR_CURRENT {135,135,135,135,135} // Values 0-255 (RAMBO 135 = ~0.75A, 185 = ~1A)

### Steps to Reproduce

1. I'm tried to change microstep modes to 8 and 4
2. I'm tried to change digipot_motor_current 135 to 185
3. I'm tried to change steps per mm around 200-650

**Expected behavior:** [What you expect to happen]
I want to turn faster my extruder motor in this way I will have more clay flow control.
**Actual behavior:** [What actually happens]
Currently, I have a vibration and noisy extruder or slowly but quiet extruder. What I need is spin faster and quiet extruder.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2315151/Marlin.zip)

[video-1534964937.zip](https://github.com/MarlinFirmware/Marlin/files/2315163/video-1534964937.zip)

#### Additional Information
I shared one video of my print process to how was run my extruder motor
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-09-06:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-09-06 by thinkyhead

chetan98250, 2019-07-15:
Hi
Please plug all driver jumpers in reprap boards
For 16 stepping 

chetan98250@gmail.com