MarlinFirmware/Marlin #11628: LPC1768 (Azteeg x5 GT) no serial via USB

DyonisosHunter, 2018-08-23:
### Description

Hi again. The Board arrived.
I could succesfully build the .bin and have the board make a .cur out of it but I am not able to get a communication via USB working.
Issue #11483 seemed interesting but permutating #define SERIAL_PORT and #define SERIAL_PORT_2 over -1 to 2 did not help.

OS is win10. Device manager shows the board under 'other devices' as 'Smoothie' (with an exclamation mark).

### Steps to Reproduce

Don't know if the issue is reproducable. Problem might be the OS.

**Expected behavior:** [What you expect to happen]

Establish serial communication via USB to the board.

**Actual behavior:** [What actually happens]

No serial communication via USB.

#### Additional Information

(for some reason I am not able to attach a zip)

Relevant sections:
configuration.h

```/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
```
pins_AZTEEG_X5_GT.h addition:

```
#if HAS_DRIVER(TMC2208)
  /**
   * TMC2208 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial1
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  /**
   * Software serial
   */
  #define X_SERIAL_TX_PIN    P1_18
  #define X_SERIAL_RX_PIN    P3_26
  //#define X2_SERIAL_TX_PIN   -1
  //#define X2_SERIAL_RX_PIN   -1

  #define Y_SERIAL_TX_PIN    P2_11
  #define Y_SERIAL_RX_PIN    P3_25
  //#define Y2_SERIAL_TX_PIN   -1
  //#define Y2_SERIAL_RX_PIN   -1

  #define Z_SERIAL_TX_PIN    P1_21
  #define Z_SERIAL_RX_PIN    P0_17
  //#define Z2_SERIAL_TX_PIN   -1
  //#define Z2_SERIAL_RX_PIN   -1

  #define E0_SERIAL_TX_PIN   P0_16
  #define E0_SERIAL_RX_PIN   P2_06
  //#define E1_SERIAL_TX_PIN   -1
  //#define E1_SERIAL_RX_PIN   -1
  //#define E2_SERIAL_TX_PIN   -1
  //#define E2_SERIAL_RX_PIN   -1
  //#define E3_SERIAL_TX_PIN   -1
  //#define E3_SERIAL_RX_PIN   -1
  //#define E4_SERIAL_TX_PIN   -1
  //#define E4_SERIAL_RX_PIN   -1
#endif
```

Thanks :)

DyonisosHunter, 2018-08-23:
I am really sure I did that already.

[image]

This is what the device manager shows. There is no other com port to select.

Running smoothie with the suggested config works just fine. So hardware should be fine.
Don't know if it makes any difference but at the first flash the red LED went dark after some time. Ever since I always have to pres reset oder unplug main power to make it turn off. Anyway, a .cur and eeprom.dat gets created on the SD.

p3p, 2018-08-23:
Showing Smoothie(!) in device manager means the LPC1769 had a watchdog reset (crashed), it gets stalled in the smoothieware bootloader and doesn't reboot until it's reset.

As for the issue, I'm unsure, but to narrow it down try and boot Marlin with the default example, if that works add your customisations until it doesn't. I'm going to assume it's something todo with TMC2208s?

DyonisosHunter, 2018-08-24:
Hi,
I replaced both config files with the examples and commented out all extra pin definitions in pins_AZTEEG_X5_GT.h. Except that I had to deactivate #define TEMP_STAT_LEDS it build just fine.
BUT I'm seeing the same behavior as before -.-

Yes, 4 TMC2208s are going to be used. Working flawlessly on the UNO+RAMPS and I'm going to try to keep the UART functionality.

p3p, 2018-08-24:
We need to try and isolate the issue you are having, so start from the minimum changes necessary (just the board and serial to -1) to the default config (not example this time, was hoping they would just work), I know you want TMC2208s but we have to get the minimum working first. The fun of using experimental beta firmware ;)

DyonisosHunter, 2018-08-24:
True ;)

Sooo, I'm still using the example config. PIO refused to build out of the sudden caused by some permission issues with the working folder. So I deleted .pioenvs and piolibdeps and recompiled. Now COM7 popped up and I have access to the sdcard via file explorer. One step closer I guess ;)

Anyway, still no serial communication working...

p3p, 2018-08-24:
Make sure `#define SERIAL_PORT -1` it isn't in the example

DyonisosHunter, 2018-08-25:
Hey, sorry, had to leave a little sudden...
Anyway, I got it to work (working config attached below). As stated in Issue #11483 and as you said `#define SERIAL_PORT -1` does the trick. But still it took a while to figure, that the sd card I used seems to be defective. The same fw - no changes - some times just works but most of the time I get the issues stated above.

Now after adding settings from my ramps setup something breaks serial again. I took care to only add those settings that should not mess up any pin definitions. Stuff like `#define EXTRUDERS 1` or `#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 400, 100 }` for instance.

Is there a particular power off/on sequence to initialize a new firmware.bin that I possibly miss?

[Azteeg_obedient_serial.zip](https://github.com/MarlinFirmware/Marlin/files/2320540/Azteeg_obedient_serial.zip)

p3p, 2018-08-25:
> Is there a particular power off/on sequence to initialize a new firmware.bin that I possibly miss?

On reset the bootloader looks for a bin, if its there flashes it, so any old reset will do. Make sure you are ejecting the USB drive before reset, Your OS lies to you about how fast transfers are.

I also recommend setting `#define FLASH_EEPROM` in your config until it becomes the default, it's more likely to be stable on non Re-ARM boards.

DyonisosHunter, 2018-08-25:
Pressing and releasing the reset trigger too fast doesn't work reliably. Pressing and holding it for like a second worked out just fine :)

So, by now I have transfered all my previous settings and it turned out, that activating `#define PIDTEMPBED` breaks something. I have double checked a few times - everytime its activated the opening issue gets triggered.

I think I can go without `#define PIDTEMPBED` .

There is no `#define FLASH_EEPROM` :/

closed 2018-08-25 by DyonisosHunter

p3p, 2018-08-25:
> Pressing and holding it for like a second worked out just fine :)

As long as the board resets it will start the boot loader that handles the firmware update, odd that you need to hold it on this board.

> that activating #define PIDTEMPBED breaks something.

I use PIDTEMPBED on LPC176x based boards, can't think of a reason it would stop the board booting

> There is no #define FLASH_EEPROM :/ -> found it. It's inside persistent_store_api.h

Ah sorry, I meant just add it to your config file not uncomment, its a bit of a hidden option at the moment.

gloomyandy, 2018-08-25:
You probably need the RX pins to be interrupt capable, other than that it may depend upon what sort of components are hanging off the pins (pull up resistors, etc.).

But does software serial even work on this processor? It looks like a key table that sets the delays for various baudrates is not filled in at all, so perhaps it the code still needs further work?

DyonisosHunter, 2018-08-26:
Hi :)

Writing does not work either. May be that it still need some work done. There is a SoftSerial library available at mbed so I guess yeah, *should* be doable (while I can't find any docs about what pins are capable).