MarlinFirmware/Marlin #11631: TEMP_SENSOR_CHAMBER , not displayed anywhere

creatron, 2018-08-23:
If the chamber sensor is defined the temperature is read and converted to float, but nobody uses this. I find that RUMBA pins does have a pin allocated and use this with RAMPS 1.4 modified, This all works but nothing is displayed. I found what I think was the originator of this code and that code has the following

  if (PAGE_UNDER(28)) {
    // Extruders
    HOTEND_LOOP() _draw_heater_status(5 + e * 25, e, blink);

    // The temperature in the chamber (inside the printer).
    #if EXTRUDERS <= 2 && TEMP_SENSOR_CHAMBER >= 1
      u8g.drawFrame(52, 0, 24, 26);
      _draw_centered_temp(thermalManager.current_temperature_chamber + 0.5, 54, 15);
    #endif

    // Heated bed
    #if HOTENDS < 4 && HAS_TEMP_BED
      _draw_heater_status(81, -1, blink);
    #endif
I think the idea was to draw the chamber temperature after the extruder and before the Bed and FAN temperature image,.

creatron, 2018-08-24:
I manage to display the temperature with the following addition in "status_screen_DOGM.h" line 254
  //
  // Temperature Graphics and Info
  //

  if (PAGE_UNDER(28)) {
    // Extruders
    HOTEND_LOOP() _draw_heater_status(STATUS_SCREEN_HOTEND_TEXT_X(e), e, blink);
    // Heated bed
    #if HOTENDS < 4 && HAS_HEATED_BED
      _draw_heater_status(STATUS_SCREEN_BED_TEXT_X, -1, blink);
    #endif

    #if HAS_FAN0
      if (PAGE_CONTAINS(STATUS_SCREEN_FAN_TEXT_Y - 7, STATUS_SCREEN_FAN_TEXT_Y)) {
        // Fan
        const int16_t per = ((fanSpeeds[0] + 1) * 100) / 256;
        if (per) {
          u8g.setPrintPos(STATUS_SCREEN_FAN_TEXT_X, STATUS_SCREEN_FAN_TEXT_Y);
          lcd_print(itostr3(per));
          u8g.print('%');
        }
      }
    #endif
    
    // gerrie start
    // The temperature in the chamber (inside the printer).
    #if TEMP_SENSOR_CHAMBER >= 1
      u8g.drawFrame(8, 20, 23, 9);
      _draw_centered_temp(thermalManager.degChamber() + 0.5, 10, 28);  // degChamber()
    #endif
    // gerrie end 
  }

AnHardt, 2018-08-25:
You can't put the chamber at a fixed place. 
There has to be some rules where to display what.
For example:
We have N places to display M extruders, the chamber, the bed and the fan,
If there is a part cooling fan, display it at the rightmost place.
If there is a part cooling fan, and you have a bed, display left of the fan, else at the rightmost place.
If there is a part cooling fan, and/or you have a bed, display the the chamber left of them, else rightmost.
If the leftover places are >= the number of extruders display them all, else if the left over places are 2 draw the active extruder at the leftmost place and alter all extruders at the second place, if only one place is remaining alter all the extruders at the leftmost place.

Or alternative when you think extruder temps are more important than the others you could iterate over fan, bed, chamber at the rightmost place if you need to.

thinkyhead, 2018-08-30:
In the future we'll draw the fixed things on Graphical display just once, then only update the parts that change. That will allow us to more easily display the Nozzle, Bed, Chamber icons programmatically, where now we have to have separate `_Statusscreen.h` blobs for every possible variant.

boelle, 2019-02-20:
@thinkyhead maybe "convert" this to a feature request?

creatron, 2019-02-21:
I done it and it works fine, I can supply the code and pictures how it is
displayed on the screen etc. I currently is installed on the Ender 3, CR10
S4 and A8 (convertered to MKS BASE)  printers. I currently use 1.1.9, but I
tested it also on 2.0 as well ( I think on my  MKS SBASE as well. only
hardware , no printer real estate at this stage, but with all the stepper
and sensors connected (SD card was not yet operational at that time, the
only reason it not yet on a printer).

If interested let me know how to proceed.

Kind Regards
Gerrie

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.)

Phone +27(0)11 024 3832
Phone +27(0)11 026 5631
Fax   086 552 2890
Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

closed 2019-05-16 by thinkyhead