MarlinFirmware/Marlin #11635: Marlin 1.1.9 does not recognize negative offsets in X and Y axis.

sigourosandreas, 2018-08-24:
### Description

Hello,
I was using marlin 1.0.2 for the last 4 years now, recently changed to 1.1.8 and now to 1.1.9., always using the some setup.
The 1.1.8 has been fixed [here](https://reprap.org/forum/read.php?415,823273), because it had almost the same bug.
The 1.1.9 is always giving me this error during compiling: 
SanityCheck.h:895: error: static assertion failed: LEFT_PROBE_BED_POSITION is outside the probe region.

static_assert(LEFT_PROBE_BED_POSITION >= MIN_PROBE_X, "LEFT_PROBE_BED_POSITION is outside the probe region.");

^

SanityCheck.h:896: error: static assertion failed: RIGHT_PROBE_BED_POSITION is outside the probe region.

static_assert(RIGHT_PROBE_BED_POSITION <= MAX_PROBE_X, "RIGHT_PROBE_BED_POSITION is outside the probe region.");

^

SanityCheck.h:897: error: static assertion failed: FRONT_PROBE_BED_POSITION is outside the probe region.

static_assert(FRONT_PROBE_BED_POSITION >= MIN_PROBE_Y, "FRONT_PROBE_BED_POSITION is outside the probe region.");

^

SanityCheck.h:898: error: static assertion failed: BACK_PROBE_BED_POSITION is outside the probe region.

static_assert(BACK_PROBE_BED_POSITION <= MAX_PROBE_Y, "BACK_PROBE_BED_POSITION is outside the probe region.");

^

SanityCheck.h:960: error: static assertion failed: Z_SAFE_HOMING_X_POINT is outside the probe region.

static_assert(WITHIN(Z_SAFE_HOMING_X_POINT, MIN_PROBE_X, MAX_PROBE_X), "Z_SAFE_HOMING_X_POINT is outside the probe region.");

^

SanityCheck.h:961: error: static assertion failed: Z_SAFE_HOMING_Y_POINT is outside the probe region.

static_assert(WITHIN(Z_SAFE_HOMING_Y_POINT, MIN_PROBE_Y, MAX_PROBE_Y), "Z_SAFE_HOMING_Y_POINT is outside the probe region.");

### Steps to Reproduce

1. I have attached the configuration files:
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2317400/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2317401/Configuration.txt)

Paste them in the 1.1.9 marlin and run the program. It should give you the same error like mine.
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2317428/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2317429/Configuration.txt)

**Expected behavior:** Run the program without any error.

**Actual behavior:** The new firmware will be uploaded in the board correctly.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

oakdesign, 2018-08-24:
Readig through your post on reprap I won't consider this as a bug as it's a very special desing of your Printer with a probe not probing on the Bed but outside of it. 
The SanityChecks are just there to avoid it to let a user set up probe positions outside the bed area to avoid unwanted crashes.
Regard

sigourosandreas, 2018-08-24:
I had a mistake in my first post, sorry for that.
So how is it possible to fix this problem and complete the uploading?

Εστάλη από το BlueMail<http://www.bluemail.me/r?b=13506>

sigourosandreas, 2018-08-27:
Thank you for the help. It works perfect.
This our own design, custom made printer.

closed 2018-08-27 by sigourosandreas

thinkyhead, 2018-08-29:
> probing outside of the bed area

Probing is constrained to the defined bed area. To probe farther out, increase the configured min-max movement settings.

tonyjobson, 2019-04-25:
To anyone else finding this page and still scratching their head.

(I was worried if i probed at X_min_pos the bltouch woudl fall off the side of the bed and not make contact.)

 You don't need to consider your probe to nozzel offset when working out your probe area (it's calculated into the prob area). I thought I did and spent a long time looking at this error trying to work out why it was bigger than my print bed.

Justinjhcs1, 2019-07-10:
this is pissing me off too the probe stops about 4mm short of the bed and the nozzle is aporx 3mm away when it is supposed to be 0.20mm 1/2 the Nozzle Diameter of 0.4mm WTF I think the probe has a 4mm 
Z is parking in the middle X85 y100 z5.... Ill try doing a 15mm Cube print and see if this will keep on like this.....

AnHardt, 2019-07-10:
Did you try more endstop/probe noise filtering?