MarlinFirmware/Marlin #11636: Z homing and then G1 Z0 not going to zero, Z offset ignored?

Gaz83, 2018-08-24:
I have been using Marlin 1.1.8 for a while now and today I came to print and noticed an issue. When I homed everything and set the Z axis to go to zero, it looks like its a good few mm off the bed.

I am using a Bltouch and before today, everything worked fine.

I decided to clear my Z offset and do it all again. My offset comes out at -1.8, same as before.

I Homed everything again and then moved Z to 0. Same problem.

So I decided to download latest version 1.1.9 and set everything up again. Problem is still there.

I even tried a stupid number like Z-2.5, no difference. It's almost like Z offset is being ignored.

gloomyandy, 2018-08-24:
How are you setting the Z offset? If the changes are in your configuration.h file and you are also have eeprom enabled, did you reload the default settings after you made the changes? Always a good idea to do an M502 followed by M500 to make sure the eeprom values match the defaults.

If this does not help then you may need to provide more details of what board etc you are using and upload your config files.

Gaz83, 2018-08-24:
yeah I am using M500 commands. Saving it, loading it etc. But as I say, this was working and suddenly doesn't without any changes.

My board is MKS Gen-L. I will upload config file when I get in later.

Gaz83, 2018-08-24:
Added Configuration

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2318776/Configuration.zip)

Gaz83, 2018-08-25:
Issue not related to Marlin. Printed a new Bltouch holder but it did not print to correct size so the Bltouch was not in same position.

closed 2018-08-25 by Gaz83