MarlinFirmware/Marlin #11638: [1.1.9] SD_FINISHED_RELEASECOMMAND defined but not used

marcio-ao, 2018-08-24:
The configuration option "SD_FINISHED_RELEASECOMMAND" has been replaced by "SD_FINISHED_STEPPERRELEASE". As a consequence, "SD_FINISHED_RELEASECOMMAND" is no longer used anywhere and should be removed.

thinkyhead, 2018-09-12:
I see them both. And in both 1.1.x and 2.0.x codebases. For example:
```cpp
#if ENABLED(SD_FINISHED_STEPPERRELEASE) && defined(SD_FINISHED_RELEASECOMMAND)
  if (!cleaning_buffer_counter) enqueue_and_echo_commands_P(PSTR(SD_FINISHED_RELEASECOMMAND));
#endif
```

marcio-ao, 2018-09-12:
@thinkyhead: That's odd. I could have sworn I grepped through the entire code base. I guess I missed that. Good catch. I'll close this ticket as irrelevant.

closed 2018-09-12 by marcio-ao