MarlinFirmware/Marlin #11639: [bufix-2.0.x latest] LPC1768 onboard SD card access

ModMike, 2018-08-24:
### Description

Use perssitentstore api to store EEPROM to onboard hardware and use SD card for printing

### Steps to Reproduce

In the  folder:

1) in HAL_LPC1768/main.cpp disable USB initialization:
`//USB_Init()`

2) in HAL_LPC1768/peristant_store_api.h enable flash:
`#define FLASH_EEPROM`

3) in HAL_LPC1768/spi.h enable on board sd card:
```
/** onboard SD card */
#define SCK_PIN           P0_07
#define MISO_PIN          P0_08
#define MOSI_PIN          P0_09
#define SS_PIN            P0_06
```

**Expected behavior:**

USB port is disabled, Marlin sees onboard SD card, Display works, printing from WEB UI is enabled

**Actual behavior:** [What actually happens]

1) Compiles but display is killed.  

2) Persistentstore seems to work because eeprom is no longer on SD card.

3) No access to SD card

4) ESP32 Web UI no longer talks to Marlin

**Actual behavior:** [What actually happens]

gloomyandy, 2018-08-24:
Firstly I'd like to point out that what you are trying to do is not really a supported feature at the moment, so there is a very high chance it will not work. The LPC1768 support is very much a work in progress and definitely not production code. That said you may be able to get something working, but it would help if you provide as much information as possible. 

What board are you trying to do this with? Are you sure those are the correct pins for your hardware? How is the ESP32 talking to the LPC1768? Are you trying to use software or hardware SPI?

If you want to access the on-board SD card at the very least you will need to comment out the call to MSC_SD_Init(0) from main as this is what kicks off the sharing of the SD card via USB. But there is a very good chance that things will still not work. You may just have to wait until things become more stable.

ModMike, 2018-08-24:
I understand but I am told this might work with the right config.  I am using the Azteeg X5 Mini Wifi (updated post to reflect).

I am pretty sure they are the correct pins because it is based on the smoothie.

p3p, 2018-08-24:
You are asking for a way to disable USB entirely and use the onboard SD card with Marlin, not what your description days, 

You already use the PersistentStore api as that is how all persistent changes are saved within Marlin, you currently use the SDCard version, to use the Flash based version you need `#define FLASH_EEPROM
` in your config.

As I've said previously disabling the USB stack to enable the onboard card is a hack and outside what is tested, you would need to at least comment out this from main
```cpp
  (void)MSC_SD_Init(0);

  USB_Init();                               // USB Initialization
  USB_Connect(TRUE);                        // USB Connect

  const uint32_t usb_timeout = millis() + 2000;
  while (!USB_Configuration && PENDING(millis(), usb_timeout)) {
    delay(50);
    #if PIN_EXISTS(LED)
      TOGGLE(LED_PIN);     // Flash quickly during USB initialization
    #endif
  }
```

and probably this from SysTick_Handler
```cpp
disk_timerproc();
```

define the spi pins in the pins file for your board, don't modify the defaults.

> I understand but I am told this might work with the right config

pretty sure I said it might work if you are adventurous and make the right modifications, but I wasn't going to walk you through it ;)

ModMike, 2018-08-24:
P3P, thank you for your input and I understand.  I figured if I started a described path, people could add to the solution.  What really frustrates me is that we have a board that works in every respect AND was a built in WEB UI.  The web ui has only one fault, it cannot print because Marlin cant access the SD card.

I find the idea of having to buy an RPI for octoprint (which I have and love) wasteful for users when we are so close to an all in one solution.  I could care less about the USB port.  If we can kill it and make this work, we have an all in one solution.

p3p, 2018-08-24:
I understand your frustration but you have to be patient, one of the first things I said to you was that this was on my todo list (ie. I'm working towards it), If the above doesn't work and you are sure you have the SPI pins correct for the board I will look further into a temporary hack for you, time permitting. (just a guy doing his hobby here)

ModMike, 2018-08-24:
P3P.  I certainly understand what you are saying and don't want to anger you.  On the flip side, if it's just a few lines of code and someone can help me out, then great.

Quick update:  So LCD is back, persistent store works (responds to M500 & M501 properly), and ESP32 is talking to Marlin again.  Issue is still accessing SD card through marlin.  I believe the pins are ok but will triple check.

Do I need to enable a 2nd serial port for Marlin to talk the SD card?  Right now I only have the first port set to 0 for the ESP32 Web UI host to speak to Marlin.

ModMike, 2018-08-24:
Just looking through the spi_pins.h file where I enabled onboard card SPI.  Do I need to disable external sd card?

/** external */
#ifndef SCK_PIN
  #define SCK_PIN           P0_15
#endif
#ifndef MISO_PIN
  #define MISO_PIN          P0_17
#endif
#ifndef MOSI_PIN
  #define MOSI_PIN          P0_18
#endif
#ifndef SS_PIN
  #define SS_PIN            P1_23
#endif
#ifndef SDSS
  #define SDSS              SS_PIN

p3p, 2018-08-24:
> define the spi pins in the pins file for your board, don't modify the defaults.

You shouldn't be modifying the `spi_pins.h` file, custom pins should be set in the pins file for your board, the custom values will override that file

ModMike, 2018-08-24:
OK

gloomyandy, 2018-08-24:
Is there a schematic available for the board? Reading the esp3d comments it seems that the board you are using may have some sort of extra hardware to allow the ESP32 to share the SPI bus that provides access to the SD card with the LPC1768 (or is it 69 on that board?). Is this how you are trying to use it? This may or may not work with Marlin as the current code is not really setup with a shared SD card and shared SPI bus in mind. I'm currently doing some work in this area so have got to know this code pretty well! I've managed to corrupt my SD card many, many times along the way!

Looking at the pins you are trying to use for SPI to the SD card it would seem that it is on SSP1, this means you must enable software SPI as the current LPC1768 hardware SPI is hardcoded to use SSP0. So in addition to making the changes to the pins (and commenting out the USB code) you need to define LPC_SOFTWARE_SPI this may already be set in your pins file but if not you may need to define it. 

Oh and what display are you using? 

So just to take a step back, what did you have working before you started down the path of disabling USB? Did the display work? Could you talk to Marlin using the ESP32?

ModMike, 2018-08-24:
P3P, I reverted SPI changes and moved it to my boards pins file.

Gloomandy,

I believe there is one for the X5 GT which is similar.  Let me look at it to confirm pins.  

Before I disabled USB everything worked except printing from SD card.  ESP32 WEB UI did and currently talks to marlin no problem.

I did not define LPC_SOFTWARE_SPI in my boards pin file but ill do it now.  Do I need to define a second serial port?

ModMike, 2018-08-24:
Please find the schematics for the previous version (not wifi) attached.  As best as I can determine, SD pins are correct.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2319815/Archive.zip)

ModMike, 2018-08-24:
Just to be triple sure I used Dip Trace to trace the pins and they are indeed the right pin numbers.

ModMike, 2018-08-25:
Just received a response from Panucatt:

X5 mini uses a multiplexer to switch the SD card to either MCU or ESP. Most of the time it is connected to the MCU except when transferring files using ESP.

They also sent a preliminary schematic that I can consult but was asked not to distribute because it has not been cleaned up yet.  Let me know what you need.

gloomyandy, 2018-08-25:
I'd really like to see the details of how the multiplexer works, we may need to make changes to support it and ensure that there is no chance of corrupting the SD card when using it (we probably need to consider forcing a reload of the SD card directory listing if the SD card can be changed by something else, it would be nice if there was some way to detect that things may have changed). It may be that the hardware allows Marlin to lock access in some circumstances (which would be a good thing).

ModMike, 2018-08-25:
I guess software serial is not an option here but seeing that the SD defaults to MCU we should be able to get access.  Finally, if the WEB UI locks itself out of SD card access during printing, then we could be golden, correct?

I will validate with Luc.

ModMike, 2018-08-25:
By the way, the SD card only seems to be read at specific times by the Web UI:

1) When logging in the first time
2) When uploading a file
3) When refreshing the directory
4) When selecting a job to print

The more I look at this the less I think Marlin needs to worry because the Web UI interventions are on user demand.

p3p, 2018-08-25:
This is a bit troubling, avoiding corruption (in any situation) will be impossible if an external device can just disconnect the LPC176x from the SPI bus whenever it feels like. Even with it only doing so on user interaction.

I guess the ESP8266 can be the only way to control Marlin, no other external control, so that it can keep track of state and not cause issue.

ModMike, 2018-08-25:
The SD Detect pin wired straight to the 8266.  Can I add the following line to disable (or rather force) Marlin SD detect?

#define SD_DETECT_PIN -1

gloomyandy, 2018-08-25:
Hmm that is not ideal. I suppose it is no worse than just pulling the card out of the reader at some random point (which is not a good thing to do!), The good news is that Marlin does not write to the card very often. I wonder if this board has any way to tell Marlin to stop accessing the SD card (is there a gcode to do that?) and to inform it that the contents have changed. It would really help to see the actual schematic (and the source for the modified ESP3D that I think they use). 

I'm not sure what defining a pin to -1 will do, leaving it undefined might be a better option, that basically says I don't have that pin. If SD_DETECT_PIN is undefined then this will certainly trigger different behaviour from Marlin (it will add extra menu items to allow the user to refresh the SD file list for instance).

@p3p have you any thoughts on the best way to handle a "shared" SD card? I think the DUE HAL has some sort of support for this (it seems to try and share the card between USB access and direct Marlin access), but I've no idea how well it works.

ModMike, 2018-08-25:
As I mentioned, I was told the schematics were not for public consumption, I am not sure if that means they were given to me in confidence or not.  I asked Panucatt for a clarification because I take NDAs very seriously but they do take some time to respond so I have no idea when I will get permission.

I may be speaking from ignorance, as my wife claims most of the time, but I am not sure what you mean by share.  The multiplexer is an either/or type of arrangement which selects the MCU by default.

The switch itself is controlled by the 8266 and I have asked Luc how it is controlled but based on the interface I believe it is set to default to MCU and triggered only during the 5 specific events I mentioned in my earlier post.

p3p, 2018-08-25:
> Are we sure the USB is fully disabled with the instructions provided by P3P?

I put together the framework so, yea.. maybe

> What serial port settings do I need in Configuration.h if any for SD card access? I currently have the first 1 set to 0 for Web UI and 2nd disabled.

That's fine, you only use the one serial port.

> have you any thoughts on the best way to handle a "shared" SD card?

@gloomyandy The plan is to have to mount / dismount the card, if its mounted on one the other can't access it until its unmounted, The USB MTP was actually designed to get around this issue as it doesn't access the block device directly.. I did look for a nice open implementation I could instead of MSC but didn't find anything at the time.

gloomyandy, 2018-08-25:
There is no direct serial port access to the SD card so any access will need to go via gcode presumably via the ESP3D interface, so unless you want some additional serial access to another device like a PC you should not need any other settings.

The USB interface should be disabled, but if you are not plugging a USB cable in then it may not even matter. 

In terms of "sharing" what you are doing is switching the SPI access to the SD card. This has at least a couple of issues. Firstly Marlin does not expect that the SPI connection can be "turned off" at any time (unless you remove the card), so it could potentially be doing some operation on the card when the ESP3D unit jumps in and disconnects it. Because it is not expecting this, Marlin may treat this as an error and assume that the card has been removed. But possibly even worse is that Marlin does not expect the content of the card to change "while it is not looking", this includes the various system level structures that are used to manage the filesystem. When you use the ESP3D interface to say upload a file, these structures will change, at best this may mean that Marlin is simply not aware of the new file, at worst Marlin may end up writing system level structures back to the disk that conflict with the ones written by the ESP3D operations, net result a corrupt SD card. If Marlin is not writing to the card (so only opening files read only) you may get away with it, but if for instance it is using the card to store things like the power off recovery file (or emulate eeprom), then "bad things may happen". 

@p3p Yes something like MTP would have been a better basis for this sort of thing! I had a look at the DUE code and they seem to do various tests for open files and printing and whatever and if so deny access to the USB side of things, but I didn't see anything to handle the other side of things so that the SD card is owned by the USB code, or anything to force a directory refresh after changes from the USB side of things. I wonder if that DUE code actually works? Do you think it will be possible to automate the mount/dismount, or will it require user intervention?

ModMike, 2018-08-25:
Well the eeprom is saved to hardware so thats safe.  Here is a screen shot of the Web UI 1.0 (I will get 2.0 beta access soon).

[image]

Luc mentioned that the only first option is relevant to Marlin.  I tried to disable it by setting all to no.  He said the rest were Smoothie specific.  If Smoothie SD card access is managed by WEB UI, maybe Luc can adapt or repurpose it for Marlin.

For now, I would consider it a huge victory of we could just see the card and worry about SD card access handling after.

ModMike, 2018-08-25:
As a final check, can someone tell me if this is what I needed to put in my boards pin file?

//
//  SD card SPI pins
//
#define SS_PIN            P0_06
#define SCK_PIN           P0_07
#define MISO_PIN          P0_08
#define MOSI_PIN          P0_09
#undef SD_DETECT_PIN

ModMike, 2018-08-25:
In the board pin file?  By the way I just updated my previous post on ESP3D settings with the translated version of Luc's description, if it helps.

gloomyandy, 2018-08-25:
Yes in the pin file (at least it is set there for the SBASE), makes sense to keep it together with the pin changes, as it is those changes that mean you need to use software rather than hardware SPI.

ModMike, 2018-08-25:
Thats were I had put it but still no love. What is your best guess, access being blocked or wrong pins?

gloomyandy, 2018-08-25:
I'd expect the pins to be OK. What happens? Have you tried using the LCD to request that the SD card is refreshed/initialised? Do you get any sort of listing? Did you remove the call to MSC_SD_Init(0)?

ModMike, 2018-08-25:
Update from Luc:

"Yes by default Marlin will have full control of SD - it is up to ESP3D to switch to get access to SD or not.

So to answer it is really better not to parse SD from web UI when printing because it will cut even very quickly the SD access from Main board,

2.0 has a read only mode that can be set by default to avoid this - I am still thinking how to add an automatic check to see if printing is ongoing or not, to allow (or not) to disable the read only mode

I have read the discussion on Marlin github, about SD - the mecanism is transparent for Marlin and 2.0 work already with RE-ARM + Specific daughter boards using the second SD card reader present on Screen like Viki2 or GLCD. So no need to focus on Azteeg wifi specificity with ESP3D - it is not the problem."

ModMike, 2018-08-25:
Here are my files to confirm I did it properly. 

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2320869/Archive.zip)

If you guys can give me your email address, I can send you the schematics as long as you do not distribute without Panucatt permission.  Do you have Dip Trace?  It's free but if its too much hassle I can convert to .pdf

By the way, how do I get my contributor badge?

ModMike, 2018-08-25:
Sorry Andy, forgot to answer:

I think I removed the call properly (attached files in previous post) and I did try to use the LCD to init the card but it just hangs for about a second.  I also tried to manually mount the card thorough the Web UI using M21 and read with M21.  No files listed.

gloomyandy, 2018-08-25:
You have SD_DETECT_PIN defined later on in the file (in the LCD section, not sure if that part is used or not), I would put the new defines at the end of the file just to be safe. So what happens when you run this? Do you get the menu items to let you refresh the SD? What happens if you use them? If you do not get the extra menu items then you must have SD_DETECT_PIN defined somewhere!

ModMike, 2018-08-25:
I do get 2 extra menu items:

NO SD Card
Init. SD CArd

ModMike, 2018-08-25:
So no difference when at end of file.

Update from Luc concerning Web UI SD card access:

1) When uploading a file
2) When deleting a file or creating directory
3) When refreshing the directory

There is no access to SD at all when selecting print job and printing, it just sends just send GCODE to printer.

gloomyandy, 2018-08-25:
Hmm couple of things you could try...
1. Try changing SPI_SD_INIT_RATE in SdFatConfig.h to 6 (from 5) as you have the faster 1769 cpu I think.
2. Try changing the SPI speed settings in Configuration.h to see if a slower speed helps.
Other than that I'm afraid you are into serious debugging try looking at the Sd2Card.cpp file and perhaps adding some trace debug (which you may struggle to see as you have turned off USB access and don't have a serial interface hooked up to a PC). 

Good luck!

p3p, 2018-08-25:
okay .. had time to look into this, and everything we said would have been correct if there wasn't a bug in the software spi initialisation ;) 

The bootloader initialises the (Hardware) spi bus and attaches the pins in order to check and update the firmware on boot, this means that pins cannot be toggled by software, in the software spi initialisation the pins where assumed to be in default state (0) and not attached to anything so the direction was set but not the mode, meaning any output to those pins had to effect,

long story short, in `HAL/LPC1768/SoftwareSPI.cpp` change the swSpiBegin function to this:
```cpp
void swSpiBegin(const pin_t sck_pin, const pin_t miso_pin, const pin_t mosi_pin) {
  pinMode(sck_pin, OUTPUT);
  if (VALID_PIN(miso_pin)) pinMode(miso_pin, INPUT);
  pinMode(mosi_pin, OUTPUT);
}
```

I have successfully accessed the onboard sdcard from Marlin using it.

forkoz, 2018-08-26:
So not possible to disable MSC sharing of the card and using SW SPI to just access the card through marlin?

gloomyandy, 2018-08-26:
@forkoz yes it should be, but the code needs to be tweaked a little to make it work. That is what most of this thread is about.

forkoz, 2018-08-26:
I thought he was disabling all of USB. I wanted to try just MSC disabled but keep the USB control. I have the locks shut off and can load eeprom from the card. In theory marlin should also be able to load gcode in this state (yes, risking corruption) but that's probably stopped by the SDSUPPORT implementation taking full control of the card. I tried it on a lark but can never get anything except .. visible on the sd.

p3p, 2018-08-26:
You shouldn't need to disable the entire usb stack, I was walking @ModMike though relatively complex code modifications I hadn't tried before, so I was just removing things that may have conflicted, 

The important parts are:
- define FLASH_EEPROM
- comment out MSC_SD_Init(0) 
- update your boards pin file
  - update spi pins
  - define LPC_SOFTWARE_SPI
- fix swSpiBegin

> I tried it on a lark but can never get anything except .. visible on the sd.

Don't know what you mean by that

forkoz, 2018-08-26:
I mean that I tried to have eeprom, msc and marlin sd at once.

ModMike, 2018-08-27:
> The important parts are:
> 
> define FLASH_EEPROM
> comment out MSC_SD_Init(0)
> update your boards pin file
> update spi pins
> define LPC_SOFTWARE_SPI
> fix swSpiBegin

My 8266 is still locked, so hoping Panucatt gives me a reset procedure today.  In the mean time i've (hopefully) isolated the 8266 from the system by removing it's TX and RX header pins and now Marlin boots properly.  The 8266 could still be causing other issues but Marlin did boot properly after the jumpers were removed.

I still can't read the SD card, here are the changes I made based on this threads instructions:

1) `#define FLASH_EEPROM` in Marlin/src/HAL/HAL_LPC1768/persistent_store_api.h

This works because Marlin is able to save and read memory and no eeprom.dat file is written to SD

2)  `//(void)MSC_SD_Init(0)` in Marlin/src/HAL/HAL_LPC1768/main.cpp

Seems to have disabled the USB port.  Computer still sees the port but cannot access it.  Also tested with USB cable disconnected.

3) Added this to the end of my boards pin file `Marlin/src/pins/pins_AZTEEG_X5_MINI_WIFI.h`:

```
//
//  SD card SPI pins
//
#define SS_PIN            P0_06
#define SCK_PIN           P0_07
#define MISO_PIN          P0_08
#define MOSI_PIN          P0_09
#undef SD_DETECT_PIN
#define LPC_SOFTWARE_SPI
```

4) Replaced the following in my `Marlin/src/HAL/HAL_LPC1768/SoftwareSPI.cpp`:

```
void swSpiBegin(const pin_t sck_pin, const pin_t miso_pin, const pin_t mosi_pin) {
  pinMode(sck_pin, OUTPUT);
  if (VALID_PIN(miso_pin)) pinMode(miso_pin, INPUT);
  pinMode(mosi_pin, OUTPUT);
}
```

forkoz, 2018-08-27:
Try to comment out the rest of the USB stuff like in this post: https://github.com/MarlinFirmware/Marlin/issues/11639#issuecomment-415853441

I had the same problem with what I tried, SD contents are never read.

gloomyandy, 2018-08-27:
Do you have any gcode on the sd card? I think only gcode files (and directories)  show up on the LCD. You will have to init the SD as well. You could try commenting out the USB startup code, but I don't think it should really be required.

ModMike, 2018-08-28:
Sorry about the 8226 induced absence.  All sorted now and decoupled from SD.

Yes I do have Gcode on it but still no love. Any other ideas?

ModMike, 2018-08-28:
Andy, still interested in the schematics?

gloomyandy, 2018-08-29:
Did you try it with the USB stuff disabled? Yes to the schematics my email is andy at gloomy-place.com let me know when you see this and I'll edit my post to delete the email address.

ModMike, 2018-08-29:
Yes.

MT

ModMike, 2018-08-29:
Ok delete your email.

ModMike, 2018-08-29:
Andy, was it .net or .com?  By the way so want to build an R/C sub!

gloomyandy, 2018-08-29:
Either should get to me! I got your email and replied. Will be a day or two before I can really look at this again.

ModMike, 2018-08-29:
No problem, thanks!

MT

ModMike, 2018-08-30:
Just a quick note to let you all know that I installed a version Luc's Web UI that has been stripped of SD functionality.  Still not working.  I also tried to flash:
```
void setup() {
}

void loop() {
}
```
so it would be as blank as possible.  Unfortunately Marlin refused to start, which is very weird.

ModMike, 2018-09-08:
@p3p @gloomyandy 

Hi guys, any progress on this?  I know P3P said he was re-writing the frame work but I haven't heard anything since.

p3p, 2018-09-09:
@ModMike I'm putting what time I can into doing that, life gets in the way, pulling apart the framework and the HAL is not trivial at this point and I don't want to add regressions in functionality.

I stopped responding here mostly because the procedure I outlined works for me on the Re-ARM and SBase, I didn't have the hardware (or time atm) to go deeper into understanding why you are still not having any luck.

ModMike, 2018-09-09:
Thank you for the update and I do understand.  I updated my repo and will try again.  While you are here, can you validate something for me?

I did not set Serial port 2 to -1.  Wouldn't that preclude the marlin host from communicating?  I understand it is not a real serial port, am I totally off?

Lastly, can you send me your working re-arm config?  I'll work with that and leave you alone to finish up?.

Thanks again!

p3p, 2018-09-09:
> I did not set Serial port 2 to -1. Wouldn't that preclude the marlin host from communicating? I understand it is not a real serial port, am I totally off?

If you are using the USB to send serial data then you need at least one of the Serial ports to be set to -1 to enable that, if you are only using the hardware serial then only enable one serial port with the appropriate hardware port number.

> Lastly, can you send me your working re-arm config?

I'l have to recreate it from my guide above

gloomyandy, 2018-09-09:
I'm still working on things in this area, but I got side tracked looking into a problem that causes the USB serial connection to lose characters if when a reset is caused by the SD card lock code. Anyway I had hoped to get the PR for this over the weekend, but some family things have come up (I'm in effect a carer for my Mother) which means I didn't get it done. With luck I may get to it during the week. It may help with what you are trying to do, but I still don't really understand why what you have now is not working to some extent at least.

ModMike, 2018-09-09:
Thanks for the update, I look forward to getting it sorted.

[8 more replies]

closed 2018-10-17 by ModMike