MarlinFirmware/Marlin #11643: Printer increases temp to 325C when set to 235C

landodragon141, 2018-08-25:
### Description

Printer increases temperature to 325C on the second layer when it is set at 235C
This occurs on the latest stable branch downloaded 2018-08-24
This happened to me first on the initial release of 1.1.9 so I downloaded the latest revision from the stable branch yesterday and tried it again with the same result. I only noticed originally because the filament which was orange was becoming more neon yellow and started stringing like crazy. I'm so glad I stayed to watch the first couple layers.

### Steps to Reproduce

1. Sliced this file. 
2. Start print
3. wait for layer 2 and clench

**Expected behavior:** [What you expect to happen]
Printer switches from 240C (first layer temp) to 235C

**Actual behavior:** [What actually happens]
Printer increases to 325C while oddly still showing that the set point is 235C on the screen. 

#### Additional Information
I'll attach a picture of the LCD screen if desired showing that it still thinks it's set to 235 although the temp reading goes up to 325C also I do not have thermal runaway disabled. 
[MarlinIssueFiles.zip](https://github.com/MarlinFirmware/Marlin/files/2320878/MarlinIssueFiles.zip)

thinkyhead, 2018-08-30:
Try a different slicer and see if it makes any difference.

landodragon141, 2018-08-31:
I can. I also I have the G code and my configuration attached in that zip if anyone feels like trying to run the gcode dry. I actually checked the raw gcode and the values for the M140 are correct.

landodragon141, 2018-08-31:
So here's an additional question. How is it getting to 325 when my max hotend temp is set to 300? (I'm using an e3d pt100) To my understanding the thermal runaway should be triggered way before it gets to 325.

thinkyhead, 2018-09-01:
You're right that it makes no sense. You might want to rebuild and re-flash Marlin to see if it corrects itself. Also, `M502`, `M500` after the flash.

landodragon141, 2018-09-01:
Well the reason I posted is that I was having the issue with the original release of 1.1.9 so I updated to the latest hoping it would be resolved. Alas the rest is history. Also I always do the M502 M500 thing when I re-flash.

AnHardt, 2018-09-01:
Temperature table 20 is relatively rarely used - compared to the normal thermistors; so not very well tested.
Table 20 has the opposite slope than the normal thermistors. Low ADC-values correspond to low temperatures here. That may change the way the raw max- and - min-temperatures have to be calculated. It also changes the test for what is in- and what is outside the *good* interfall. In one of the both there may be an up to now undiscovered error.

Happy macro expanding.

AnHardt, 2018-09-01:
Maybe I got it.
Either it needs an extra condition to prevent 'HEATER_0_USES_THERMISTOR' from being set in Conditionals_post when 20 is selected.
Or it needs a condition not to redefine the in table_20 already correctly defined HEATER_0_RAW_HI_TEMP and HEATER_0_RAW_LO_TEMP in thermistortables.h .

AnHardt, 2018-09-02:
I did not, but have an other suspicion what i'm investigating.

AnHardt, 2018-09-03:
```
    const bool heater_on = 0 <
      #if ENABLED(PIDTEMP)
        soft_pwm_amount[e]
      #else
        target_temperature[e]
      #endif
    ;
    if (rawtemp > maxttemp_raw[e] * tdir && heater_on) max_temp_error(e);
```
If we have PIDTEMP and the current temperature is already high enough for the PID to decide we do not need any further heat, so adjusting the soft_pwm_amount to 0, we can't get a max_temp_error.

@thinkyhead 
Can we base the decision upon heater_on on  target_temperature always?
Can't find why we do use soft_pwm_amount.

AnHardt, 2018-09-04:
Booting with a stuck heater is still undetected because of the changes needed for the fu**ing "high temperature thermistors".

AnHardt, 2018-09-04:
See further comments in https://github.com/MarlinFirmware/Marlin/pull/11718#issuecomment-418394884

thinkyhead, 2018-09-05:
The patches have been merged, so try the bugfix branch to see if it's behaving better now.

thinkyhead, 2018-09-05:
> Can we base the decision upon `heater_on` on `target_temperature` always?
> Can't find why we do use `soft_pwm_amount`.

Probably just preserving existing behavior from long, long ago. We can follow the Git Blame history to trace back to the first commit where this was used and see what the justification was at that time.

thinkyhead, 2018-09-06:
I suspect the whole reason for using the PWM instead of the target temp is that the temp used to be a 4 byte float. Now it's a 16-bit integer, so using the PWM is no longer an optimization. So, we're all good!

AnHardt, 2018-09-06:
Would you please follow my links.
"Only issue MAXTEMP when heating" #5082.
Was pure madness. Defining  `BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE` (omitting the kill()) would have done the job.

closed 2018-09-10 by thinkyhead

landodragon141, 2018-09-16:
FYI I figured out what caused my temperature to go so high. I had enabled PID extrusion scaling with the default Pc of 100. I print infil at the same speed as a Prusa i3 MK3 about 180mm/s. Apparently that's enough to send the temps over 300. On a side note with the new fix it actually through an error and stopped my print so huzzah to that!

landodragon141, 2018-09-16:
Crap I lied it still starts to increase on the second layer.

thinkyhead, 2018-09-16:
I don't understand PID scaling very well, but it seems that we should throw an error and have the printer shut down if the temperature of the nozzle goes over the `HEATER_n_MAXTEMP` setting in any situation. There doesn't seem to be a check against that anywhere at the current time.

In the meantime, it's probably best to avoid using `PID_EXTRUSION_SCALING` since it's causing your heater to stay on for too long.

AnHardt, 2018-09-16:
> Crap I lied it still starts to increase on the second layer.

Next layer or part cooling fan?

> I don't understand PID scaling very well, ...

### The problem
It needs energy to melt filament. To heat the filament the heater block around the nozzle is cooled down. The cooling takes some time to reach the temp (dT1). sensor. Now the PID can begin to work against that cooling and gives some more power to the heater. The additional heat starts creeping thru the heater bock. After some time (dT2) the heat reaches the nozzle and after an other time (dT3) the heat reaches the sensor. (dT2 can be larger than dT3 or the other way around. )
```
--------- Heater 
| H N S | Nozzle    Heaterbock. H,N,S can have any sequence and distance.
--------- Sensor
```
### The idea
Avoid oscillations of the nozzles temperature by putting in the melt energy early, so the nozzle does not have to cool down until the energy is provided by the PID.
### The code
```cpp
        #if ENABLED(PID_EXTRUSION_SCALING)
          cTerm[HOTEND_INDEX] = 0; // cTerm[] is only used for debug output
          if (_HOTEND_TEST) { // Is this the right heater?
            const long e_position = stepper.position(E_AXIS);
            if (e_position > last_e_position) {
              lpq[lpq_ptr] = e_position - last_e_position; // amount of planed! extrusion since the last call ~1/5second before
              last_e_position = e_position;
            }
            else
              lpq[lpq_ptr] = 0;

            if (++lpq_ptr >= lpq_len) lpq_ptr = 0; //lpw_len is a delay in ~1/5second
            cTerm[HOTEND_INDEX] = (lpq[lpq_ptr] * planner.steps_to_mm[E_AXIS]) * PID_PARAM(Kc, HOTEND_INDEX);
            pid_output += cTerm[HOTEND_INDEX]; // PID_output is adjusted directly. No chance to throw errors
                                                                              // based on MAX_TEMP. Array could be omitted.
          }
        #endif // PID_EXTRUSION_SCALING
```
### What is the code doing?
Determine how much filament is planned to extrude. Apply the energy delayed by some fixed time.
### What's wrong with the code?
The code tries to **delay** the additional melt energy by some **fix** time.
From the problem we can see we have to apply the heat before to avoid the drop. 
So there can only be one reason for a delay - bridging the time between planning and executing (in the planner buffer). Sadly this time varies widely from a few ms to several seconds.
### Conclusion:
***Can't work as expected!***
### What's needed?
Provide more energy a fixed time before the filament is melted, delayed by the amount of time in the planner buffer. If the sum of these times becomes negative, apply the energy immediately. This will be to late but hopefully faster than waiting for the temperature drop.   

> shut down if the temperature of the nozzle goes over the HEATER_n_MAXTEMP

This is already fixed now.
Except you mean we should hinder to order temperatures above MAXTEMP. I haven't looked that up now, but i remember we limited the input temperature in the menus to MAXTEMP - 15?, to avoid the overshot to go above MAXTEMP. I can remember this is different for setting the temp by g-code, but i don't, if not checked at all or only the margin is omitted.

landodragon141, 2018-09-16:
Sorry let me clarify my previous comment.
I thought that the issue that was causing the temperature to increase out of control was PID_Extrusion_Scaling. I failed to properly confirm my ASSumption before I posted here that I had identified the problem. After disabling PID_Extrusion_Scaling the moment the second layer starts the temperature just starts increasing for no apparent reason. As I previously mentioned the good news is that I can confirm that the fix to max temp works since it will now fatal error instead of continuing to print with the heater at full. Later tonight I'll do a test and start this same print while monitoring it with repetier server. That will let me see not just what the temperatures are doing but the heater itself. 

This is what should happen:
1st Layer E0-240C Bed-67C 
2nd Layer E0-235C Bed-67C

This is what actually happens
1st Layer E0-240C Bed-67C
2nd Layer E0-TO THE MOON! (Right now my max temp is set at 275C so it hits that and kills the printer) Bed-67C

landodragon141, 2018-09-16:
@AnHardt 
I think there are two possible ways to really execute PID Scaling correctly.
1st cobble together a way of doing it in the slicer. This would require some kind of calibration to be run to inform the slicer about the printers heating and cooling characteristics.
2nd we would need a much much much larger planner buffer so that we could actually compensate for temperature multiple seconds in advance of E movements.

AnHardt, 2018-09-16:
I still don't understand.
Is the **target** temperature set to infinity at layer 2, or is the right **target** temperature displayed but the **is** temperature rises until the printer resets.

Please cite the g-code at the layer change.

AnHardt, 2018-09-16:
Please also append the config files and the used pins file.

landodragon141, 2018-09-16:
@AnHardt The right target temperature is displayed but the actual continues to rise until it resets. 
With regards to configs you will find both the config files and the exact gcode file in the attachment posted towards the top of this thread.

landodragon141, 2018-09-16:
I have made some changes to the MKS_GEN_13 file but not to the fan or heater.
[pins.zip](https://github.com/MarlinFirmware/Marlin/files/2387109/pins.zip)

AnHardt, 2018-09-16:
Yes. Your changes should not cause the problem. Will dive into pins_RAMPS.h tomorrow. Would be a great help for me if you coud state where you connected fan and heater and if the fan works - to eliminate a few dozen cases. Could also be a physical short between them.

landodragon141, 2018-09-17:
As I said earlier 
Fan D9
Hotend D10
Fan works although I have noticed that it is starting up when the hotend turns on..... I'll have to investigate a possible short.

AnHardt, 2018-09-17:
Bingo!

AnHardt, 2018-09-17:
As a workaround you could try to edit  pins_RAMPS.h from
```
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN     10
#endif
```
to
```
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN     7
#endif
```
And connect the heater to D7.

landodragon141, 2018-09-18:
Good point hopefully I'll have a chance to look into it tonight.

landodragon141, 2018-10-03:
FYI - I discovered my problem and it turns out that once again I am my own worst enemy. What ended up happening is that I had a few wires switched around. Because I use 24 gauge ethernet cable for my wiring I use two pairs for the hotend heater but only one pair for the layer cooling fan. I some how made a bozo move and wired one hotend pair and the fan to the E0 mosfet while connecting the second hotend pair to the fan mosfet. So in the end it was entirely my fault although as a result we did fix a critical bug in the thermal runaway code so I feel good about that.