MarlinFirmware/Marlin #11644: Marlin 1.1.9 Heatbed no longer heating or reading temperature

Gaz83, 2018-08-25:
I am using Octopi to control my printing and ever since upgrading to Marlin 1.1.9, my heatbed no longer heats up and does not read the temperature.

My heat bed is from a duplicator i3. 

I set the bed thermistor to 99 and downloaded a custom thermistor table to match. This has worked fine in version 1.1.8 but does not work in 1.1.9

[MarlinData.zip](https://github.com/MarlinFirmware/Marlin/files/2320882/MarlinData.zip)

kAdonis, 2018-08-25:
you have set:

```
#define TEMP_SENSOR_BED 0
#define TEMP_SENSOR_CHAMBER 99
```
what you want is:
```
#define TEMP_SENSOR_BED 99
#define TEMP_SENSOR_CHAMBER 0
```

Gaz83, 2018-08-26:
Well I am a dumb ass lol. Right so I corrected that and now I have a new issue. Every time I try a print I get this.

Error:MINTEMP triggered, system stopped! Heater_ID: bed

I had a quick look in the configuration file and noticed #define BED_MINTEMP 5. When I monitor in Octoprint it says the bed is off, where as the extruder is giving me a temperature reading of about 20 degree c.

So I change the 5 to a 1. Now I don't get the error but I did notice that in Octoprint that the bed says off until it hits 10 degrees c, then I get a reading.

Does this seem normal?

kAdonis, 2018-08-26:
This it not normal
The temperature readings from bed and hotend should both show room temperature, when the printer is cold.
Your thermistortable is wrong.
You could try one of standard tables and measure the real temperature yourself to find the best table for your thermistor

In the example config for the duplicator 6:

`#define TEMP_SENSOR_BED 1`

AnHardt, 2018-08-26:
From your thermistor table:
> // 100k bed thermistor (Thermistor table 1 but -20 Deg)
const short temptable_99[][2] PROGMEM = {

Does it make you wonder?

Gaz83, 2018-08-26:
Nope, because I'm a newbie lol.

boelle, 2019-02-20:
@Gaz83 is this still an issue when using bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-17 by p3p