MarlinFirmware/Marlin #11648: TMC 2130 overtemperture error

BlackWolf57, 2018-08-26:
Hi all.

Last week I have an overtemperature on my Z driver with the latest 2.0 firmware.
The config is always the same, I just activate the "Monitor driver status".

The problem is it decrease the current, but don't know why, it fall to 0 and jump to a 0xFFFFFFFF value and decrease with this new value.

I think thats an unsigned value, so can you add a condition to stop decrease when it reach 0 !
Not upload on my printer, but something like:
```
    #if CURRENT_STEP_DOWN > 0
      // Decrease current if is_otpw is true and driver is enabled and there's been more than 4 warnings
      if (data.is_otpw && st.isEnabled() && st.getCurrent() > 0 && otpw_cnt > 4) {
        st.setCurrent(st.getCurrent() - CURRENT_STEP_DOWN, R_SENSE, HOLD_MULTIPLIER);
        #if ENABLED(REPORT_CURRENT_CHANGE)
          _tmc_say_axis(axis);
          SERIAL_ECHOLNPAIR(" current decreased to ", st.getCurrent());
        #endif
      }
    #endif
```
on the tmc_util.cpp

Thanks.

teemuatlut, 2018-08-26:
Yes there's a sanity check that has been implemented already but is not in upstream yet.
The main question however is why you're getting over temp errors at zero current settings.

When everything is working properly, the current would not get to very low values as the over temp condition should get cleared with lower settings.

Set `CURRENT_STEP_DOWN` to `0` if you wish to bypass this feature.

BlackWolf57, 2018-08-28:
I don't know why I have an overtemp at 0 current.
I run fine with 750 mA setting and passive cooling. But I don't use the monitor feature. With the latest firmware I activate monitor feature and see this.

I take a look on the code and the TMC datasheet and I don't really understand.
On code when there is an pre-warn, current decrease but I don't see any pre-warn clear flag.
On the datasheet they don't say that pre-warn flag need to be cleared. Perhaps it can clear itself ?
The only clear flag is called with M912 but it's a manual operation.

boelle, 2019-02-20:
@BlackWolf57 tried actice cooling+heatsinks and latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead