MarlinFirmware/Marlin #11652: Unknown Commands M122 / M906; HAS_TRINAMIC flag seemingly not correct

Treynolds416, 2018-08-27:
### Description

I discovered this issue while trying to troubleshoot my corexy setup with all TMC2130 drivers for persistent layer shifting (will open another issue regarding that in more detail if necessary). I just converted my 1.1.8 config files to a fresh install of 1.1.9 and made no real changes (files attached below). 

### Steps to Reproduce

1. Enable TMC_DEBUG in advanced config
2. Connect printer, and try M122 S1 and/or M906
3. Host (repetier in my case) claims that both are "Unknown commands"
I tried commenting out the if checks containing the gcode functions in Marlin_main.cpp to confirm that the problem is HAS_TRINAMIC, but had limited time and no luck compiling.

#### Additional Information

Should I be using one of the bugfix branches instead of the 1.1.9 release?

[Marlin-1.1.9-configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/2322382/Marlin-1.1.9-configfiles.zip)

teemuatlut, 2018-08-27:
You need to define the driver types.

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L532-L556

Treynolds416, 2018-08-29:
Yep, that was the issue... I specified the drivers and forgot to uncomment... :/

Thank you for the help.

closed 2018-09-09 by thinkyhead