MarlinFirmware/Marlin #11655: Ramps 1.7 and DUE

boelle, 2018-08-27:
Hi all

@MrAlvin havde made the next version of the Ramps board but i have some issues getting it to work, just reading temps and getting a response from M119 is royal pain in the rear

he made a test program which is here: RAMPS_1.7/Arduino test sketches/test everything/ee_ana_blink_end_step/ee_ana_blink_end_step.ino

everything works fine with that

he also for dev purpose made a fork with settings ready to go: https://github.com/MrAlvin/Marlin/tree/bgfx-20x-R17-dev

that one does not work, no temps and no response from M119, M502 does not give a response either

we have tried both arduino ide 1.8.6 and 1.9 and platform io

anyone that can help us spot what is going wrong for us?

workheart, 2018-08-27:
change line nr 24 [env_default = megamega2560](url)
in platformio.ini
to` env_default = DUE`
it compiled without error in platformio :-) 
@boelle, i was to fast but now it worked

boelle, 2018-08-27:
its not compile errors that are the problem

and yes i did change that line in platformio.ino

the problem is when the code is uploaded to the board.... nothing works

boelle, 2018-08-27:
btw... i understand both english and german (and danish) if that helps

not sure about @MrAlvin  but he understands English and danish :-)

workheart, 2018-08-27:
hard to find without the board, guess you have to wait for someone with more DUE expertise,
what does Work on the new board? how many did you make?

boelle, 2018-08-27:
we just have a few prototypes

everything works with the test program, but not with marlin

is it @Bob-the-Kuhn that does a lot of 32 bit stuff? or maybe @Wurstnase ?

boelle, 2018-08-27:
i know that @MagoKimbra did some due stuff once but not sure if he is still active and throw some light on it

MrAlvin, 2018-08-27:
As I have more test tools and options available, I did further testing, and I got the bed and hot-end temperatures to work, by switching bed pin from (analog) pin 5 to pin 7. Pin 7 is marked T3 on the PCB.  Pin 5 is marked Bed on the PCB. 

While things were running (meaning; hot-end and bed were being heated), I also tested the volt reading (using a multimeter) for analog pins 5, 6 and 7.  And with Marlin running, analog pin 5 was stuck at the same voltage, despite physical temperatures were changing.

When running my test programs (as linked to above by Boelle), analog pins 5, 6 and 7 all works, and the multimeter shows voltage changes for all three pins.  With my simple test programs running, analog pin 5  is NOT stuck, and the voltage changes, when the physical temperatures change. 

So a work-around has been identified, but it would still be nice if analog pin 5 could  be used as; an analog pin.

Wurstnase, 2018-08-28:
It's probably a year ago that I've compiled the latest Marlin with PlatformIO on my Due. I had no issues. I need to remove some libs that time, because that didn't worked in the past with the Due. 

On the SAM3X is pretty simple to use the analog inputs programmatically. You don't need any DMA or other things. I didn't check the Marlin code, but probably the analog pin 5 is used also somewhere as digital pin? Please check the numbers. Digital pins and analog pins have different numbers but could share the same one.

MrAlvin, 2018-08-28:
@Wurstnase  Ah, yes.  That was indeed the problem. I had not connected the dots: A5 =D59. And D59 had been typed, where it should have been D58. Typo is now fixed, and analog 5 (bed) is reporting temperatures. 
Thank you :-)

boelle, 2018-08-28:
i will have a go too then :-D

boelle, 2018-08-28:
@MrAlvin do you also get a response when you send M119

me still unzipping

MrAlvin, 2018-08-28:
hmmm.... M119 is on a different issue. But I will do some testing, and let you know.  Ok?

boelle, 2018-08-28:
hehe, the reason why i asked is that before i could not get any response from anything :-)

ie M119 would not reply if endstops was triggered or not

M114 shoould tell you where the board thinks the nozzle is, always 0,0,0 after reset

boelle, 2018-08-28:
yaaa :-D

i get response from both M119 and M114

and i can control bed heating and nozzle heating

big thanks to @Wurstnase so far

boelle, 2018-08-28:
The only thing i have not beed able to test with success are

Fan on with M106 and off with M107, there is no response on serial output and no reaction on board

the board has an extra fan connector on digital 8, no reaction with M42 P8 S255 or  M42 P8 S0, not on board or serial

X max does not change status

my printer is the only one that has drivers that run on 3.3V logic and that one is doing a print so will have to wait testing the steppers

workheart, 2018-08-28:
I recommend the m m43 t option to triger pins for dwbuging purpose. Not suute about the 43 but google for set pin state it has the possibility to test lets say 10 pins and check with a multimeter if the pin pex: 5 is wher you think it is it tells you what is the internal and the analog and the external pin name etc. Verry handy on board dev.

workheart, 2018-08-28:
> i will try that
> how to debug output? also with M43?

Yes its easy follow the marlin docu,
http://marlinfw.org/docs/gcode/M043-T.html  set the time to .. 15seconds
that you have terminal output what pin gets triggerd (it switches high low
high low...) And informatin on the pins,
I do disconect my drivers etc so i can probe eerything else if it is not
where or what i expected. If i than dont find it i run the code without the
shield, .... On step at a time :) its time consuming but it helps trouble
Shooting

MrAlvin, 2018-08-28:
Hi boelle,

I am today working to get the standard (5V) LCD displays up and running with the 3V Due.  I have made a universal bi-directional adapter board, so all pins on aux3 and aux4 are level shifted  between 3V and 5V. This should allow for any existing display to be able to be used with this RAMPS 1.7 + Due. 

When this is done (or when I take a break from the LCD stuff) , I will look into the other issues you mention, like: 

- Fan on with M106 and off with M107, the board has an extra fan connector on digital 8, but no reaction with M42 P8 S255 or M42 P8 S0, not on board or serial

- X max does not change status. 

- my printer is the only one that has drivers that run on 3.3V logic and that one is doing a print so will have to wait testing the steppers. 
A comment: so far the steppers I have tested with (A4988 and DRV8825) have been working like usual.

boelle, 2018-08-28:
strange... i use M43 W to tell me if a pin changes state

nothing happens when i short the X max connector, and as i understand things it should at least tell me a pin number that changes state

will try M43 I to get state of all pins and see if one of them changes

boelle, 2018-08-28:
only pin 54 changed

`
PIN: 00        RXD                                    Input  = 1
PIN: 01        TXD                                    Input  = 1
PIN: 02        E_MUX0_PIN                             Input  = 1
.              X_MAX_PIN                              Input  = 1
PIN: 03        X_MIN_PIN                              Input  = 1
PIN: 04        SERVO0_PIN                             Input  = 1
PIN: 05        SERVO1_PIN                             Input  = 1
PIN: 06        E1_CS_PIN                              Input  = 1
.              FIL_RUNOUT_PIN                         Input  = 1
.              SERVO2_PIN                             Input  = 1
.              Z2_CS_PIN                              Input  = 1
PIN: 07        SERVO3_PIN                             Input  = 1
PIN: 08        E0_AUTO_FAN_PIN                        PWM = 0
PIN: 09        FAN_PIN                                PWM = 0
.              RAMPS_D9_PIN
PIN: 10        HEATER_BED_PIN                         Output = 0
.              RAMPS_D8_PIN                           Output = 0
PIN: 11        HEATER_0_PIN                           Output = 0
.              RAMPS_D10_PIN                          Output = 0
PIN: 12        PS_ON_PIN                              Output = 0
PIN: 13        LED_PIN                                Input  = 0
PIN: 14        <unused/unknown>                       Input  = 1
PIN: 15        E_MUX1_PIN                             Input  = 1
.              Y_MAX_PIN                              Input  = 1
PIN: 16        LCD_PINS_RS                            Output = 0
PIN: 17        LCD_PINS_ENABLE                        Output = 0
PIN: 18        <unused/unknown>                       Input  = 1
PIN: 19        E_MUX2_PIN                             Input  = 1
.              Z_MAX_PIN                              Input  = 1
PIN: 20        <unused/unknown>                       Input  = 1
PIN: 21        <unused/unknown>                       Input  = 1
PIN: 22        E1_ENABLE_PIN                          Output = 0
.              Z2_ENABLE_PIN                          Output = 0
PIN: 23        LCD_PINS_D4                            Output = 0
PIN: 24        E1_STEP_PIN                            Output = 0
.              Z2_STEP_PIN                            Output = 0
PIN: 25        LCD_PINS_D5                            Input  = 1
PIN: 26        E1_DIR_PIN                             Output = 0
.              Z2_DIR_PIN                             Output = 0
PIN: 27        LCD_PINS_D6                            Input  = 1
PIN: 28        <unused/unknown>                       Input  = 1
PIN: 29        LCD_PINS_D7                            Input  = 1
PIN: 30        E0_ENABLE_PIN                          Output = 0
PIN: 31        BTN_EN1                                Input  = 1
PIN: 32        <unused/unknown>                       Input  = 1
PIN: 33        BTN_EN2                                Input  = 1
PIN: 34        E0_STEP_PIN                            Output = 0
PIN: 35        BTN_ENC                                Input  = 1
PIN: 36        E0_DIR_PIN                             Output = 0
PIN: 37        BEEPER_PIN                             Output = 0
PIN: 38        E0_CS_PIN                              Input  = 1
PIN: 39        <unused/unknown>                       Input  = 1
PIN: 40        MAX6675_SS                             Input  = 1
PIN: 41        KILL_PIN                               Input  = 1
PIN: 42        Y_CS_PIN                               Input  = 1
PIN: 43        <unused/unknown>                       Input  = 1
PIN: 44        <unused/unknown>                       Input  = 1
PIN: 45        <unused/unknown>                       Input  = 1
PIN: 46        X_CS_PIN                               Input  = 1
PIN: 47        <unused/unknown>                       Input  = 1
PIN: 48        Z_CS_PIN                               Analog in =     0
PIN: 49        SD_DETECT_PIN                          Input  = 1
PIN: 50        MISO_PIN                               Input  = 1
PIN: 51        MOSI_PIN                               Input  = 1
PIN: 52        SCK_PIN                                Input  = 1
PIN: 53        SDSS                                   Output = 0
.              SS_PIN                                 Output = 0
[... 5 lines cut ...]
PIN: 59 (A 5)  TEMP_BED_PIN                           Input  = 1
PIN: 60 (A 6)  TEMP_0_PIN                             Input  = 1
PIN: 61 (A 7)  TEMP_1_PIN                             Input  = 1
PIN: 62 (A 8)  Y_STEP_PIN                             Output = 0
PIN: 63 (A 9)  Y_DIR_PIN                              Output = 0
PIN: 64 (A10)  Z_MIN_PROBE_PIN                        Input  = 1
PIN: 65 (A11)  <unused/unknown>   Analog in =  1023   Input  = 1
PIN: 66        <unused/unknown>                       Input  = 1
PIN: 67        Z_ENABLE_PIN                           Output = 0
PIN: 68        Z_STEP_PIN                             Output = 0
PIN: 69        Z_DIR_PIN                              Analog in =     0
PIN: 70        <unused/unknown>                       Input  = 1
PIN: 71        <unused/unknown>                       Input  = 1
PIN: 72        <unused/unknown>                       Input  = 1
PIN: 73        <unused/unknown>                       Input  = 1
PIN: 74        <unused/unknown>                       Input  = 1
PIN: 75        <unused/unknown>                       Input  = 1
PIN: 76        <unused/unknown>                       Input  = 1
PIN: 77        <unused/unknown>                       Input  = 0
PIN: 78        <unused/unknown>                       Input  = 1
`

with x max shorted it changes to

PIN: 54 (A 0)  <unused/unknown>   Analog in =   942   Input  = 1

boelle, 2018-08-28:
With M43 I T S8 R5 W1000 i can get the extra fan connector make my voltmeter go from 0 to 12 and back

so it its on D8 but strange is that i cant change state with M42 P8 S255

boelle, 2018-08-28:
ahh protected pin

SENDING:M42 P8 S255
Error:Protected Pin
[ERROR] Error:Protected Pin

now where is that set i dont know

boelle, 2018-08-28:
the part cooling fan on D9 have a LEd and it blinks only very shortly and voltmeter is not fast enough

but if i put the voltmeter on the pin from the DUE is changes from 3.3V to 0V and back

D11 which is the nozzle/extruder heater does not react when i try to pulse it

boelle, 2018-08-28:
so i reheated the pin for D2 and now i cant get a response from M502 or M43 I T S2 R5 W1000. M119 and M114 is also non responsive

will take shield of and check pins ( again :-) )

MrAlvin, 2018-08-28:
I have got M106 and M107 working now. For both P0 and P1 (Fan0-D10 and Fan1-D8)
To get it to work, a FAN_1_PIN definition was in need of being added to the pins_RAMPS_17.h file

I also corrected something about Bed pin assignment, that would otherwise have Bed pin as Fan0 pin.

I hope this helps.

boelle, 2018-08-28:
oki... will upload the test sketch to confirm everything still works

then i will download the new pins file and see if that solved it

boelle, 2018-08-28:
M106 P0 works but not M106 P1

M42 P8 S255 does not work either

AnHardt, 2018-08-28:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L7294-L7302
https://github.com/MarlinFirmware/Marlin/blob/ed86550061ab94f10dd1d310c994bd74db582740/Marlin/pins.h#L778-L786

AnHardt, 2018-08-28:
[M42](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L7310-L7342) Have a look at the special handling for the fans. If a **soft** PWM is running at that pin the just written value is  overwritten by that a moment later. Could (should?) be similar for the heaters.

AnHardt, 2018-08-28:
For a test better just use the heater commands.

AnHardt, 2018-08-29:
The heaters are never really **OFF**.
If the environment temperature would be below zero, the heaters would try to heat to zero - provided MINTEMP is not biting before.

boelle, 2018-08-29:
now trying to download again from https://github.com/MrAlvin/Marlin/tree/bgfx-20x-R17-dev and this time make no changes to config to see if something changes

if it works it must be my config that is screwed up somehow, even thou i doubt it since it worked a few hours ago

AnHardt, 2018-08-29:
A naked DUE has no EEPROM - just an emulation (that must be activated?)

boelle, 2018-08-29:
there is an eeprom on the shield/board :-)

boelle, 2018-08-29:
but that is the only thing i will do, put on the shield and upload code as it is on github

boelle, 2018-08-29:
a strange thing, when i upload from platform io i get this

Forcing reset using 1200bps open/close on port COM13
Uploading .pioenvs\DUE\firmware.bin
No device found on COM13
*** [upload] Error 1

but windows can see the board fine

will try a few times

boelle, 2018-08-29:
managed to upload the code

still no response from M502, not from M119 or M114 either

using pronterface and the temp does not update

will try the test program again

boelle, 2018-08-29:
yep, test program confirms everything working

hmm... maybe its best to get some good sleep

boelle, 2018-08-29:
yep time for some sleep to me

managed to flash the copy of marlin linked in first post and for some reason that does not work for me

boelle, 2018-08-29:
output from serial:

Connecting...
start
Printer is now online.
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Aug 29 2018
echo: Free Memory: 86628  PlannerBufferBytes: 1408
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0

what i note is that it loads hard coded default so it has no contact with eeprom on shield

also temp graphs does not update and it does not repond to commands

https://i.imgur.com/cVMkbQO.png

MrAlvin, 2018-08-29:
There is indeed an eeprom on the board, but to use it, it must first be activated.  

If you look at the top of the configuration.h file on https://github.com/MrAlvin/Marlin/tree/bgfx-20x-R17-dev you will see a list of notes and comments about what has been done, and what needs to be done/tested, in order to get the correct settings in Marlin activated, so all things on the board will eventually end up working. 

On the ToDo list/"changes that needs to be done", is: eeprom, SDcard, LCD displays. But more details will also need to be tested. 
And I will say that you are doing a great job testing, and discovering which commands needs attention, and adjustments in various files of the settings in Marlin, so the hardware and the firmware will eventually come together.  So please keep up the good work. 
It might however help to lower your level of frustration, to remember that right now is a phase of prototype and discovery. Okay? 

But still, please keep up the good work, with the testing of gcodes. The testing is greatly appreciated. Okay?

MrAlvin, 2018-08-29:
I have now activated eeprom in configuration.h, and tested that it works. 

I used M503 to see all active settings, then M500 to store in eeprom. 
Then tested by changing M204 to different numbers, M503 to verify. Then M500 to store in eeprom.  
Then reset button, to restart Due. 
Then M503 to see that eeprom values were used at boot. 
Then M502 to reset to hardcoded defaults, and M503 to verify that M204 changed back to hardcoded default. 
Then M501 to read from eeprom, and M503 to verify that M204 had changed. 

And it worked.

boelle, 2018-08-29:
morning

yep i know its still prototype, what made me maybe just a little frustrated is that due to double checking i started to think that my DUE's was bad

but then again why should 2 be bad at the same time

so i will try and relax more :-D

it was very late before i got to sleep so it will be a few hours yet until i can test the latest changes, still feel lile a zombie after only 5 hours of sleep

boelle, 2018-08-29:
just flashed the firmware again, no changes done to me other than editing platformio.ini to say: env_default = DUE

when connecting i get: 

start
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Aug 29 2018
echo: Free Memory: 86620  PlannerBufferBytes: 1408
Error:EEPROM CRC mismatch - (stored) 37377 != 18576 (calculated)!
[ERROR] Error:EEPROM CRC mismatch - (stored) 37377 != 18576 (calculated)!

echo:Hardcoded Default Settings Loaded

boelle, 2018-08-29:
trying to load defaults do not work:

>>>M502
SENDING:M502

will try the other due to see if that changes anything. if not i will try and take off the shield

even without the eeprom it should respond to M502

boelle, 2018-08-29:
with shield on it does not repsond to M502

without it does

its the same on both due's

workheart, 2018-08-29:
Well, with shield ther is eprom and without shield ther is none, that is
reasonable. I would start with some arduino eeprom test\example script test
if its marlin or board related.

boelle, 2018-08-29:
the test program in the first post tests everything on the shield.... including the eeprom

and everything works fine with the test program

boelle, 2018-08-29:
some more sad news

i have been in contact with arduino support and found out that both usb ports work perfectly fine if no shield  is on

if the shield is on then the navtive port stops working (does not show up in device manager) and the programming port shows up but when trying to program it will say: No device found on COM13

its the same for both due's i have only difference is the serial port number

MrAlvin, 2018-08-29:
About eeprom. 

When you first load a firmware, the eeprom is not being changed, notr loaded with anything.
This is why you first do M503, to see all the default settings, as set in the config files.
 
Then you store those settings into eeprom with M501.  
This is the first time that the eeprom contains any information, useful to the Marlin firmware. 

After this you can use M502 to see what is stored in eeprom. 

You need to use the same procedure whenever a newer firmware is storing new variables in eeprom. Not different values, but different variables, or variables in a different order. 
Marlin has a way to signal this change, in storing new variables, or in a different order, in eeprom, but I do not right now know how this method works.   It used to be that the MACHINE_UUID would be used (possibly many years ago) but I imagine that a better method is used today. It could be a CRC check. 

And this could be why, when you first load a firmware, with eeprom enabled, you get a CRC error. Because the eeprom would contain whatever the little test programs, I have made, would leave in the eeprom. The test program basically leaves  ten numbers in sequence in the first ten positions of the eeprom, and this is definitely different from what marlin would store in the first ten bytes of eeprom. 

I hope this sequence of doing eeprom (M503, M501, M502), will result in something useful to you.

MrAlvin, 2018-08-29:
About com ports going wrong. 

I have experienced similar situations in the past, and in windows I have removed them by un-installing the driver associated with that com port. 

One challenge about un-installing a driver (in windows) to a com port that is not working, is that the com port does not show up in the device manager list - until you find a way to have hidden ports shown, that is.  Then the hidden ports will be shown in lighter gray. 

In the windows 10 I have on my machine, there is a menu option for device manager, when right clicking on the start button. 
Otherwise (again on my windows10 version) it is possible to do the search magnifying glass, and type in "device manager", and then a clickable link appears. 

In device manager, chose menu:"view", and select "show hidden devices. 
Then go to "ports (com lpt), and expand the list. This should now show all the com ports that have ever been set up on you computer.  
By right clicking on a com port, it becomes possible to select "Uninstall device". 

Next time the device is plugged in, then a new com port is set up, for that particular serial port/usb device. 

I hope this will help you fix the com port situation. 

Com port troubles can some times arise when too many different environments try to access a particular comport at the same time. 
For 3D printers then,  the Arduino interface, the host software (I user Repetier host), and "platform.io", might all try to access the same com port at the same time, and this can, at times, cause software hick-ups, at the driver level.  And most of the time this can be fixed by removing the com device, and hopefully this will also reset things within each of the environments. 

If you use windows 7, there might be an additional set, to have the device manager show ALL hidden devices. As I recall, an environment variable needs to be set. 

But hopefully your device manager will show the hidden com ports, and hopefully it will allow for new registration of the com ports on your Due's.

boelle, 2018-08-29:
removed all hidden items under ports and under usb

still the same and have sent you some screenshots

could there be a short or something on the board that would make this happen? if so where should i look?

boelle, 2018-08-29:
a note to everyone that is interested, i flashed the test program linked in the first post and everything that the program test works

and i spent the better part of an hour emailing with arduino support to make sure it was not the due or cables that was at fault

dinner time but after it time to get the magnifying glass out and eyeball every little solder joint 

what are the chances that something in the copy of marlin linked in the first post has a stupid little mistake that makes the show stop?

MagoKimbra, 2018-08-31:
Hi, I read only now ...
You can try the basic configuration with the 1.7 ramps with MK4duo, the part of the displays is not yet complete. But a test only serial engines and temperature you can try.
https://marlinkimbra.it/configurator/v4_3_6/firmware_configurator.php
Choice ARDUINO DUE processor and RAMPS 1.7 HFB.

MrAlvin, 2018-09-01:
Hi @MagoKimbra , thank you for making this shield selectable on the MK4Duo configuration list. 
As soon as I get the last few details working with Marlin 2.0, then I will do tests with the MK4Duo.

boelle, 2018-09-01:
Morning everyone..

@MagoKimbra 

i opened the configurator and just selected due and ramps 1.7 and then downloaded, got some compile errors and can see they are spi related:

In file included from sketch\src\platform\HAL_DUE\../../../src/platform/platform.h:41:0,

                 from sketch\src\platform\HAL_DUE\../../../MK4duo.h:112,

                 from sketch\src\platform\HAL_DUE\HAL_spi.cpp:44:

sketch\src\platform\HAL_DUE\HAL_spi.cpp: In static member function 'static void HAL::spiBegin()':

sketch\src\platform\HAL_DUE\../../../src/inc/conditionals_post.h:301:21: error: 'SPI_FULL_SPEED' was not declared in this scope

   #define SPI_SPEED SPI_FULL_SPEED

                     ^

sketch\src\platform\HAL_DUE\../../../src/platform/HAL_DUE/communication.h:104:57: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                (Com::print(val, ## __VA_ARGS__))

                                                         ^

sketch\src\platform\HAL_DUE\../../../src/platform/HAL_DUE/communication.h:123:49: note: in expansion of macro 'SERIAL_MV'

 #define SERIAL_EMV(msg, val, ...)           do{ SERIAL_MV(msg, val, ## __VA_ARGS__); SERIAL_EOL(); }while(0)

                                                 ^

sketch\src\platform\HAL_DUE\HAL_spi.cpp:159:7: note: in expansion of macro 'SERIAL_EMV'

       SERIAL_EMV("SPI_SPEED:", SPI_SPEED);

       ^

sketch\src\platform\HAL_DUE\HAL_spi.cpp:159:32: note: in expansion of macro 'SPI_SPEED'

       SERIAL_EMV("SPI_SPEED:", SPI_SPEED);

                                ^

In file included from sketch\src\platform\HAL_DUE\../../../MK4duo.h:106:0,

                 from sketch\src\platform\HAL_DUE\HAL_spi.cpp:44:

sketch\src\platform\HAL_DUE\../../../src/inc/conditionals_post.h:301:21: error: 'SPI_FULL_SPEED' was not declared in this scope

   #define SPI_SPEED SPI_FULL_SPEED

                     ^

sketch\src\platform\HAL_DUE\HAL_spi.cpp:160:15: note: in expansion of macro 'SPI_SPEED'

       spiInit(SPI_SPEED);

               ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

MagoKimbra, 2018-09-01:
Yes i found this error, one moment...

boelle, 2018-09-01:
no stress, its weekend :-)

any chance you are good with hardware also? i have a strange issue with the ramps 1.7 board i got from @MrAlvin 

i can only flash code with the shield off the due, but his test program shows that everything works

the issue is that with the shield off both usb ports work, with the shield on the native port cant be recognised by windows, and i cant flash via the programming port

MagoKimbra, 2018-09-01:
Ok fixed...
I made the change to SPI_SPEED last night and I did not realize that if the SD is not defined the value of SPI_FULL_SPEED are not defined.

boelle, 2018-09-01:
it can happen to all of us when its late and you should be sleeping

are you good with hardware also?

[23 more replies]

closed 2018-10-22 by boelle