MarlinFirmware/Marlin #11658: AD8495 Calculation Discrepancy with Datasheet

etogburn, 2018-08-27:
### Description
I have a machine that will (eventually) use 5 thermocouples tied to AD8495 amplifiers. I have done extensive research into the chip and the code in Marlin as it exists today. The calculations directly from temperature.cpp are listed below.

`#define TEMP_AD595(RAW)  ((RAW) * 5.0 * 100.0 / 1024.0 / (OVERSAMPLENR) (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET)`
`#define TEMP_AD8495(RAW) ((RAW) * 6.6 * 100.0 / 1024.0 / (OVERSAMPLENR) *(TEMP_SENSOR_AD8495_GAIN) + TEMP_SENSOR_AD8495_OFFSET)`

The calculation for AD595 is perfectly fine according to the datasheet. 10mV/C is specified which is 1/500th of a 5V signal. I can buy into that. I have found that the AD8495 works perfectly if I used the AD595 from older versions (that do not have the AD8495 separated yet) if I use a gain of 2 (5mV/C typical). I can easily change the code to read this:

`#define TEMP_AD8495(RAW) ((RAW) * 10.0 * 100.0 / 1024.0 / (OVERSAMPLENR) * (TEMP_SENSOR_AD8495_GAIN) + TEMP_SENSOR_AD8495_OFFSET)`

This code would match the AD8495 datasheet exactly and allow the gain to typically be set to 1. The offset would need to be used in some cases depending on the reference voltage, but I would consider that fine. 

I did not want to directly modify the code as I did not know any background information into why the 6.6 value was in the formula to begin with. The AD8495 has a 1000C range of readings. It does not make sense to me to have a gain of 10.0/6.6 entered for a typical use of the 8495s. 

Please help verify my logic here. Thanks. Also please forgive me if I've missed something that I should have included with this issue. This is my first time posting in a long time.

etogburn, 2018-08-27:
That's a very good point. It should be able to be derived based on the MCU used on the control electronics. I'd like to see the two formulas for AD595 and AD8495 be based on a consistent reference voltage. This could be as simple as 
#define ADV_REFERENCE_VOLTAGE 
right above where the gains are defined in Configuration_adv.h. I'm unsure if this would be required elsewhere in the code. If it is, it would be cleanest to derive it based on the board.

AnHardt, 2018-08-27:
As far as i know, in all other cases we are measuring resistances, where the reference voltage are shorted out of the equations.

thinkyhead, 2018-09-05:
I suppose we could have a reference voltage in every pins file where it is something other than 5V, and when not defined, assume 5V.

AnHardt, 2018-09-05:
thermistortabe_20 is an other example, depending on 5V reference voltage. 
Here the instrument amplifier needs 5V to set its output proportional to the PT100 resistance. Above a certain temperature it will fry a 3.3V ADC.
Should throw a warning when used at a 3.3V system.

thinkyhead, 2018-09-12:
Any simple way to apply a calculation during runtime to convert the 5V tables to values suitable for 3.3V, or do entirely new tables need to be generated?

AnHardt, 2018-09-12:
There is nothing what we could do but warn.
From the [AD595 Datasheet](https://www.sparkfun.com/datasheets/IC/AD595.pdf)
> Wide Power Supply Range: +5 V to 615 V

-> you can't connect it to 3.3V Vcc

> Thermocouple Type K AD595
Temperature Voltage Output
°C    mV      mV
...
0      0          2.7
...
300 12.207 3022
320 13.039 3227
340 13.874 3434
...
500 20.640 5107
...

-> above ~330°C it will begin to fry a 3.3V ADC and you need a new table.

You could make a custom voltage divider to scale down the AD595s 5V output to 3.3V. In theory you could then use the current table. But in reality you would have to make a custom conversion table for the specific resistor combination you can get.

etogburn, 2018-09-13:
Would it be possible to do the calculation based on a thermistor table? We are still reading an abolitionist input so it is essentially the same thing. The exact software for a thermistor table should work just fine for these as well. Then there could be two separate tables for a 3.3v and 5v reference. I may be over simplifying it, but I do think it could work.

thinkyhead, 2018-09-17:
The thermocouple code would need to be rewritten to accommodate a table. Presumably a table is not needed because they are well-proportioned.

boelle, 2019-02-20:
@etogburn is this one still relavant?

etogburn, 2019-02-20:
@boelle I have my own working copy with a few code changes to separate the gains in the configuration files for AD595 and AD8495. It's not the most elegant, but it does work. I can't imagine this will impact many people, so it is likely ok to leave it as is and close this issue.

boelle, 2019-02-20:
@etogburn only you and @thinkyhead can close :-/

the green button below

closed 2019-02-20 by etogburn