MarlinFirmware/Marlin #11661: Turning part cooling fan on heats extruder nozzle on rigidbotv1 printer. Setting nozzle temperature does not heat nozzle.

Bluelinegecko, 2018-08-27:
### Description

Control pins for Rigidboard compile incorrectly in version 1.1.9

I have 2 virtually identical Rigidbot printers using the original v1 rigidboard. One printer has been running 1.1.8 bugfix for a while and works fine. I decided to upgrade the 2nd to 1.1.9 a few days ago. After doing so, I noticed several of the pins appear to be incorrect. At first I could not get the single extruder to heat, but the thermistor is operating correctly. I assumed it was a configuration error somewhere or a bug. Today I noticed that the part fan for the extruder also does not work, and if I set a speed on the part fan the nozzle starts to increase in temperature even when the nozzle temp is set to zero. (setting a nozzle fan temperature does not appear to start the fan).

Everything else appears to work correctly. I can move axis, heat bed, thermistors work, etc. It looks like the Pins.h settings are getting confused somewhere andare incorrect for at least the fan0 and heater0 pins. I've looked through the code (not a strong point for me) and can't see any real changes to the pins_rigidboard file or anywhere where this could be happening. I also tried replacing only the config files from the example folder and changing just the controller (FGSC) and board (rigidboard) settings and I get the same behavior. 

Flashing back to the previous version of firmware 1.1.4 for that printer works fine so it is not a hardware problem. My 2nd identical rigidbot runs 1.1.8 fine with no issues.
 
### Steps to Reproduce

1. Flash 1.1.9 to rigidbot printer using v1 board
2. Increase nozzle temperture and/or part fan speed
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Nozzle temperature should heat to nozzle setting, Part cooling fan should turn on.

**Actual behavior:** [What actually happens]
Nozzle doesn't heat. When part fan is turned on, fan doesn't spin but nozzle heats up.
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.
[configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/2325653/configfiles.zip)

Bluelinegecko, 2018-08-27:
I think this is related to the fact that the Rigidboard v1 is a dual extruder board regardless of whether both extruders are used or not. Therefore the correct default configuration in pins.h should be for RAMPS_EEF. I think the configuration somewhere is being defined as RAMPS_EFB which would be setting the heater0 pin that should be pin 9 on the rigidbot to pin 8 instead which is the rigidboard Fan0 pin. 

I don't know if the definition of this could have changed somewhere between 1.1.8 and 1.1.9.

Like I said following this code is not my strong point.

Bluelinegecko, 2018-08-28:
Ok. I think I found the real issue and solution. 

I was a bit confused earlier and thought the RAMPS_EFF/RAMPS_EFB variables applied to my set up as the rigidboard is a RAMPs derivative and were overriding the pin assignments, but the configuration doesn't actually implement those lines of code.

The bug is related to the changes made by Thinkyhead on June 7, 2018 (Allow FAN_PIN override, pins cleanup (#10957))

https://github.com/MarlinFirmware/Marlin/commit/582845fd72af67026d116aa2eae67b444c6e62a3#diff-9eebc2382dd9b06f72b413388eb35878

The problem is that the Fan0 pin is not being defined first in the pins_RIGIDBOARD.h file.

The RAMPS_D10_PIN variable is first defined as pin 9 in the pins_RIGIDBOARD.h on line 39.
Then on line 42 pins_RAMPS.h is included.
Since FAN_PIN has not yet been defined, no changes are made to the heater/fan pins assignments until line 266 of pins_RAMPS.h.
Since FAN_PIN hasn't been defined yet, the #ifndef FAN_PIN section of code is triggered. Since the rigidboard isn't defined as any of the RAMPS variants, the #else statement defines FAN_PIN as pin 9 (the setting already defined as RAMPS_D10_PIN and is used for heater0)
Then when pins_RAMPS.h finishes and it returns to process the remainder of the pins_RIGIDBOARD.h code, when it gets to line #88, since FAN_PIN has already been defined in pins.RAMPS.h if doesn't execute line #89 which would return FAN_PIN to the correct setting of pin 8.

So effectively, the heater for extruder0 is defined as pin9 first, then pin 9 is later defined to FAN. After compiling, hardware signals to turn on the fan actually start heating the extruder and not turning the fan, and hardware signals to heat the extruder are ignored.

I confirmed this and fixed it in my configuration by simply moving lines #88-#90 in Pins_rigidboard.h above line #42 so that the FAN_PIN is defined prior to calling pins_RAMPS.h. I complied the firmware, uploaded it and everything appears to be functioning correctly. A test print is running now.

Modifying the code so that it defines the FAN_PIN before including pins_ramps.h is probably the best, solution, as just reverting the code to undefine/define the FAN_PIN as it originally appeared in pins_RIGIDBOARD.h would not prevent the fan pin assignment from being assigned to the same pin as the heater and may still cause a conflict even if the fan pin is correctly assigned later.

I didn't submit a fork, as I'll leave it to Thinkyhead or one of the other maintainers to decide how to implement the actual bug fix, as it looks like a lot of work went into the cleanup and trying to make the numerous pins.h files uniform especially where the FAN_PIN variable is concerned. None the less, this is a show stopping bug for anyone trying 1.1.9 on a system with a rigidboard or rigidboardv2 which may affect many users trying to upgrade Marlin. 

I've included my modified pins_rigidboard.h [pins_rigidboard.h.zip](https://github.com/MarlinFirmware/Marlin/files/2329866/pins_rigidboard.h.zip) file though if it helps anyone.

Bluelinegecko, 2018-08-30:
In looking at the bugfix 2.0 code I'm confident the bug exists there too as the pins_* files  are the same as the ones used in 1.1.9.

thinkyhead, 2018-09-09:
TL;DR — What value should the `FAN_PIN` be assigned in `pins_RIGIDBOARD.h`?

edwilliams16, 2018-09-09:
I think pin 8.

M43 for working 1.1.x:
[image]

M43 for 1.1.9:
[image]

Bluelinegecko, 2018-09-12:
> There are two ways this might be resolved:
> 
> 1. Revert `pins_RIGIDBOARD.h` to put this back as it was…
>    ```c++
>    #undef FAN_PIN
>    #define FAN_PIN             8 // Same as RAMPS_13_EEF
>    ```
> 2. Or, put the following line at the top above the include of `pins_RAMPS.h`…
>    ```c++
>    #define RAMPS_D9_PIN 8
>    ```
> 
> Please try both of these and see which one produces the best result, from the standpoint of being able to pick a layout like we can with RAMPS, such as EEF, EFF, EFB, etc. It seems to me that option number 2 is the better one, since it prevents the altered "D10" MOSFET from colliding with the unaltered "D9" MOSFET that RAMPS ends up using for the middle connector, while still allowing the plug order to be customized.

I agree with you that the 2nd choice is probably the safest and why I configured mine that way by  moving the #define RAMPS_D9_PIN 8 line before the include. It looked like you had put a lot of thought into unifying the format of the pins files of all the RAMPS variants, so I thought I'd leave it up to you and the rest of the Marlin crew to figure out how to best implement the bug fix and keep everything as uniform as possible.

Line #89 of pins_RIGIDBOARD.h has the comment: "// Same as RAMPS_13_EEF". It is entirely possible that just defining the board as the EEF variant somewhere before including 'pins_RAMPS.h' would also work, as the ifndef FAN_PIN statement in "pins_RAMPs.h" would then correctly set the fan pin to D8, instead of having it set to D9 by the 'else' statement. This may cause conflicts somewhere else though with pin definitions but I suspect that might be how it was originally set up. I also suspect that in dual extruder set-ups, line #208 of pins_RAMPs.h might correctly define the board as RAMPS_EEF but I havent checked to see if the bug in the rigidboard file affects my dual extruder version or if this would correct the problem yet.

All that being said, I do agree with Edwilliams16's comments about removing the RAMPS references entirely and just assigning all the pins in a pure pins_RIGIDBOARD.h file. The main reason being that the Rigidboards are fully socketed boards using JST or ribbon cable connectors on all connections. The only common user change is to switch the extruders as Edwillaims16 wrote. They really don't need or have the flexibility of a simple RAMPS board and it would likely make configuration easier to just define all the pins in one file.

There are very few socketed pins that are unused on the rigidboard. By default they only use minimum endstops, but the socket connection are there for people who have wanted/needed to add maximum endstops as well. If those aren't currently defined it makes sense to add those to the rigidboard file as well. There has been some discussion in the Rigidbot Google+ group of addressing the Fan1 pin for individual control of the 2nd extruder's part cooling fan on dual extruder configurations too. Finally, while we are at it, with the addition of all the increased RGB LED functionality that has been added in the last few revisions of Marlin, it might be worth it to assign the correct pins in the rigidboard pin file for the factory RGB strip connector to make future implementations easy. I haven't added a strip to mine, but anyone who has and might know the correct pin definitions for the JST connector on the board might be able to help us include that as well. I can see if we get any more feedback in the rigidbot forums on these unused connections and the correct pin definitions if we are open to including those definitions to whatever the final bugfix looks like.

thinkyhead, 2018-09-12:
The main reason we try to keep pins files limited to the differences –when possible– is to make it easier to see the variant w/r/t the board design to which they are most closely related. At the present time the RigidBoard maintains the EEF order, and there's no user-facing facility to rearrange the ordering. But there are defines that can be added to change the order, basically there for the experimentalists.

ardyesp, 2018-09-14:
> There are two ways this might be resolved:
>
>     1. Revert `pins_RIGIDBOARD.h` to put this back as it was…
>        ```c++
>        #undef FAN_PIN
>        #define FAN_PIN             8 // Same as [...]

I am using Option 1 and it works. Option 2 is probably a better fix.

edwilliams16, 2018-09-14:
I changed pins_RIGIDBOT.h to read

#define FAN_PIN           8  // EXTRUDER 1

#define FAN1_PIN          11  // EXTRUDER 2
#define HEATER_1_PIN       12    
#include "pins_RAMPS.h"

#define HEATER_0_PIN       9  

which works to fix the other pin errors.

closed 2018-11-02 by thinkyhead