MarlinFirmware/Marlin #11662: [1.1.9] Printbed bigger than printable Area raises an Error

DurenBlake, 2018-08-27:
Hi,
im trying to compile Marlin for my selfbuild 3D-Printer.

I designed the printer to have an bigger buildplate than the actual area it can print on,
therefore the nozzle will never leave the printbed.

I´m currently running on Marlin 1.1.6.

Trying to update to 1.1.9 raises the following Error:

> SanityCheck.h:378: error: static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.
> 
>  static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,

My configuration for the printbed and [X/Y]_[MIN/MAX]_POS is as follows:

> // The size of the print bed
> #define X_BED_SIZE 214
> #define Y_BED_SIZE 214
> 
> // Travel limits (mm) after homing, corresponding to endstop positions.
> #define X_MIN_POS 0
> #define Y_MIN_POS 0
> #define Z_MIN_POS 0
> #define X_MAX_POS 203
> #define Y_MAX_POS 193
> #define Z_MAX_POS 227

The error does makes sense, because the nozzle can never leave the printbed and therefore can never contain the whole printbed. But i designed it to be this way.

My question now is:
Can i just comment this Check out and be happy or
does some part of Marlin require the printer to contain the whole printbed withing its movement range and im stuck with 1.1.6?

p3p, 2018-08-27:
The bed size is how you define the printable area, your configuration is contradicting that by having a bigger printable area than your printer has access to, have you got what these mean inverted?

DurenBlake, 2018-08-27:
They are not inverted. 
My Bed is 214x214 in size.
And my usable area is 203x193.

But i guess you already answered my problem.
If i set the bed_size to what my print area is (203x193) it will work.

I only hope Marlin doesn`t run into trouble, because my actual printbed is larger.

Maybe i got the meaning of "size of the print bed" wrong by typing in the number for the actual hardware rather than the area i have access to with the nozzle .... maybe i starred to long at the screen for today :D

closed 2018-08-28 by DurenBlake