MarlinFirmware/Marlin #11664: Stepper puls timing

maukcc, 2018-08-28:
Does anyone know the time between a change on the DIR pin to the next pulse/step?
Can this be set somewhere?

GMagician, 2018-08-28:
With new version it is automatically set when you define appropriate driver type in configuration.h (X_DRIVER_TYPE, Y_DRIVER_TYPE, etc).
You may still override values in configuration_adv.h (MINIMUM_STEPPER_DIR_DELAY)

maukcc, 2018-08-28:
Ooh, thats new. I like it !
Where are all the settings for the separate drivers stored?

GMagician, 2018-08-28:
drivers.h file has default values. For 2.0 is located in core subfolder

maukcc, 2018-08-28:
Thanks for the feedback.
#define A4988 0x001 refers to somewhere else I am guessing?

GMagician, 2018-08-28:
No, is just a number assigned arbitrarily to identify it from other driver but any number may have been used. Important is that every driver has an univocal value

closed 2018-08-30 by thinkyhead