MarlinFirmware/Marlin #11667: CR-10 upg to Marlin 1.1.8 fan issue (always on)

piotrrkozera, 2018-08-28:
Hi,
I upgrade my CR-10 software to Marlin 1.1.8. I want add to my printer BLTouch option. Everything work fine but my cooling fan is always on now. M107, M106 S0 and M106 Px S0 don't work and I cant turn off cooling fan.
This is so annoying because I used this fan to control laser.
I attached my Configuration.h, Configuration_adv.h and pins_ for Melzi_Creality, SANGUINOLOLU_11 and SANGUINOLOLU_12 if needed.
Please for help :)
Best regards,
Piotr
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2327544/Marlin.zip)

thinkyhead, 2018-09-05:
Marlin 1.1.8 is obsolete. Try Marlin 1.1.9 and see how it fares.

thinkyhead, 2018-09-05:
Marlin 1.1.8 is obsolete. Try Marlin 1.1.9 and see if it's any better.

boelle, 2019-02-20:
@piotrrkozera have you tried latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead