MarlinFirmware/Marlin #11668: Error compiling with Extruder Runout Prevention enabled - 1.1.9

welo78, 2018-08-28:
### Description

Error compiling for board Mega 2560 when enabling Extruder Runout Prevention

### Steps to Reproduce

1. Download and unzip Marlin 1.1.9
2. Configure Marlin with basic settings like drivers, steps, discount reprap graphic display
3. Configure serial buffer size (expand)
4. Turn on some other advanced features (whole configuration files in attachment)
5. Enable extruder runout prevent
6. Get error compiling
7. Disable extruder runout prevent
8. Succesfully compiled

**Expected behavior:** Compile succesfully

**Actual behavior:** Compiling fails at end with Error

#### Additional Information

* Included zip file
* Picture of error

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/2328112/Configuration.Files.zip)

[image]

boelle, 2019-02-20:
@welo78 did you try as thinky suggested?

welo78, 2019-02-21:
The problem was in te arduino IDE. I dont know why but I get these segmentation faults from time to time. 
Yesterday I used a vanilla Marlin 1.1.9 with only:
- drivers selected in configuration.h (compiled here succesfully)
- pins changed in ramps.h pinout (compiled here succesfully)
- tmc debug and tmc monitoring enabled in configuration_adv.h (segmentation fault)

Nothing else was configured I think, since I only wanted to test if my TMC2130 SPI drivers are working. When I tried to compile after selecting drivers and changing pins it compiled succesfully. After uncommenting TMC debug and monitoring, it got a segmentation fault. **I restarted computer and it compiled succesfully.**

boelle, 2019-02-21:
you need to use bugfix 2.0

welo78, 2019-02-21:
Isn't bugfix 2.0 only for 32 bit boards ? I am using a Ramps 1.4 :)

boelle, 2019-02-21:
no, its both for 8 bit and 32 bit

boelle, 2019-02-21:
also use arduino ide 1.9.0

boelle, 2019-02-21:
https://www.arduino.cc/download_handler.php?f=/ide_beta/arduino-1.9.0-beta-windows.zip

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead