MarlinFirmware/Marlin #11670: cant compile

littlepedro78, 2018-08-29:
Arduino error: pasting "/* (bed)*/" and "_RPORT" does not give a valid preprocessing token
any help

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

cjsoong, 2018-08-29:
@littlepedro78  Do not use Arduino IDE V1.86, change back to V1.85.

littlepedro78, 2018-08-29:
thanks solved

closed 2018-08-30 by thinkyhead