MarlinFirmware/Marlin #11671: bltouch not work help please

marco1703, 2018-08-29:
[Marlin-1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/2333354/Marlin-1.1.9.zip)

### Description

hello guys please help me I have a problem with Bltouch .. you turn on the connections are right and 'connected to a ramps 1.4 gen .. seems to work but in truth do not level the printing floor power please check my fw .. I leave it attached thank you
### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

GMagician, 2018-08-29:
try removing `//` before `ENDSTOP_INTERRUPTS_FEATURE`, ramps support this

you should also remove `//` before `?_DRIVER_TYPE` (for installed steppers) and replace `A4988` with your driver type name (but not related to bltouch issue)

Why have you defined SERVO0_PIN in configuration?
You may chose between BOARD_RAMPS_14_EFB and BOARD_RAMPS_13_EFB

marco1703, 2018-08-30:
hello thanks for the answer I have tmc2100 as a driver .. servo pin 11 .. what do you think?

GMagician, 2018-08-30:
?_DRIVER_TYPE must be TMC2100.
Since you are using ramps 1.4 you should not change servo pin. it's automatically assigned and you need to connect bltouch to D11 (3 pin near power connector). Remember jumper to give 5V to it.

Does it stow and deploy when commanded by LCD?

marco1703, 2018-08-30:
I did not give extra power because I saw that it works .. the problem is that even if it does not work, the plan does not save the printing plan compensation data

GMagician, 2018-08-30:
Wait a moment, your issue is bltouch that doesn't works or UBL data not stored?

marco1703, 2018-08-30:
UBL data not stored my problem

GMagician, 2018-08-30:
Oh, that's another story despite your issue title.

read notes at:

> https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

To store G29 S1.
To use after G29 L1

marco1703, 2018-08-30:
you can tell me the gcode procedure to use step by step

GMagician, 2018-08-30:
Read link I wrote above to acquire data
M420 S1 should be used as first commands in your gcode

marco1703, 2018-08-30:
but i can try the bltousch before the the firtst gcode?

GMagician, 2018-08-30:
What do you means with:

> but i can try the bltousch before the the firtst gcode

You already stated that bltouch is working...

marco1703, 2018-08-30:
the bl, have color red, and if i send g28 / g29 and after this send 420 V, send me this message 
bed lev ok 
feed height off

GMagician, 2018-08-30:
'Fade Height off' means that you dont have active such function (it reduce compensation till removing it after some mm in Z)
'bed lev ok' if really is 'Bed Leveling on' it means that you have activated UBL compensation
bl red is normal when it is stowed

marco1703, 2018-08-30:
but i have #define  fade height ....you can see my fw

GMagician, 2018-08-30:
> but i have #define fade height ....you can see my fw

Have you changed it by LCD? sure you have done M500 after?

marco1703, 2018-08-30:
now i try, pls

marco1703, 2018-08-30:
thks for your support ;)

marco1703, 2018-08-30:
can you try a write the fw for me, have ramps 1.4 gen, ehit lcd 12264,  tmc2100, and bltouch, can try pls?

GMagician, 2018-08-30:
I just see now you are using AUTO_BED_LEVELING_UBL not AUTO_BED_LEVELING_UBL then some details given before may be not correct.

You configuration.h is pretty good. This is what I should do with no more details than the ones given
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2336180/Configuration.zip)

marco1703, 2018-08-30:
special thks , i can try later,i send news

ebraiman, 2018-10-14:
If this helps I made a video on how to install and test.  https://youtu.be/d0akA_5mm-8

closed 2018-10-26 by thinkyhead

padcom, 2019-02-26:
There is one thing I have found out Today about BLTouch (and clones) that is very useful (I think). The point is that when you connect the BLTouch sensor it is relevant that the pins are properly connected and not the other way around. I have had mine turned around and it just didn't work...

marco1703, 2019-02-26:
I should mount on a mks 32 bit ramps with tmc2208 the bltouch .. with marlin 2.0 you know how to configure it? regards

Autofficina Migliori

Questa mail è inviata unicamente alle persone o enti ai quali è indirizzata. Essa può contenere informazioni la cui riservatezza è tutelata dalle leggi vigenti. Sono vietati la riproduzione e qualsiasi uso di questa mail e/o delle informazioni in essa contenute in mancanza di autorizzazione del destinatario. Se avete ricevuto questa mail per errore, vogliate cortesemente contattarci quanto prima per telefono o fax cancellate questa e-mail dal vostro computer.