MarlinFirmware/Marlin #11673: Marlin 1.1.9 , Dual Z endstop didn't work

ANYCUBIC-3D, 2018-08-30:
Hello, the mainly config like below:
#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  #define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _ZMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif

#define USE_ZMAX_PLUG

 #define Z_MAX_PIN        43
Without auto leveling, now dual Z motors are stoped by ZMIN  endstop when homing, dose I mis-configed something?

ANYCUBIC-3D, 2018-08-30:
@GMagician  thanks for your answer. I had tested M119, when I trigger 'ZMAX' endstop manually, it shows 'z2_min: TRIGGERED', otherwise it shows 'z2_min: open'.
Also yeah, the two underscore of ZMAX was disappeared when I publish this comment.

GMagician, 2018-08-30:
Do tou have ENDSTOP_INTERRUPTS_FEATURE or not?
What about ENDSTOP_NOISE_FILTER? sure you don't have EMI issues?

ANYCUBIC-3D, 2018-08-30:
I didn't enable ENDSTOP_INTERRUPTS_FEATURE and ENDSTOP_NOISE_FILTER , The same config to marlin 1.1.0 run on the same machine, it works fine. 
Should I must be enable ENDSTOP_INTERRUPTS_FEATURE?

GMagician, 2018-08-30:
Yes if you can (if your HW support interrupt on endstop inputs)

ANYCUBIC-3D, 2018-08-30:
Sorry.  Could you test Dual Z endstop with disable ENDSTOP_INTERRUPTS_FEATURE? my Y_MIN_PIN had used for UART3,  so I can't open interrupt endstop inputs.

ANYCUBIC-3D, 2018-08-30:
OK, thanks , I opened ENDSTOP_NOISE_FILTER, the result is same, so maybe not EMI issues.

GMagician, 2018-08-30:
Sure that when you manually trigger z1 endstop z2 will stay open?

ANYCUBIC-3D, 2018-08-30:
Yeah, I am very sure, I had use marlin old version two years

GMagician, 2018-08-30:
That's weird... everything seems ok. Can you attach full configuration.h and configuration_adv.h?

Really I'm using 2.0 but it should be the same logic.

ANYCUBIC-3D, 2018-08-31:
Hello there, here are full config:

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
[... 3484 lines cut ...]
  #define MAX7219_DEBUG_PLANNER_HEAD  3  // Show the planner queue head position on this and the next LED matrix row
  #define MAX7219_DEBUG_PLANNER_TAIL  5  // Show the planner queue tail position on this and the next LED matrix row

  #define MAX7219_DEBUG_PLANNER_QUEUE 0  // Show the current planner queue depth on this and the next LED matrix row
                                         // If you experience stuttering, reboots, etc. this option can reveal how
                                         // tweaks made to the configuration are affecting the printer in real-time.
#endif

/**
 * NanoDLP Sync support
 *
 * Add support for Synchronized Z moves when using with NanoDLP. G0/G1 axis moves will output "Z_move_comp"
 * string to enable synchronization with DLP projector exposure. This change will allow to use
 * [[WaitForDoneMessage]] instead of populating your gcode with M400 commands
 */
//#define NANODLP_Z_SYNC
#if ENABLED(NANODLP_Z_SYNC)
  //#define NANODLP_ALL_AXIS  // Enables "Z_move_comp" output on any axis move.
                              // Default behaviour is limited to Z axis only.
#endif

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

#endif // CONFIGURATION_ADV_H

GMagician, 2018-08-31:
I checked your config and it seems ok, nothing related to your issue.

Something that is not Marlin standard:
  `#define OutageTest`
  `#define TFTmodel`
Something strange:
  `#define USE_XMAX_PLUG`
  `#define E1_DRIVER_TYPE A4988`
  `#define Z2_USE_ENDSTOP ZMAX`
  `#define HOMING_FEEDRATE_XY (4060)`
  '#define HOMING_FEEDRATE_Z (660)'

Today I'll be able to fix my printer and I'll try configuration without interrupts, just to see it this is the cause of your trouble

ANYCUBIC-3D, 2018-08-31:
Thanks again.
//#define OutageTest
//#define TFTmodel 
write by myself, now I disabled.
//#define USE_XMAX_PLUG 
about '#define E1_DRIVER_TYPE A4988, I used E1 motor  as another Z motor. and E1 driver type is A4988
about #define HOMING_FEEDRATE_XY (4060)
'#define HOMING_FEEDRATE_Z (660)' there is 'myltiply' in my project, but disappered when I comment to here. 

Now another Strange thing, when I home all  axis, X Y works well(trigger endstop two times), but Z axis trigger one time and stop there, about 1 minutes(or maybe later) later,it raise and trigger one time again by itself.

GMagician, 2018-08-31:
> about '#define E1_DRIVER_TYPE A4988, I used E1 motor as another Z motor. and E1 driver type is A4988

you have to `#define Z2_DRIVER_TYPE A4988` for second Z axis

> trigger one time and stop there, about 1 minutes

this seems a issue already reported by other people.

To help you may compile with DEBUG_LEVELING_FEATURE enabled and execute home. Maybe can help to signal in echo messages where pause is

thinkyhead, 2018-08-31:
@ANYCUBIC-3D — Please read this before posting any code or other text with characters like asterisk, underscore, etc.: https://guides.github.com/features/mastering-markdown/

ANYCUBIC-3D, 2018-09-01:
Thanks to all.

I had compile with DEBUG_LEVELING_FEATURE,  NOTHING feedback, and didn't response G/M code until Z axis move again.

GMagician, 2018-09-01:
@ANYCUBIC-3D after compiling with DEBUG_LEVELING_FEAUTURE, and before doing home, you have to execute M111 S255

ANYCUBIC-3D, 2018-09-01:
This is all feedback:
M111 S255

echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION
ok
G28
echo:G28
Error:Printer halted. kill() called!(this message feedbacked at the second time of Z  endstop trigger)

How can I fix this issue?

GMagician, 2018-09-01:
Uhm... this was message with Z_DUAL_ENDSTOPS_ADJUSTMENT = 0 was fixed time ago.. Please try with 0.001. You you still have it maybe we are facing some regression

ANYCUBIC-3D, 2018-09-03:
I had set `Z_DUAL_ENDSTOPS_ADJUSTMENT` =0.001;
I had enable  `DEBUG_LEVELING_FEATURE` and `M114_DETAIL`;
then re-flash and reboot system.

Below is the log:

M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
ok
G28
echo:G28
>>> G28
Machine Type: Cartesian
Probe: NONE
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 2.00
>>> do_blocking_move_to(0.00, 0.00, 2.00)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -630.00, [40.00])
  current_position=(0.00, 0.00, 2.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, [40.00])
  current_position=(0.00, 0.00, 2.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 10.00)
  current_position=(0.00, 0.00, 2.00) : sync_plan_position
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -10.00
 soft_endstop_max = 410.00
> home_offset[X] = 0.00
  current_position=(-10.00, 0.00, 2.00) : 
<<< set_axis_is_at_home(X)
  current_position=(-10.00, 0.00, 2.00) : sync_plan_position
  current_position=(-10.00, 0.00, 2.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -615.00, [40.00])
  current_position=(-10.00, 0.00, 2.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, [40.00])
  current_position=(-10.00, 0.00, 2.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 10.00)
  current_position=(-10.00, 0.00, 2.00) : sync_plan_position
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 410.00
> home_offset[Y] = 0.00
  current_position=(-10.00, 0.00, 2.00) : 
<<< set_axis_is_at_home(Y)
  current_position=(-10.00, 0.00, 2.00) : sync_plan_position
  current_position=(-10.00, 0.00, 2.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, -682.50, [6.00])
  current_position=(-10.00, 0.00, 0.00) : sync_plan_position
G29
**About 1 minute later(This item is not log)**
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 8.00, [6.00])
  current_position=(-10.00, 0.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -16.00, 1.50)
  current_position=(-10.00, 0.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
>>> do_homing_move(Z, 0.00, [6.00])
  current_position=(-10.00, 0.00, 0.00) : sync_plan_position
Error:Printer halted. kill() called!

**Reboot machine manually (This item is not log)**
M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED
z2_min: TRIGGERED
ok
M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z2_min: open
ok

ANYCUBIC-3D, 2018-09-05:
@thinkyhead  Any news?

thinkyhead, 2018-09-05:
> Any news?

I fixed your formatting for readability. Please read [this page](https://guides.github.com/features/mastering-markdown/) to learn how to use Markdown formatting.

As for the logs, they only reveal that the Z endstop is not showing as triggered when homing is complete. It could be a wiring issue. It's impossible to tell without more hands-on diagnostics.

As a temporary workaround, you can comment out all occurrences in Marlin of…
```cpp
endstops.validate_homing_move();
```

closed 2018-09-11 by ANYCUBIC-3D

fikretduru, 2020-04-18:
only work NC

thinkyhead, 2020-04-20:
> only work NC

```cpp
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.
```