MarlinFirmware/Marlin #11678: help wanted failing to start printing at 0.2

khomouda, 2018-08-30:
Hello all, I have a problem in printing, this is the first time to try to print with the a coreXY printer called v-king by Roy @ http://v-king.no, that I just built. When the g-code file Homes, then start to print and runs <<G1 Z0.200 F1002>> to go to the printing position, it makes sound at the worm gear , the controller changes z to 0.2 and starts printing although the actual position is still 11
 I tried changing the speed of z movement to no avail, jogging, homing, and even G0 Z0 works fine, just the code in simplify3d output file

; G-Code generated by Simplify3D(R) Version 4.0.1
; Aug 30, 2018 at 9:27:24 PM
; Settings Summary
;   processName,Left
;   applyToModels,100x100mm
;   profileName,HICTop (modified)
;   profileVersion,2018-07-19 04:21:55
;   baseProfile,Default (modified)
;   printMaterial,ABS
;   printQuality,Medium
;   printExtruders,
;   extruderName,Left Extruder
;   extruderToolheadNumber,0
;   extruderDiameter,0.4
;   extruderAutoWidth,0
;   extruderWidth,0.4
;   extrusionMultiplier,1
;   extruderUseRetract,1
;   extruderRetractionDistance,1
;   extruderExtraRestartDistance,0
;   extruderRetractionZLift,0
;   extruderRetractionSpeed,1800
;   extruderUseCoasting,0
;   extruderCoastingDistance,0.2
;   extruderUseWipe,0
;   extruderWipeDistance,5
;   primaryExtruder,0
;   layerHeight,0.2
;   topSolidLayers,3
;   bottomSolidLayers,3
;   perimeterOutlines,3
;   printPerimetersInsideOut,1
;   startPointOption,2
;   startPointOriginX,0
;   startPointOriginY,0
;   sequentialIslands,0
;   spiralVaseMode,0
;   firstLayerHeightPercentage,100
;   firstLayerWidthPercentage,100
;   firstLayerUnderspeed,0.8
;   useRaft,0
;   raftExtruder,0
;   raftTopLayers,2
;   raftBaseLayers,2
;   raftOffset,6
;   raftSeparationDistance,0.45
;   raftTopInfill,20
;   aboveRaftSpeedMultiplier,0.3
;   useSkirt,0
;   skirtExtruder,0
;   skirtLayers,2
;   skirtOutlines,5
;   skirtOffset,0
;   usePrimePillar,0
;   primePillarExtruder,999
;   primePillarWidth,12
;   primePillarLocation,7
;   primePillarSpeedMultiplier,1
;   useOozeShield,0
;   oozeShieldExtruder,999
;   oozeShieldOffset,2
;   oozeShieldOutlines,1
;   oozeShieldSidewallShape,1
;   oozeShieldSidewallAngle,30
;   oozeShieldSpeedMultiplier,1
;   infillExtruder,0
;   internalInfillPattern,Rectilinear
;   externalInfillPattern,Rectilinear
;   infillPercentage,30
;   outlineOverlapPercentage,15
;   infillExtrusionWidthPercentage,100
;   minInfillLength,5
;   infillLayerInterval,1
;   internalInfillAngles,45,-45
;   overlapInternalInfillAngles,0
;   externalInfillAngles,45,-45
;   generateSupport,1
;   supportExtruder,0
;   supportInfillPercentage,10
;   supportExtraInflation,0
;   supportBaseLayers,2
;   denseSupportExtruder,0
;   denseSupportLayers,0
;   denseSupportInfillPercentage,40
;   supportLayerInterval,1
;   supportHorizontalPartOffset,0.3
;   supportUpperSeparationLayers,1
;   supportLowerSeparationLayers,1
;   supportType,0
;   supportGridSpacing,4
;   maxOverhangAngle,45
;   supportAngles,45,-45
;   temperatureName,Bed,Left Extruder
;   temperatureNumber,0,0
;   temperatureSetpointCount,1,1
;   temperatureSetpointLayers,1,1
;   temperatureSetpointTemperatures,100,250
;   temperatureStabilizeAtStartup,1,1
;   temperatureHeatedBed,1,0
;   temperatureRelayBetweenLayers,0,0
;   temperatureRelayBetweenLoops,0,0
;   fanLayers,1
;   fanSpeeds,0
;   blipFanToFullPower,0
;   adjustSpeedForCooling,1
;   minSpeedLayerTime,15
;   minCoolingSpeedSlowdown,20
;   increaseFanForCooling,0
;   minFanLayerTime,45
;   maxCoolingFanSpeed,100
;   increaseFanForBridging,0
;   bridgingFanSpeed,100
;   use5D,1
;   relativeEdistances,0
;   allowEaxisZeroing,1
;   independentExtruderAxes,0
;   includeM10123,0
;   stickySupport,1
;   applyToolheadOffsets,0
;   gcodeXoffset,0
;   gcodeYoffset,0
;   gcodeZoffset,0
;   overrideMachineDefinition,0
;   machineTypeOverride,0
;   strokeXoverride,200
;   strokeYoverride,200
;   strokeZoverride,200
;   originOffsetXoverride,0
;   originOffsetYoverride,0
;   originOffsetZoverride,0
;   homeXdirOverride,-1
;   homeYdirOverride,-1
;   homeZdirOverride,-1
;   flipXoverride,1
;   flipYoverride,-1
;   flipZoverride,1
;   toolheadOffsets,0,0|0,0|0,0|0,0|0,0|0,0
;   overrideFirmwareConfiguration,0
;   firmwareTypeOverride,RepRap (Marlin/Repetier/Sprinter)
;   GPXconfigOverride,r2
;   baudRateOverride,115200
;   overridePrinterModels,1
;   printerModelsOverride,HicTop_3DP12_-_Bed.stl
;   startingGcode,M280 P0 S160 ; BLTouch alarm release,G4 P100 ; delay for BLTouch,G28 ; home,M420 S1,
;   layerChangeGcode,
;   retractionGcode,
;   toolChangeGcode,
;   endingGcode,M104 T0 S0 ; turn off left extruder,M104 T1 S0 ; turn off Right extruder,M140 S0 ; turn off bed,G91 ; relative mode,G1 Z15 F1500 ; raise nozzle 15mm ,G90 ; absolute mode,G1 X30 Y240 F3000,M84 ; disable motors
;   exportFileFormat,gcode
;   celebration,0
;   celebrationSong,Random Song
;   postProcessing,,
;   defaultSpeed,3600
;   outlineUnderspeed,0.3
;   solidInfillUnderspeed,0.8
;   supportUnderspeed,0.8
;   rapidXYspeed,4800
;   rapidZspeed,1002
;   minBridgingArea,50
;   bridgingExtraInflation,0
;   bridgingExtrusionMultiplier,1
;   bridgingSpeedMultiplier,1
;   useFixedBridgingAngle,0
;   fixedBridgingAngle,0
;   applyBridgingToPerimeters,0
;   filamentDiameters,1.75|1.75|1.75|1.75|1.75|1.75
;   filamentPricesPerKg,46|46|46|46|46|46
;   filamentDensities,1.25|1.25|1.25|1.25|1.25|1.25
;   useMinPrintHeight,0
;   minPrintHeight,0
;   useMaxPrintHeight,0
;   maxPrintHeight,0
;   useDiaphragm,0
;   diaphragmLayerInterval,20
;   robustSlicing,1
;   mergeAllIntoSolid,0
;   onlyRetractWhenCrossingOutline,1
;   retractBetweenLayers,1
;   useRetractionMinTravel,0
;   retractionMinTravel,3
;   retractWhileWiping,0
;   onlyWipeOutlines,1
;   avoidCrossingOutline,0
;   maxMovementDetourFactor,3
;   toolChangeRetractionDistance,12
;   toolChangeExtraRestartDistance,-0.5
;   toolChangeRetractionSpeed,600
;   externalThinWallType,0
;   internalThinWallType,2
;   thinWallAllowedOverlapPercentage,10
;   singleExtrusionMinLength,1
;   singleExtrusionMinPrintingWidthPercentage,50
;   singleExtrusionMaxPrintingWidthPercentage,200
;   singleExtrusionEndpointExtension,0.2
;   horizontalSizeCompensation,0
G90
M82
M106 S0
M140 S100
M190 S100
M104 S250 T0
M109 S250 T0
M280 P0 S160 ; BLTouch alarm release
G4 P100 ; delay for BLTouch
G28 ; home
M420 S1
G92 E0
G1 E-1.0000 F1800
G1 Z0.200 F1002
; process Left
; layer 1, Z = 0.200
T0
; tool H0.200 W0.400
; outer perimeter
G1 X60.200 Y85.200 F4800
G1 E0.0000 F540
G92 E0
G1 X159.800 Y85.200 E3.3127 F864
G1 X159.800 Y184.800 E6.6254
G1 X60.200 Y184.800 E9.9381
G1 X60.200 Y85.200 E13.2508
G1 X60.600 Y85.600 F4800
G92 E0
G1 X60.600 Y184.400 E3.2861 F864
G1 X159.400 Y184.400 E6.5722
G1 X159.400 Y85.600 E9.8583
G1 X60.600 Y85.600 E13.1444
G92 E0
G1 E-1.0000 F1800
; layer 2, Z = 0.400
G1 X60.200 Y85.200 F4800
G1 Z0.400 F1002
G1 E0.0000 F1800
G92 E0
G1 X159.800 Y85.200 E3.3127 F1080
G1 X159.800 Y184.800 E6.6254
G1 X60.200 Y184.800 E9.9381
G1 X60.200 Y85.200 E13.2508
G1 X60.600 Y85.600 F4800
G92 E0
G1 X60.600 Y184.400 E3.2861 F1080
G1 X159.400 Y184.400 E6.5722
G1 X159.400 Y85.600 E9.8583
G1 X60.600 Y85.600 E13.1444
G92 E0
G1 E-1.0000 F1800
; layer end
M104 T0 S0 ; turn off left extruder
M104 T1 S0 ; turn off Right extruder
M140 S0 ; turn off bed
G91 ; relative mode
G1 Z15 F1500 ; raise nozzle 15mm 
G90 ; absolute mode
G1 X30 Y240 F3000
M84 ; disable motors
; Build Summary
;   Build time: 0 hours 1 minute
;   Filament length: 52.8 mm (0.05 m)
;   Plastic volume: 126.98 mm^3 (0.13 cc)
;   Plastic weight: 0.16 g (0.00 lb)
;   Material cost: 0.01
thanks to all for help

thinkyhead, 2018-08-30:
Try lowering your Z max feedrate and Z acceleration. It sounds like Z is trying to move faster than the hardware can manage. After re-flashing, use `M502`, `M500` to ensure the motion settings are all saved to EEPROM.

khomouda, 2018-08-31:
[image]
Less than this?

khomouda, 2018-08-31:
I reduced Feedrate to 7 and acceleration to 70 and it worked, does this mean bad motors? these are NEMA 17 48 motors which are supposed to be high torque.
thank you very much for your help

AnHardt, 2018-08-31:
max steprate = (steps / mm) * (mm / s)
6400 steps/mm is about M6 with 32 microsteps.
For a first test on a 8 bitter stay below 10.000 with the steprate.
=> 1.5 mm/s

6400 * 10 = 64000 steps/second is enough to stress some 32bitters.
6400 * 7 = 44800 still exceeds the abilities of a 8 bitter  but should be ok for a 32bit system.

Does a z-resolution of 1/6400 mm = 0,00015625 mm make sense to you?
With a lower microstepping you could have much higher speeds.

khomouda, 2018-08-31:
thanks a lot, I changed the Drive micro-stepping in the Ramps through changing the jumpers from 1/32 to 1/8, and divided the steps per unit by 4, but the motors moved very little and the Z was stuttering, so I kept trying the steps numbers till I returned back to the same numbers to get the calibration right. what am I missing

khomouda, 2018-09-04:
Hi, was that a dumb question or was it difficult, I am sorry but I am newbie trying to understand more

AnHardt, 2018-09-04:
Not a dumb question - just not well fitting here. This is a bug tracker and for discussing the further development of Marlin. Not a forum, not a chat, not a blog, not for basic configuration related questions, not for newbies (not able to distinguish between hardware, firmware, configuration problems or missing features) - even when constantly misused.

thinkyhead, 2018-09-05:
> was that a dumb question or was it difficult

Not at all dumb. Here are some good places to get help with general troubleshooting. Some of the geeks who work on Marlin can sometimes be found hanging around these sites:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

closed 2018-09-05 by thinkyhead