MarlinFirmware/Marlin #11679: Automatic bilinear bed leveling doen't seen to be working properly

bill-orange, 2018-08-30:
Even though I am using  bilinear autoleveling (3 X 3), I have  high and low spots on large test prints.  My grid looks like this: 
1 2 3
4 5 6
7 8 9

'7' is always too low and '6' is always too high.  I need advice.  Here's my setup:

I am running an Anet A8 using Octoprint and Marlin 2.x.  I have successfully added an inductive probe.  I have added G29;M501;M420 S1; to my start code after the last G28.  I have tested my probe accuracy and found an acceptable standard deviation (actually, I think the correct term is population standard deviation).  When I run the bed visualizer in OctoPrint, I get a visualization of a symmetrical bed with a 0.4 mm hump in the middle (apparently not unusual).   My Z axis steppers are level.  My leadscrews and Z axis guide rods are no better and no worse than anyone else.  My best Z offset is -2.3 even though to just pass a sheet of paper I need to be down to -3.1 (odd).  Printing with a Z offset of -3.1 practically grinds into the bed.  I dont see this oddity as a problem as long as I know Z-2.3 is good.

What can I do to rid my printer of these high and low spots?  How can I find and fix the root problem.  If the root problem can not be addressed is there a way to program compensations into Marlin?  Should I be running a 5 X 5 pattern maybe?

Thoughts and advice?

Thanks,

Bill

thinkyhead, 2018-08-30:
Make sure there is no twist in your X gantry. Even a small twist, or something pulling on the wires, etc., may be enough to misalign the probe during the procedure and produce garbage readings.

RudyBenner, 2018-08-31:
M420 S1 in your start script AFTER any G28. This was supposed to be fixed but is still problematic. I use M420 S1 Z2 to include fade height.

bill-orange, 2018-08-31:
Put the M420 S1 after G28, okay. Before or after the G29 or does it matter?

RudyBenner, 2018-08-31:
After the last G28.

bill-orange, 2018-08-31:
On the second set of start code, doesn’t the G28 Z erase the G39 data you just collected?  I would think that you would need an M500 after the G29 and an M502 after the G28 Z. 

In any case, I will move my M420 around and see if it matters. At this point I need, for starters, to figure out if the problem is mechanical or software.

bill-orange, 2018-08-31:
Ah, not M502. M500 is what I am suggesting. It will store your settings.

closed 2019-02-13 by thinkyhead