MarlinFirmware/Marlin #11680: my Z steps per mm need to be proportional to the position (like a Scissor Lift)

Bartolobot, 2018-08-31:
my Z steps per mm need to be proportional to the position (like a Scissor Lift) I may have made an altogether new type of kinematics ( Cartesian with a twist). I am not very good at writing code more of a moron still sending morose code through my telegraph. lol. 
       I'm sure its a simple couple lines of code to calculate the Z motor travel with the actual bed displacement the math is easy enough  thanks to the Pythagorean theorem    
[https://courses.lumenlearning.com/waymakercollegealgebra/chapter/distance-in-the-plane/](url)

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AnHardt, 2018-08-31:
Instead of modifying Marlin, think about a preprocessor.
A, for example python script, reading in your g-code, searching the z-values, modifying them with your formular and writing them back to another file.

Usually this is much more time efficient (for you) than a proper integration of a new cinematic into Marlin.
DELTA, SCARA, COREXY all begun that way.

AnHardt, 2018-08-31:
Would make me wonder if the relation in a scissor lift would be linear/[proportional](https://en.wikipedia.org/wiki/Proportionality_(mathematics)). Please reinvestigate.

a² + b² =  c'² where c is constant. (Pythagoras)
 
[image]

Palatis, 2018-08-31:
I think implementing a new kinematics will be more appropriate.
else we can just take  preprocessor for DELTA and CoreXY, too.

the point is to run the same piece of `generic' gcode across different machines they should behave the same.

thinkyhead, 2018-09-05:
> …like a scissor lift…

Is it similar at all to the [Scott Russell Linkage](https://en.wikipedia.org/wiki/Scott_Russell_linkage) as used in the [Kappa](https://www.kickstarter.com/projects/ionic3dp/kappa-the-worlds-first-straight-line-mechanism-3d)?

AnHardt, 2018-09-05:
For example

[image]

Yes, the math should be about the same.

thinkyhead, 2018-09-06:
Interesting… https://www.wikihow.com/Find-the-Height-of-a-Triangle

Bartolobot, 2018-10-11:
Here's a few actual images of my machine! enjoy!

[image]

[image]

Bartolobot, 2018-10-14:
I think the Graph would look more like this 
[image]

thinkyhead, 2018-10-26:
Kinematics is simply solving a right triangle.

You know the length of the scissor arm, `C`.
You know the height you want, `B`.
So you just need to solve the base length, `A`.

```
A = SQRT(C*C - B*B)
```

For example, if the scissor arm is 1000mm and you want to move Z to 20mm height:

```
A = SQRT(1000*1000 - 20*20)
A = 999.799
```

Bartolobot, 2018-11-25:
Thinkyhead, 
        Thank you for the advice.  but I don't think a programmer is really necessary I'm not making the machine to market or gain any profit from it. As you can see I have been open with all the details And its here for anyone to use in their personal projects. This is just a hobby, I do all the work myself and am happy to share my Ideas with the Marlin community. and I really can't afford a programmer. 
I only ask for guidance if you are willing to help me and possibly anyone else with similar  Kinematics design features. and open the doors for out of the box Thinking for new improved designs! 
Isn't that what its all about here, I mean if you feel your advice is valuable then write a book! 
I may buy it! Otherwise, I would appreciate any guidance On where I would implement something like what you described in your previous post.

AnHardt, 2018-11-25:
I thought a bit about your kinematic.
I see more and more problems.
In a real system you neither can reach a = 0 nor b = 0.
When you hit an endstop or the hard limit you are at a-d=0 or b-e=0.
You have to know either d or e very exact to find out where on the curve you are. Adjusting endstop is very critical. Much more difficult than in a linear system because not only the absolute position but also the steps/mm at that point is affected. So if you don't get it right linearity is messed up.

Scissor lifts are able to lift big loads. But they perform best near their maximum height. At the bottom they perform badly, move a lot, but have less power than at the top. Beds of 3D-printers need the opposite characteristic. The load is biggest when the bed is far away from the nozzle - the part has grown up to there and is heaviest. 

Convince me of the advantages of a scissor lift compared to a conventional solution.

Bartolobot, 2018-11-26:
Well, the primary advantage is it overall economically makes having a huge build area (bed) more feasible and/or Multiple Build areas some sections may be heated or cooled with adhesive or not, without the need of clearing the plate for subsequent prints and this remains to be seen hopefully allowing it to print for very long periods of time on multiple objects unattended, (I do have a webcam on the list of additions)  My build area is around 900mm x 460mm x 500mm ! When considering such a sizeable build bed it becomes quite evident that eliminating the substantial reinforcements and equipment needed to move and keep the bed both rigid and level, While simplifying the overall design makes sense. With the one exception, of course, the Scissor! and I did consider this being somewhat of an issue but I did not imagine how bad it would actually be, I figured shouldn't be too hard just tweak a couple lines of code and well be on our way, On the other hand and (before I list the several more advantages) lets consider what it would take to make a bed of that size move and stay level and ridged. 1) The cantilever design although an efficient design it has many inadequacies especially the bigger it gets the worse the problems are And so the liner rods and bearings would have to be huge! Also, with all other configurations that I've seen out there, the Printer Is Inherently as big as its build Volume plus the support structures required to achieve its build volume. That is if you have a build volume of 250mm cubed that means your machine is around 300mm+ cubed. Where my design starts very compact and flat. Making my Design a Large format printer that's portable, storable and out of sight if need be- Under your bed! Let's consider another more effective design for A large format printer that's no longer portable or storable and you can't hide it from your destructive nephews! That is a Large and expensive Lead screw at each corner and ......its complexities..... I think you get the picture right? Scissor Lift Bingo! easy with lots of added benefits only one catch Modify the code. 
**I hope you'll agree with me That modifying the code just a tiny bit has got to be outweighed by orders of magnitude by these other alternatives!**

**_Alleged Issues,_**

    Let me assure you the Issues you mention Have been addressed and are in fact not an issue with my design 1)  I do have hard stops and end stops incorporated on all axis  2) with the addition of a rotary encoder attached to the Nema 23 Z stepper, at the moment only mechanically attached and working Just haven't figured out how to wire it to my Ramps 1.5/then how to use the feedback to make position corrections should that be necessary mid print. 3) Of course Homing the machine is imperative prior to starting a print and as long as No! Z steps are lost or incorrectly executed all will be copacetic  Furthermore this starting point is the bottom of compression and should always return to that exact spot in relation to the nozzle/bed distance every time. end stops don't get harder than a completely compressed scissor lift.
4)Your right about the performance **increase** at the maximum height and increased load at the bottom This has been mitigated with the addition of heavy spring energy that works with opposing force to help equalize the performance. at the start, they have maximum energy stored at the bottom start pos. and at about mid hight they have released all of the kinetic energy back into the system as they are only needed most at the start where the loads are very high. This type of system can be tuned accurately and reliably. I might add that a NEMA 23 high load stepper with acme ball lead screw offer a considerable mechanical load/advantage on their own. One more thing to add on this issue also applies to the next is usually on a scissor lift you will have multiple stages more than one X stacked on top of each other. This would compound both the advantages of performance and stability for which I have no need for both so my design has only one stage. 5) Rigidity 

 **_More advantages,_**  substantially less need for costly and constant leveling of the bed surface. with less setup time and Better print quality while conveniently keeping most wiring safe and stable (no repeated bending of conductors ) prolonging wire/terminal life. Stable Environment under the bed making upgrades and additions easy and worry free of added weight loads on any moving parts. making the electronics area concealed and controlled as far as cooling plus dust and debris from extruder can be kept out! So more than 50% of the entire mass Is under the bed keeping the upper half Sleek, Clean, Compact, and most important of all FAST! 
I have More but I think I made my case.
A rather Lengthy reply so I rest my case. And wait for the verdict
I appreciate any help or guidance just point me in a fruitful path to neverending Plastic Sky's And I will not hesitate to do the same for you my fellow swordfish Anglers! 
                                 With Regards
                                   Bartolobot 13.

Bartolobot, 2018-11-27:
Did someone say elaborate my point?
Forgive me.
Anhardt, Thinkyhead
      I know its lengthy.  I'm aware of my poor grammar and run-on sentences from hell 
But please bare with me and I'll try to keep it short 
If I was blessed with any talent I know it was in the field mechanical design and problem solving, If only I could say the same for writing and understanding code.

thinkyhead, 2018-11-27:
The core Marlin team is busy for the next few months working on bug fixes, cleaning up code, polishing existing features, and preparing 32-bit support, so we don't have time to put into this issue. I recommend the following resources to ask for advice and recruit help:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

AnHardt, 2018-11-28:
> I do have a rotary encoder physically attached to the z stepper motor .
But I have not figured out how to get position feedback from a rotary encoder implemented into marlin!

Rotary encoders usually do not help to find a absolute position, but they can tell how far it is between two points (in encoder steps unit) if you count their steps. Mounting the encoder to the z-stepper (a) will only help to find out if the stepper lost steps (comparing stepper steps with encoder steps). Much more useful would be to know about  z (b). With comparing delta a to delta b at 2 or 3 points one could set up the transition curve properly. 

If you get that few positive feedback (besiteds of "Uh - that looks sollid.") , you are probably the only one for whom a solution of this problem is useful. For us GPL-programmers it's like a 'public amateur crossword puzzle contest' without a concurrent, where just the organizer is watching, but the spectators are absent - likely unsatisfying.

Bartolobot, 2018-11-30:
The basic problem I need to solve first and foremost is the Z step correction of diminishing returns with every z step progression relative to position .
I would just like to know what part of Marlin firmware would have to be modified to correct the steps 
Assuming I have an absolute starting position every time after homing, No steps are lost along the way, and Position is known if we have all that,
    Then what file or part is z steps calculated if that was all I need to do? 
And I do understand that those things will need to be also dealt with to have reliable results
but need to start somewhere? 
 a simple request to start for my comprehension of what the current code may look like and maybe I can gain insight as to what might be adapted or modified perhaps even implement a fix myself and test and learn that simple.
anyone Know?

Bartolobot, 2018-12-02:
at AnHardt

Yes, I know this and agree with you. But,
Shouldn't the system work just like any other printer? without an absolute position sensor 
in a relative way not absolute considering of course 
A Correct formula mathematically is codified into marlin for correcting the displacement issue in my z-axis
and give that a go 
In the meantime, I'll be looking at those magnetic position sensors and have one the way here soon.

boelle, 2019-03-13:
@Bartolobot still working on this?