MarlinFirmware/Marlin #11681: Z Probe Offset with large values (-16.8) not working

davidmilunic, 2018-08-31:
The Anycubic Kossel Delta has a large Z Probe Offset. Mine is -16.8 mm. This number is not being handled correctly when LCD Z Probe Offset is used to adjust it.
When I adjust Z Probe Offset through the LCD not all of the number is displayed, only one digit before the decimal is displayed. i.e. "-6.8" instead of "-16.8". Sometimes the number is totally wrong.
After adjusting the Z Probe Offset a wildly different number gets stored to EEPROM, usually around -200 so that the printer starts printing in mid air.
The only way that I have been able to reliably set the Z Probe Offset has been to edit the EEPROM values through Repetier Host.
The problem seems to be the worst when I set the Z Probe Offset through the menu item under Delta Calibration. I have also disabled the babystep z offset feature for M851, otherwise babystepping z causes the Z Probe Offset to get wonky. I am currently using 1.1.x-bugfix from just a few days ago (today is 31AUG2018).

davidmilunic, 2018-09-01:
When setting z probe offset under delta calibration, the extruder will move halfway down to the bad and then it raises a bit, then continues. It looks like it is backing off by the current z probe offset.
After setting and saving, the value of z probe offset in EEPROM is twice as big.
I think it is adding the old and new values together instead of adjusting the z probe endstop by the difference between the original value and what is measured during set z probe offset under delta calibration.
How to fix? I went from 1.1.9-bugfix to 2.0-bugfix with the same results.

davidmilunic, 2018-09-01:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2342358/Configuration.zip)
I have attached my configuration.h and configuration.adv.h
What is the weirdness with z probe offset adjust?

boelle, 2019-02-20:
@davidmilunic still having issues?

closed 2019-02-20 by thinkyhead