MarlinFirmware/Marlin #11682: Marlin 2.0 Extruder motor take to hight speed

3DRaft, 2018-08-31:
Delta printer with Mks Sbase 1.3 / Marlin2.0 last bugfix.
Configured as i think corectly. Files Atached. 
Problem with extruding feedrate. When printer go print speed of filament feedrate turn very fast, seems like speed * 2.  
Played with other function like district e/ smoothstepping/ and pulses but no different.
When printer in home position and I put filament moove he move correctly how i set 10mm.
I wath g code 
G92 E0
G1 E10 F120 
So Repetier host set speed F120. But in the slicers Cura or slic3r speed set for all line of code 
like G1 x/// y/// e/// f1200
and the extruder motor give this speed but it's no correct. 
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2340223/Config.zip)
So i try to set feedrate for extruder for minimum but it's not help. 
What could be the reason ?

3DRaft, 2018-08-31:
Update:
I have Diamnd Nozzle.
It's trouble with mixing extruder.
When i turn off  mixing extruder and set it to first extruder, this work fine. 
So it's need to make to make 3 extruders and edit sbase pins for make same signal from tempereture and hotend. I am working for it.

3DRaft, 2018-08-31:
Assignment pins to one extruder not working.

3DRaft, 2018-08-31:
UPDATE:
It's work with Single_Nozzle

thinkyhead, 2018-09-09:
Many changes were made to the planner / stepper code for 1.1.9 by @ejtagle, myself, and a few others. Perhaps we broke the `MIXING_EXTRUDER` in the process.

Be sure to do `M502`, `M500` after any re-flash to refresh the EEPROM and current settings.

e-Mole, 2018-09-09:
The same problem in Marlin 1.1.9 (A. Mega + RAMPS) - extruder motor speed is too high if MIXING_EXTRUDER is ON.

ejtagle, 2018-09-09:
@e-Mole : Are you using Linear Advance ? ... It is not supported with mixing extruder ... I still have to figure out how to make it work, but it is very hard to do.
Make sure to completely disable Linear advance!

e-Mole, 2018-09-10:
@ejtagle : LIN_ADVANCE is disabled (I forgot to write that my printer is delta)

3DRaft, 2018-09-10:
I tried with disabled LA, Junction Deviation and S-Cruve and the same problem.

e-Mole, 2018-09-12:
@3DRaft : Do you also use a Delta printer or Cartesian printer?

thinkyhead, 2018-09-17:
Duplicate of #11310.

closed 2018-09-17 by thinkyhead