MarlinFirmware/Marlin #11691: BSD, MIT, or Apache license?

thinkyhead, 2018-09-01:
It's a bit of a hassle of dealing with GPL violations, especially in the PRC and other remote regions, which is required in order to maintain the integrity of the license. We cannot compel vendors to comply with the license by our own means, since it costs $ to bring lawsuits. The FSF recommends using negotiation and dialogue to obtain compliance, and sometimes that does work. But if a vendor firmly resists, the only remedy left to copyright holders is to file IP Infringement claims with resellers.

With all this in mind, it occurred to me that the whole problem could be solved by simply moving Marlin to a more permissive license, such as the BSD, MIT, or Apache license. But so much of Marlin is either derived from GPL software or actually statically linked to GPL software that I think this is an impossibility.

The GPL actually does a lot more than just keep the software open as it gets passed around and modified. It also prevents fragmentation by ensuring that variants and odd forks can be merged back to the main project.

Anyway, even if changing licenses is impossible I thought it would be interesting for us developers to speculate on and discuss the potential effects, hazards, and benefits that would accrue if Marlin were placed under a more permissive license.

- What would such a change do to the marketplace?
- How would it affect clone makers, especially in the PRC?
- Would it lead to greater fragmentation, as lots of closed but clearly derivative firmwares appear bearing funny names?

Moyster, 2018-09-01:
> What would such a change do to the marketplace?  

More closed-source derivatives without needs to redistribute modifications/sources.  

> How would it affect clone makers, especially in the PRC?  
Licenses are just words, they don't care about words.

> Would it lead to greater fragmentation, as lots of closed but clearly derivative firmwares appear bearing funny names?  
Nilram Firmware incoming, some regexp to replace marlin name / website, *insert fancy colored LCD support*, disruptive firmware !

RudolphRiedel, 2018-09-01:
I believe the GPL is just right for a project like Marlin where lots of companies make money from it.
If anything the license should be more restrictive and not more open.

If a company is in violation, try to get in touch.
If a company ignores this, point them out, on Twitter, on the website, on Facebook, in the repository.
If the company still ignores the GPL, remove the support for their printer(s). Yes, this is more a signal than a real option, but hopefully somewhat louder than Twitter.

And since this gets highly frustrating pretty quickly, find someone who is willing to reach out to the companies for the team.

thinkyhead, 2018-09-03:
Thanks for these comments! Of course, I strongly believe in the GPL and would never mess with that aspect of Marlin.

It's been a frustrating time dealing with non-complying vendors, especially in the PRC, and attempts to highlight these violations —on Twitter in particular— have generated mixed reactions. On the one hand, there is support for the GPL and the community for the most part is annoyed by the vendors who simply ignore the license, or apply it selectively, or who just see it as an aspect of marketing and not an ethical and legal issue. On the other hand, attempts to call out non-compliance are sometimes met with sneers and jeers, especially from those who depend on affiliate links.

Some of the companies currently in violation and refusing to give the source to users include:
- **Creality 3D** has complied for a few of their machines, but not all.
- **Anet** simply refuses to open up the source, because they don't want to.
- **Tronxy** simply doesn't respond to queries.
- **Airwolf 3D** (Costa Mesa, CA) hasn't yet been contacted.

We're tracking known license violations at the website http://www.3dprintergploffenders.com/

As for compliance in the PRC… Initial blowback from the community did get Creality 3D to comply to the small extent that they have. It's not clear, but it appears to me that they are making symbolic gestures rather than honest efforts to comply.

For some reason, instead of addressing our concerns Creality also took it upon themselves to make the Ender 3 an OSHWA-compliant design. Feeding off of the RepRap project parasitically, they seem to view these moves as part of a marketing strategy rather than a natural extension of an ethical core (like, for example, the highly principled AlephObjects).

A well-known maker in China with a large YouTube following advocates for GPL compliance on her channel, and she believes she can influence the industry and move Creality and others to comply with the GPL. Since their initial gestures, Creality has shown no movement, so far failing to produce the rest of the source code promised. But I've been persuaded to take a wait-and-see approach to see what she can accomplish over the next 12 months.

Chris Bennett of TinyMachines3D is also advocating for us. During ERRF I had dinner with him and Jack Chen of Creality to discuss the GPL. Jack said "we understand the GPL" and then proceeded to ask for permission to violate the GPL (for 6 months) — which of course is not something anyone can grant.

I'm not sure what Jack was actually asking for. Did he want 6 months to get their stuff together? Did he want to be able to keep the sources from customers for 6 months after _any_ new machine release? Or, did he want a 6 month period where no one would openly criticize them? I still don't know. Maybe all of the above.

So clearly there is more education needed.

These companies mostly know what the GPL says on paper, and those who comply seem to honestly appreciate the importance of honoring the license. For those who don't comprehend the GPL or stand up for it, will they really be educated through a popular YouTube channel? And if they do comply, will they comply for the right reasons? Will they comply one day and then decide to stop as soon as the wind changes direction?

What I ultimately want is not just compliance, but a permanent and abiding enlightenment! If all else fails, we might just have to call in RMS for an intervention.

forkoz, 2018-09-03:
I don't really get the motivation to keeping the source closed on these printers. All it does is make it harder to configure marlin. Have any of the released sources even had major modifications?

AnHardt, 2018-09-03:
The motivation is usually to reduce the cost for support.
If you have a fix, unconfigurable firmware you can have it  virtually bug free after a while, for that special configuration. What stays is likely to be a hardware problem.

If you have only one, or a few, Marlins distributed in hex, you have to support only one, or a few, different configurations. If you open the source, suddenly you have to support a "bazillion" of configurations causing a "!(basilion)" of different problems, your supporters don't know about.

thinkyhead, 2018-09-04:
Apparently AirWolf 3D does supply source code upon request to customers, but you do have to write to request it. Strictly speaking, if their HEX files are published openly on the web, then at least the GPL license needs to be included with it, and the source code should _probably_ also be included, since _anyone_ who is given the HEX file, customer or not, is really entitled to the source code too.

> Have any of the released sources even had major modifications?

Not really. I'm told that Creality "paid a lot" for the Power-Loss Recovery feature. If you examine [their contractor's code](https://github.com/ShenzhenChuangxiang3D/Marlin/commit/9469395a516edfd8aea50ae76ccb9b44b6956436), you'll see it was basically a weekend project for someone who already knows Marlin pretty well, but I'm guessing it took the contractor extra time to get familiar with Marlin before they could write the feature. And, it is not what I would call a "recommended" feature due to the wear it places on the SD card, especially for vase-mode prints, which will write to the card basically constantly.

(Which reminds me, I really need to add a G-code switch to turn off Power-Loss Recovery. Otherwise vase mode prints will be bad news for SD cards and might also end up running slower, since the SD card is being written to several times per second.)

thinkyhead, 2018-09-04:
> If you open the source, suddenly you have to support a "bazillion" of configurations causing a "!(basilion)" of different problems, your supporters don't know about.

Sort-of. Strictly speaking a vendor has every right to say "If you modify the firmware or replace the board, etc., then your warrantee is void and we can't offer support." But of course the customer has to volunteer that information for the vendor to know. That said, I think most people who are capable of modding and re-flashing the firmware are less likely to call the company support line.

fiveangle, 2018-09-04:
> With all this in mind, it occurred to me that the whole problem could be solved by simply moving Marlin to a more permissive license, such as the BSD, MIT, or Apache license.

Removing the right-to-source license does not "solve" the problem of derivative projects not providing source.  It simply redefines their current behavior from "illegal" to "legal".  How does this "solve" anything except for them ?

> I strongly believe in the GPL and would never mess with that aspect of Marlin

😳 

-=dave

AnHardt, 2018-09-04:
Changing the license would need explicit confirmation of all contributors for every file. That's illusory. You'd need very good arguments to get mine.

marcio-ao, 2018-09-04:
> Changing the license would need explicit confirmation of all contributors for every file. That's illusory. You'd need very good arguments to get mine.

I second this. I know that my company in the past has not been willing to allow such re-licensing of our contributions.

> Removing the right-to-source license does not "solve" the problem of derivative projects not providing source. It simply redefines their current behavior from "illegal" to "legal". How does this "solve" anything except for them ?

^^ This. If the GPL can't be enforced, at least such vendors can be shamed, which is way cheaper than enforcement :)

thinkyhead, 2018-09-09:
Of course the license can't _**actually**_ be changed, as it is deeply rooted in several GPL projects and is statically linked with several GPL libraries.

I'm just asking for an exercise of the imagination, to come at the problem of untouchable license violation from an oblique angle and see if anything shakes out.

In any case, some DMCA takedown forms did get filled out by one or more Marlin copyright-holders, and some products were removed from Amazon as a result. So we know that IP infringement claim forms do have a direct effect on license violators. Perhaps GearBest will remove Anet product listings next…?

Roxy-3D, 2018-09-16:
> In any case, some DMCA takedown forms did get filled out by one or more Marlin copyright-holders, and some products were removed from Amazon as a result. So we know that IP infringement claim forms [...]

Actually...   That is a good point.   eBay would also honor a DMCA take-down form.   So if we learned how to do this and got proficient at it...  We could deny any infringer access to both eBay and Amazon.   **THAT** might just get their attention.

thinkyhead, 2018-09-17:
The form on Amazon is pretty easy to locate. Anyone who has contributed code has standing to file an IP violation complaint. GearBest only provides a PDF that must be filled out and … emailed (?) I guess. I haven't looked at other sites. The machines that should be addressed right now are ANET. They simply refuse to release source, and so far have not shown any openness to discussion with us.

boelle, 2019-02-20:
my take on this is that we cant do much about it, 

you could go sue them in china, but i think that you would have better  luck with a lottery ticket than convincing a china judge that they did something wrong

there are only 2 options that comes to mind but they are not an option

make people but printers that comply with license, not very likely to happen since people will get what is cheap
go closed source, not an option

@Roxy-3D idea is the best bet... thou i will doubt if we get much lucky with china based sites like gear best

boelle, 2019-03-06:
@thinkyhead i think we can close this one

all we can do is suggest people dont get these printers and that maybe we dont support them more than we do allready

boelle, 2019-03-13:
@thinkyhead maybe have a pinned issue that list the printers that dont comply and which people should stay clear of?

not sure if we should go as far as not support them directly

Moyster, 2019-03-13:
> not sure if we should go as far as not support them directly

Yeah, let's put the blame on the end users 👍

boelle, 2019-03-13:
i mean we can have example configs etc

of course we should not harm the end users, just try to put out a strong message that these printers do more harm than good

but its a hard one, but we can be sure that the company in china does not give a rats arse about it as long they sell printers they are good

boelle, 2019-10-24:
will close this one as it has served the purpose

the only thing we can do is suggesting what to buy and pick sellers that repects the GPL

but to be realistic we cant do much else

closed 2019-10-24 by boelle