MarlinFirmware/Marlin #11693: [Request] Prusa Mk3 laser optical sensor available on aliexpress now .. need implementation

studiodyne, 2018-09-01:
Hi , thanks for all are interested 

The laser sensor or the prusa MK3 is sold on aliexpress 
[here](https://fr.aliexpress.com/item/Prusa-i3-MK3-3D-Imprimante-Filament-Capteur-D-tecter-Coinc-Filament-Et-Offrir-L-utilisateur-une/32907430687.html?spm=a2g0w.search0303.3.1.3d0e7a65BKY717&ws_ab_test=searchweb0_0%2Csearchweb201602_0_10084_100031_10083_451_10304_10821_10843_452_10307_10618_10065_10068_10059_10103_10696_10820_10302%2Csearchweb201603_0%2CppcSwitch_0&algo_pvid=de99257e-d850-425e-9849-be82a0a4b45b&algo_expid=de99257e-d850-425e-9849-be82a0a4b45b-0)

Someone to implement PRUSA code in 2.0 ?
This sensor will be usefull for runout , clog , jam , it's a very good improvement if implemented
Low size and easy to install on the carriage or the printer

Thks

AnHardt, 2018-09-01:
As far as i know Prusafirmware currently does not use/include the movement detection, just the presence. And even that is not always reliable.

studiodyne, 2018-09-01:
The Prusa I3 MK3 have new types of sensors , recently added only on the last release MK3 
This sensor have so many advantages 
No contact , no braking , no friction 
Light weight and size , to be installed close to the gear in the carriage
Can inform about  all bad movements , clogged , jammed , runout , cold nozzle , sanding of filament , and the print can be stopped and can be also ' SAVED' 

Thks for all that want to work on it

GMagician, 2018-09-01:
@AnHardt I also checked code and for what I can understand @studiodyne is right, but I agree with you on

> And even that is not always reliable.

it seems it has issues on transparent filaments

workheart, 2018-09-01:
I do remember a interview where he mentioned that the implemention in FW will be later than the shipping of the first units, but the included the sensor anyway because.. they can and shipping SW is no porto

studiodyne, 2018-09-01:
the sensor is, however, subject to default, but if we implement a comparative monitoring on the lcd, and a 'TOLERANCE' setting, we can add our 'error correction' for each filament that gives poor capture results.
For example the LCD could display the motor filament distance and the captured distance, and we could check the margin of error and set it in the 'control menu' for each print. I have heard about laser problems, and also about overheating problems.

studiodyne, 2018-09-01:
The implementation of this kind of code is not simple, it is not a simple runout, but rather a measure of distance or a count, updated in the isr, to monitor the extrusion permanently. Whoever is going to write this code is necessarily a very competent person, in any case, I can not
Marlin doesn't have this feature , or a simple feature that make a count on a pin , and we create ourself our own sensor ' rotating magnet ou rotating optical ' , it's a shame 
This feature is so important for printing

GMagician, 2018-09-01:
> but rather a measure of distance or a count,

I think this is done directly by sensor

> updated in the isr

Only thing I saw from Prusa code is that info is read by I2C/SPI, not analized more the code because at the time sensor was not available.

studiodyne, 2018-09-01:
@GMagician  I have good news, the sensor works perfectly except for a few bright filaments, and in this case, it ejects the filament only after a few minutes, probably because the difference in distance becomes huge.I propose that we use a adjustment of 'Error Tolerance' from the LCD, and that at any given moment, we reset the counter, it does not allow to have a perfect count, totally useless, but, it allows us to have a monitor d ' very reasonable activity to detect a stopped filament.
For example, we display the error on the lcd, it allows us to see how the sensor behaves with each filament and then, we will adjust the threshold in the 'control' menu, so that the printing can continue
We do not necessarily need a millimeter count, we need to save the impression.

studiodyne, 2018-09-01:
The first thing to do is to implement it , and over the time the community will improve by experiencing this

studiodyne, 2018-09-01:
@GMagician  the second difficulty is that not being a simple runout, when the extrusion interrupt occurs, it is the buffer of the isr which is stopped, and therefore, the line of gcode is not finished , we must interrupt the current gcode, and be able to restart it when coming out of pause
In short, it's really too complex for a beginner

closed 2018-09-04 by studiodyne

loctite603, 2018-10-22:
One may increase the optical sensor reability by putting it after the drive gear. The drive gear slightly dents the filament, so it is easier for sensor to recognize the movement.

InsanityAutomation, 2018-10-22:
After the drive gear will never see runout as it can't push it past the sensor.

loctite603, 2018-10-23:
There are two filament sensors: one detects presence of the filament(simple switch), the another one checks for filament movement(it can be optical sensor as discussed). If properly installed(anywhere befor drive gear), the filament presence detector is reliable and there is no need for further  improvemet.

GMagician, 2018-10-24:
There are other ideas how to use laser sensor solving its weak points, adding a bearing so sensor can read bearing movement instead.
In thingiverse where are a couple of examples.

loctite603, 2018-10-24:
The main idea behind using purely optical sensor - it has no moving parts, simple and compact construction.  By using the rotating bearing to recognise a movement you are going to implement a sort of rotary encoder. As you know, this method has been widely used in PC mouse. I have seen number of successful projects in 3d printing with this type of encoders.