MarlinFirmware/Marlin #11695: [BUG] Z_MIN_PROBE

Amachete, 2018-09-01:
Hello

I am triying to update to the last version, i have tried with both versions 1.1.9 and 2.0, and my probe does not stop at homing.

Z_min is below bed level and I use the z_probe as Z_homing.

#define Z_MIN_PROBE_ENDSTOP
#define FIX_MOUNTED_PROBE
#define Z_SAFE_HOMING

#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  //#define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _XMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif

#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  58 //old32
#endif

In a very old setup this used to work but  I am not able to do it work with this latest releases.

I will attach the three files, ramps, configuration and configuration_adv of the latest 1.1.9.

[Marlin_1.1.9_z_probe_issue.zip](https://github.com/MarlinFirmware/Marlin/files/2342222/Marlin_1.1.9_z_probe_issue.zip)

Thank you.

Amachete, 2018-09-01:
Hello again.

I have messing around the code, and changing the following made it work:

My setup:
Zmin is under level of bed as safety endstop just in case the probe does not work prperly.
Z_Probe as the real Z_min, and for bed leveling.

endstop.cpp

if (stepper.axis_is_moving(Z_AXIS)) {
    if (stepper.motor_direction(Z_AXIS_HEAD)) { // Z -direction. Gantry down, bed up.
      #if HAS_Z_MIN
        #if ENABLED(Z_DUAL_ENDSTOPS)
          PROCESS_DUAL_ENDSTOP(Z, Z2, MIN);
        #else
          #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
            if (z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
          #elif ENABLED(Z_MIN_PROBE_ENDSTOP)
            //if (!z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
            if (z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
          #else
            PROCESS_ENDSTOP(Z, MIN);
          #endif
        #endif
      #endif

Conditionals_LCD.h

//#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && (ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)||ENABLED(Z_MIN_PROBE_ENDSTOP)))

With this changes I can make the z axis stops with both Z_endstop and with the Z_Probe.

The Bed leveling works too!!!!

I just want you all to know!

I do not know If it´s worth to add those changes to the code source.

Bye.

Amachete, 2018-09-06:
Hello

Thank you for your quick replay.

My printer is a PRUSA STYLE, the Z_HOMING is ON the bed, It does not matter  if It is on the 0.0.0 corner or just in the middle of the bed (Z_SAFE).
[image]

My point of view is if you have a pretty level bed printer the mechanical Z_ENDSTOP works right for you.
But if your bed is unleveled….

[image]

Look at the picture , when the Z_HOME is done There is some places of the bed level I can not reach because of the mechanical endstop, If it were optical I could pass through, but not with mechanical.

[image]

In the last picture you can see if you make a Z_HOMING OUTSIDE of the bed  you could reach the whole bed, ishhh. the current marlín setup Works perfect, but It is not my case, yet (Dual Z_motor and Z_DUALENDSTOP, are on its way)

[image]

That´s why I thing we need at least a way to use Z_HOME sensor as Z_ENDSTOP, both at the same time, without using the same pin (my setup).

In my opinion It should not be so difficult to implement something to make the Z_end_stop and Z_probe to work together as Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN does , something like "Z_MIN_PROBE_WORK_TOGETHER"

Sorry for my English but I hope you can understand what I mean.

Cheers!!

swap81, 2018-10-15:
hey friends i use the normal configuration. 

 z min is without function! have already tried the above-mentioned moifikation with copy and paste. without success !!

swap81, 2018-10-15:
[code]
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// example_configurations/SCARA and customize for your machine.
[... 1833 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#endif // CONFIGURATION_H
[/code]

thinkyhead, 2018-10-16:
@swap81 — Please don't post your whole configuration in a comment. And `[code]` doesn't work here. GitHub comments use Markdown.

swap81, 2018-10-16:
thanks for compressing. what is the solution for the z-min with cap.sensor problem?

thinkyhead, 2018-10-26:
Many possibilities. Lower the mechanical endstop so it doesn't get in the way. Or raise the bed a little. And/or increase the triggering distance to the capacitive probe. And/or, lower the capacitive probe.

closed 2018-10-26 by thinkyhead