MarlinFirmware/Marlin #11697: [2.0.x] Too long extrusion during change filament

GMagician, 2018-09-01:
I executed command by LCD and I got title message on host. I think this is not good, long extrusion check should not be considered during filament purge after change...

Am I wrong?

GMagician, 2018-09-01:
And it's also weird.
My cnf has `EXTRUDE_MAXLENGTH 200` and:
```c++
  #define PAUSE_PARK_RETRACT_LENGTH
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      100
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH    75
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE
  #define ADVANCED_PAUSE_PURGE_LENGTH         25
  #define FILAMENT_UNLOAD_RETRACT_LENGTH      13
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8
```

I'll do it once angain to test when message is asserted

GMagician, 2018-09-01:
Done other tests...echo messages appears after execution exited M600...
I also noticed in other situation a fast extruder advance while printer just after resume

GMagician, 2018-09-02:
Ok. I found something weird...don't know how reference parameters works in c++ so I can't say if this is a code or compiler bug

Guilty code is in `Planner::set_position_mm` and line seems to be:
```c++
position[axis] = LROUND(axis_steps_per_mm[axis_index] * (v +
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      axis == Z_AXIS && leveling_active ? ubl.get_z_correction(current_position[X_AXIS], current_position[Y_AXIS]) :
    #endif
    0
  ));
```

computed extruder value is 0. Just adding round brakets after `v` variable solve.

Here I attach original code disassembler and new my fix disassambler
[original.txt](https://github.com/MarlinFirmware/Marlin/files/2343753/original.txt)
[modified.txt](https://github.com/MarlinFirmware/Marlin/files/2343754/modified.txt)

thinkyhead, 2018-09-03:
Thanks for catching this regression!

closed 2018-09-03 by thinkyhead

GMagician, 2018-09-03:
May you give me an explanation about why brackets solve this?

I now how pointers works but reference is something like pointer for a specific type (is it?) when you add a value it should not change pointer base? What really happens on code?

Thanks

GMagician, 2018-09-03:
@AnHardt  I think this is not the issue, result was 0 but if you add 0 or 1 to a v you will never get

AnHardt, 2018-09-03:
Did i say "conditions value"? I mean "conditional value", so either `ubl.get_z_correction(current_position[X_AXIS], current_position[Y_AXIS])` or `0` should be added to `v`.
if you compare `((v + axis) == Z_AXIS)` you will always get '0' - false for the condition and for that the therm behind te ':' is selected - also `0`.

GMagician, 2018-09-03:
No it's me I misunderstood you..I edited my previously as soon as I correctly understood

GMagician, 2018-09-03:
Since understood real cause, is not better to read something like:
```c++
position[axis] = LROUND(axis_steps_per_mm[axis_index] * (v +
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      (axis == Z_AXIS && leveling_active) ? ubl.get_z_correction(current_position[X_AXIS], current_position[Y_AXIS]) :
    #endif
    0
  ));
```
instead of current solution?

thinkyhead, 2018-09-04:
Code is patched, so all should be well now.

PR #11708 (2018-09-03): [2.0.x] Fix planner computation
Fix #11697
commit 59e76e5f (2018-09-03): [2.0.x] Fix planner computation (#11708)

Fix #11697