MarlinFirmware/Marlin #11698: BL Touch Triggered, but doesn't stop

1liminal1, 2018-09-01:
### Description

Just upgraded to 1.1.9 on RAMPS 1.4, when trying to auto home, everything is fine till it trys to home Z. 

The BLtouch gets triggered as normal (you can see it get triggered), but the Z motor keeps going. 

Homing Y and X is working fine. 

controlling axis though pronterface works fine

ENDSTOP_INTERRUPTS_FEATURE is on 

Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is on

BLTouch is on 

I can also run;

M280 P0 S10 
M280 P0 S90 
M280 P0 S120

as normal. 

Thanks all

Roxy-3D, 2018-09-01:
There is something else you can try.   You can try making the Z Axis move faster during probing.   And turn the ENDSTOP_INTERRUPTS_FEATURE off.     That may resolve the BL-Touch issue.

(You can test by raising your nozzle high enough you can put your hand under it to trigger the probe.   And be ready to press 'Reset' if it doesn't stop.)

RudyBenner, 2018-09-01:
Check to make sure you are on Zmin, not Zmax. I did that once, it did not work.

1liminal1, 2018-09-02:
Thanks Roxy, 

Turned off the interrupts feture and raised the speed; 

HOMING_FEEDRATE_Z  (7*60)

Im assuming this is the right one? Did seem faster when it was trying to crush my finger haha

But no change, BLtouch is triggered, but wont stop the Z motor. 

Its really strange that there is a disconnect here. The BLTouch is triggered, why would it not stop it? 

Thanks

1liminal1, 2018-09-02:
> Check to make sure you are on Zmin, not Zmax. I did that once, it did not work.

> 

I actually raised an issue before for the same thing with the Y axis before releasing that I had Y and X swapped haha.

1liminal1, 2018-09-03:
Anyone have any idea? Could it be worth trying a bugfix version?

1liminal1, 2018-09-03:
Ok, so pulled out the cables, blew on them and put them back in and it works. I guess there must have been something making the connecting a little dicey. (I had reseated the connection before so)

All's well that ends well, thanks everyone.

closed 2018-09-03 by 1liminal1

Adzetko, 2018-12-11:
I have the exact same problem, on Marlin 1.1.9 too, with a legit BLTouch (from antclabs).
`ENDSTOP_INTERRUPTS_FEATURE` is off,
`BLTOUCH` is on (and nothing else concerning the Z axis of course),
I am on Zmin, not Zmax,
It is reacting to every `M280 P0 Sxx` commands,
`M119` returns "TRIGGERED" for the Z axis in whatever circumstances (deployed or not),
It continues returning "TRIGGERED" if I disconnect it,
I've tested the black and white cables with a multimeter in the diode mode (beeping when there is current),
`M43 S` (with `PINS_DEBUGGING` on) returns:
```
Servo probe test
.  using index:  0
.  deploy angle: 10
.  stow angle:   90
. probe uses Z_MIN pin: 15
. uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
. Z_MIN_ENDSTOP_INVERTING: false
. deploy & stow 4 times
WARNING - INVERTING setting probably backwards
please trigger probe
trigger not detected
```

(sorry if this is the wrong place to share this, I will open a new issue if you tell me to)

RudyBenner, 2018-12-11:
no files?

Adzetko, 2018-12-12:
> no files?
which ones do you want?