MarlinFirmware/Marlin #11699: What does " Error: EEPROM datasize error." mean

bill-orange, 2018-09-01:
I am running an Anet A8 using Octoprint and Marlin 2.x using Bilinear Bed Leveling.

 I tried changing `#define GRID_MAX_POINTS_X 3` to `#define` GRID_MAX_POINTS_X 4'

I get the usual flash successful message but then an Octopi disconnect with Error: "EEPROM datasize error." 

On the LCD screen all appears well.

I suspect that I don't have enough EEPROM left for a 4 x 4 matrix since I am already on the ragged edge memory wise with the stock ANET board.

What does this error mean?

GMagician, 2018-09-01:
> I suspect that I don't have enough EEPROM left for a 4 x 4 matrix

I think so

thinkyhead, 2018-09-01:
> What does this error mean?

It means you need to do `M502`, `M500` to update the EEPROM, because you've changed the required storage size of the EEPROM by modifying `GRID_MAX_POINTS_X`.

closed 2018-09-01 by thinkyhead

bill-orange, 2018-09-01:
This is my post flash code so that can not be it.  M502; M851 Z-2.48; M500; M117 Firmware Updated

thinkyhead, 2018-09-01:
Un-comment this line in `configuration_store.cpp`, re-flash, and we'll be able to see the general location where the data-size differs…
```cpp
//#define DEBUG_EEPROM_READWRITE
```

bill-orange, 2018-09-02:
Done! Oddly with `#define DEBUG_EEPROM_READWRITE`  I did not receive an error!  I made no additional changes ( other than the mesh size of course).  I can not explain it.