MarlinFirmware/Marlin #11700: Backtrace function definition

silentninja1, 2018-09-01:
Im getting a compile error about a duplicate definition of backtrace. Commenting out the empty definition on line 99 of Marlin/src/HAL/shared/backtrace/backtrace.cpp

void backtrace(void) {}

Seems to resolve the issue, but it seems its been there almost 2 weeks. Anyone else seeing that, or is their compiler happily ignoring the empty definition? I wouldve expected somebody else to have seen it since the last commit here was so long ago.

thinkyhead, 2018-09-05:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We'll try compiling with them.

silentninja1, 2018-09-05:
@thinkyhead It appears to be an ardunio ide issue, with the nightly build the error went away. The current "Stable" release is anything but... Compile errors on comments and more... Platform IO has some linter issues still on conditionals. I may go back to some of the visual studio plugins again for the moment. Hard to find a reliable method of building 2.0 that I can point people to.

closed 2018-09-05 by silentninja1