MarlinFirmware/Marlin #11701: BLTouch smart does not trigger

Alantin, 2018-09-01:
I know, yet another issue with trying to configure a bltouch :(

I am working with Marlin 1.1.3, a MKS gen v1.4 board, and a bltouch smart (genuine).  Connected to z min and servos 1.

Connect BLTouch, power up controller and upload the firmware
Blue LED will be ON if a servo signal is present.  - Check
M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used. - Check
M280 P0 S10 ; should deploy the probe and the orange LED will be OFF. - Check
M119 ; should return OPEN - Check
M280 P0 S90 ; should stow the probe and the orange LED will be ON - Check
M280 P0 S60 ; puts it into the M119 test mode. - Check
The probe should remain stowed and the blue LED should be OFF. If it's ON then the unit needs to be adjusted.  - Check
M119 ; should return TRIGGERED - **Still returns open**
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2342523/Configuration.txt)
[Configuration_Adv.txt](https://github.com/MarlinFirmware/Marlin/files/2342524/Configuration_Adv.txt)
[pins_RAMPS.txt](https://github.com/MarlinFirmware/Marlin/files/2342525/pins_RAMPS.txt)

GMagician, 2018-09-01:
You will never see TRIGGERED on bltouch. signals lasts only for 10ms.

I suggest, if boards support it, to use interrupts mode for endstops

RudyBenner, 2018-09-01:
Are you not accidentally connected to Zmax? I did that once. Did not work.

Alantin, 2018-09-01:
My understand was that M280 P0 S60 causes a persistent state that will continue to show triggered with M119.
I am sure it is z min. I removed the endstop and plugged the touch straight in.

AnHardt, 2018-09-01:
At the early BL-Touche's the 'triggered' pulse lasts only 5ms, on the later 10ms.
When the output is permanently 'high', it indicates the error state.

Alantin, 2018-09-01:
Very well. G28 also crashes the hot end into the bed.   It is not a z offset issue because I have tried manually tripping the z probe with my finger without any luck.

kAdonis, 2018-09-02:
I can confirm, after M280 P0 S60 (Wiring Test) the BLTouch shows triggered permanently.
Did you check the polarity of the wires at the endstop header? White- D18, Black- Gnd

Alantin, 2018-09-02:
I did. I even flipped them to make sure.

julianCast, 2018-09-07:
I have the same configuration/connections than Alantin. 

The bltouch, randomly, after deploy its stick, wont stow after touching the bed and it'll cause a crash into it. Sometimes happens after finishing/cancelling a print or a failed probed. I'll do the checks and post it here.

GMagician, 2018-09-07:
> I did. I even flipped them to make sure.

Sure that it is not dead/damaged because of this?

julianCast, 2018-09-08:
My checks:

Blue LED will be ON if a servo signal is present. - Check
M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used. - Check
M280 P0 S10 ; should deploy the probe and the orange LED will be OFF. - Check
M119 ; should return OPEN - Check
M280 P0 S90 ; should stow the probe and the orange LED will be ON - Check
M280 P0 S60 ; puts it into the M119 test mode. - Check
The probe should remain stowed and the blue LED should be OFF. If it's ON then the unit needs to be adjusted. - Check
M119 ; should return TRIGGERED - It returns triggered

GMagician, 2018-09-08:
@julianCast 

> Sure that it is not dead/damaged because of this?

that was for @Alantin , maybe you have different causes

GMagician, 2018-09-08:
@julianCast @Alantin you may try to replace your code with this:

1.1.x (Marlin_main.cpp -> setup()):
```c++
// Make sure any BLTouch error condition is cleared
bltouch_command(BLTOUCH_RESET);
set_bltouch_deployed(true);
set_bltouch_deployed(false);
```

2.0.x (probe.h -> bltouch_init()):
```c++
FORCE_INLINE void bltouch_init() {
  // Make sure any BLTouch error condition is cleared
  bltouch_command(BLTOUCH_RESET);
  set_bltouch_deployed(true, false);
  set_bltouch_deployed(false);
}
```

I'm using bl touch after such modification with no issues but mine is a clone and can have different behaviour

@thinkyhead what do you think? That cames from #11358 and #11135

julianCast, 2018-09-08:
My mistake ! I can't believe it was just the height, I trusted a downloaded mount, I modified just 1 mm and it work everytime. Hardware problems are the best. phew. Thank you @GMagician !

closed 2018-09-12 by thinkyhead

captroper, 2018-12-20:
Sorry to resurrect this. @julianCast what specifically did you modify? The height of the mount itself? The height of the bltouch? The little screw thing in the bltouch? I'm having the same issue as @Alantin  and wondering if the cause is the same.

GMagician, 2018-12-20:
@captroper I think he means height of the mount.

ats3788, 2019-03-04:
Hi guys maybe it helps someone.
In my case I have a MKS GEN L Ver 1.4 I had to switch the red wire (to the middle) with the blue wire
on the servo Jack and I had to switch the white and the black wire on the Z-Min Stop then my BLTouch triggers