MarlinFirmware/Marlin #11702: Expand Duplication Mode, to accommodate more than 2 prints at one time.

SlickNickeL, 2018-09-01:
Expand and improve Duplication Mode.

REPETIER has a "Ditto" setting by which 4 extruders can be simultaneously used to print 4 prints, on one X carriage. The benefits of this setting, is the ability to mass produce and manufacture many parts at one time.
Can Marlin consider expanding the current configuration to at least go to 3 extruders, from the current 2 allowed extruders? This would be a great benefit to any company who wants to manufacture single-run continuous filament productions. 
Strength, speed, appearance and accuracy are greatly improved, by keeping one print to one head from start to finish. 
Size is limited to the distance of each extruder based on the X carriage spacing, which isn't a problem for small pieces. Adjustable extruders with pressure clamps, can manually move and set extruders along the X-axis. 
Since most 3D printers made to perform such tasks, have one purpose and have identically matched hotends/motors, the data to all extruders can be cloned without a significant amount of problems in the prints, by simply assigning/cloning data from EO to E1, E2 etc. One model is sliced and sent to E0. Then all other selected extruders also receive cloned code from E0. 

Serious problems arise when motors are wired in parallel from a single stepper, to include overheating of driver, heat damage and fire. This is not an option due to safety concerns.

Configuration should include the following:

Firmware settings to engage Duplication Mode as a selectable feature.

Add an allowable nozzle count, adjustable to each specific machine. E0, E1, E2, E3...

LCD selectable nozzle/extruder, for the the number of heads that will receive E0's cloned code. This will determine how many copies are made, and what nozzles are active in the process.

All selected nozzles, should have an individual LCD with selectable, preheat/extrusion/movement options for print prep.

G-code to engage and set all of the above, for SDcard automation.

Smart3DCNC, 2019-03-04:
Is this available yet?

AnHardt, 2019-03-04:
No.

SlickNickeL, 2019-03-05:
I’ve been using Repetier to do this. I use 3 heads at the same time. 
I will switch back to Marlin once they get this out.
Repetier documentation is horrible. They also zero to Zmax end stop for auto level. It’s really bad. 
Marlin has a much better documentation base and it’s written in plain English.

Smart3DCNC, 2019-03-05:
Thanks busy editing Repetier now, code seems like a cleaned up Marlin
Dont like the web config tool

Roxy-3D, 2019-03-05:
This is fairly easy to do....    For the IDEX machines, the duplication modes are handled by changing a couple of lines in the Stepper macros.    You would just add a few lines of code to those macro's to make them issue step pulses to an extra stepper motor...    

It is hardly any code at all...

SlickNickeL, 2019-03-05:
I’m good as a power user for either firmware, but I don’t code. The most I do is make complex multi sequential printing routines for a single printer, and I have to do some configuration in between parts for quality control.
Marlin has great documentation, and solid development ideas in plain English.
Repetier is German and translated. He uses the word “rotation” in documentation for auto leveling. It drives me totally crazy as a user/creator/engineer. The auto level instructions that Repetier has online, will drive an intelligent person to stupid.

Smart3DCNC, 2019-03-05:
> This is fairly easy to do.... For the IDEX machines, the duplication modes are handled by changing a couple of lines in the Stepper macros. You would just add a few lines of code to those macro's to make them issue step pulses to an extra stepper motor...
> 
> It is hardly any code at all...

in Marlin or Repetier?
Example?

SlickNickeL, 2019-03-05:
Repetier already has duplication mode up to 4 heads? Maybe more if set in firmware. Has a screen for setup and uses an M code to activate if you want it on an SD. I have a one click setup that heats the heads and prints 3 of the exact same part all at once.

Roxy-3D, 2019-03-06:
>> This is fairly easy to do.... For the IDEX machines, the duplication modes are handled by changing a couple of lines in the Stepper macros. You would just add a few lines of code to those macro's to make them issue step pulses to an extra stepper motor...

>> It is hardly any code at all...

> in Marlin or Repetier?

Look at my Icon.    I'm talking about Marlin.

> Example?

In the next 3 or 4 weeks, a 'Symmetric Duplication Mode' will be merged into the code base for IDEX machines.   If you print a 'Left Shoe' you can have the other extruder print a 'Right Shoe' at the same time.   This is a Formbot inspired feature for their IDEX machines.   Compare the lines defining Stepper Driver macros in Stepper.cpp.   Look for the lines with  symmetric_duplication_mode in them....    This Symmetric duplication mode was added with just a few lines of code.   Similar to what I'm suggesting up above.

```

volatile int32_t Stepper::endstops_trigsteps[XYZ];

volatile int32_t Stepper::count_position[NUM_AXIS] = { 0 };
int8_t Stepper::count_direction[NUM_AXIS] = { 0, 0, 0, 0 };

#if ENABLED(X_DUAL_ENDSTOPS) || ENABLED(Y_DUAL_ENDSTOPS) || ENABLED(Z_DUAL_ENDSTOPS)
  #define DUAL_ENDSTOP_APPLY_STEP(A,V)                                                                                        \
    if (homing_dual_axis) {                                                                                                   \
      if (A##_HOME_DIR < 0) {                                                                                                 \
        if (!(TEST(endstops.state(), A##_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
        if (!(TEST(endstops.state(), A##2_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
      }                                                                                                                       \
      else {                                                                                                                  \
        if (!(TEST(endstops.state(), A##_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
        if (!(TEST(endstops.state(), A##2_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
      }                                                                                                                       \
    }                                                                                                                         \
    else {                                                                                                                    \
      A##_STEP_WRITE(V);                                                                                                      \
      A##2_STEP_WRITE(V);                                                                                                     \
    }
#endif

#if ENABLED(X_DUAL_STEPPER_DRIVERS)
  #define X_APPLY_DIR(v,Q) do{ X_DIR_WRITE(v); X2_DIR_WRITE((v) != INVERT_X2_VS_X_DIR); }while(0)
  #if ENABLED(X_DUAL_ENDSTOPS)
    #define X_APPLY_STEP(v,Q) DUAL_ENDSTOP_APPLY_STEP(X,v)
  #else
    #define X_APPLY_STEP(v,Q) do{ X_STEP_WRITE(v); X2_STEP_WRITE(v); }while(0)
  #endif

#elif ENABLED(DUAL_X_CARRIAGE)
  #define X_APPLY_DIR(v,ALWAYS) \
    if (extruder_duplication_enabled || ALWAYS) { \
      X_DIR_WRITE(v); \
      if (symmetric_duplication_mode) \
        X2_DIR_WRITE(!v); \
      else \
        X2_DIR_WRITE(v); \
    } \
    else { \
      if (movement_extruder()) X2_DIR_WRITE(v); else X_DIR_WRITE(v); \
    }

  #define X_APPLY_STEP(v,ALWAYS) \
    if (extruder_duplication_enabled || ALWAYS) { \
      X_STEP_WRITE(v); \
      X2_STEP_WRITE(v); \
    } \
    else { \
      if (movement_extruder()) X2_STEP_WRITE(v); else X_STEP_WRITE(v); \
    }

#else
  #define X_APPLY_DIR(v,Q) X_DIR_WRITE(v)
  #define X_APPLY_STEP(v,Q) X_STEP_WRITE(v)
[... 67 lines cut ...]
        MIXING_STEPPERS_LOOP(j) REV_E_DIR(j);
        count_direction[E_AXIS] = -1;
      }
      else {
        MIXING_STEPPERS_LOOP(j) NORM_E_DIR(j);
        count_direction[E_AXIS] = 1;
      }
    #else
      if (motor_direction(E_AXIS)) {
        REV_E_DIR(active_extruder);
        count_direction[E_AXIS] = -1;
      }
      else {
        NORM_E_DIR(active_extruder);
        count_direction[E_AXIS] = 1;
      }
    #endif
  #endif // !LIN_ADVANCE

  // A small delay may be needed after changing direction
  #if MINIMUM_STEPPER_DIR_DELAY > 0
    DELAY_NS(MINIMUM_STEPPER_DIR_DELAY);
  #endif
}
```

InsanityAutomation, 2019-03-06:
I have buying from the contributing manufacturer to merge that code on the 15th, so under 2 weeks! The only reservation I have on the above is no hardware to test on. As @Roxy-3D said it isn't very difficult but I'd like it to be tested before merging it. Perhaps if someone is bringing one to MRRF we can develop it on the spot? Lol

Smart3DCNC, 2019-03-06:
thanks everyone , this is exactly what I am looking for , pity I am to far away to bring a machine, but more than happy to test it.

InsanityAutomation, 2019-03-06:
@Smart3DCNC to confirm, you're basically envisioning a stacker s4 from what I understand?

Smart3DCNC, 2019-03-06:
> @Smart3DCNC to confirm, you're basically envisioning a stacker s4 from what I understand?

[image]

Smart3DCNC, 2019-03-06:
> @Smart3DCNC to confirm, you're basically envisioning a stacker s4 from what I understand?

This machine is a DITTO Mirror
[image]

Roxy-3D, 2019-03-06:
@Smart3DCNC   If the extruders all move together....   Having the macros support a 3-way duplication is not a big deal.   We just scale up the non-IDEX duplication mode to do 3 (or 4) extruders instead of doing 2.   

The code to control how many (and which) extruders are doing the duplication is going to be the bulk of the code.   But if you are willing to turn on duplication mode and have ALL of the extruders printing duplicated parts...    That is very straight forward and easy to do.

Smart3DCNC, 2019-03-07:
> @Smart3DCNC If the extruders all move together.... Having the macros support a 3-way duplication is not a big deal. We just scale up the non-IDEX duplication mode to do 3 (or 4) extruders instead of doing 2.
> 
> The code to control how many (and which) extruders are doing the duplication is going to be the bulk of the code (and effort). But if you are willing to turn on duplication mode and have ALL of the extruders printing duplicated parts... That is very straight forward and easy to do.

Thanks Roxy, this is perfect,
Machine 1, Tripple duplicator
 I dont even need menu items to enable 1,2,3 extruders I will interrupt the Stepper and HEx externally.
Machine 2. Mirror Duplicator
The mirror function is in anycase mechanical hard linked in the design of the machine the X carriages move in opposite directions

SlickNickeL, 2019-03-07:
This is all supported in Repetier.
Select how many heads in firmware, and Repetier will then provide a separate control for each head, and a screen for how many copies activating each head sequentially from 0-2 copies.

I saw the S4 and I have one that is similar for my purpose with three heads. One big problem with this design is leveling. It takes on a whole new level of thinking depending on the bed adjustments. 
Auto Level is difficult but can be accomplished. The bed material must be super flat. I use 1/4” ALCA 5.  
The workflow for my purpose is to print the same part in 4 different quadrants on the print bed. The first layer must be aesthetically sound. This becomes a problem when the heads move to a different quadrant to do the next set of prints. It’s the first layer that gets all squirrelly and becomes unacceptable for the quality needed.
This is accomplished by a purge and wipe prior to each print quadrant with no skirt. In theory it works, but not for a quality product. I can’t just walk away from this setup and expect to get the results needed.

This is why Marlins autolevel out performs Repetier. It can do the necessary corrections with ease as compared to Repetiers problematic bed matrix. Only problem is, Marlin doesn’t support more than 2 heads.
You May also he an issue with sanity check in Marlin if you Use three heads and try to ditto. If my memoryserves me correctly.

This is why I have submitted this Feature. It will give Marlin a superior competitive advantage than Repetier in manufacturing. 

Repetier has great clean simple code, but lacks documentation for understanding each procedure.

Any engineer who uses the word “Rotation” In autolevelig, without defining the parameters of that rotation as an application, might as well state the engineering description of screwed. Either one would be a rotation. Neither one would explain the process correctly for what is actually taking place.

Smart3DCNC, 2019-03-07:
> This is all supported in Repetier.
> Select how many heads in firmware, and Repetier will then provide a separate control for each head, and a screen for how many copies activating each head sequentially from 0-2 copies.
> 
> I saw the S4 and I have one that is similar for my purpose with three heads. One big problem with this design is leveling. It takes on a whole new level of thinking depending on the bed adjustments.
> Auto Level is difficult but can be accomplished. The bed material must be super flat. I use 1/4” ALCA 5.
> The workflow for my purpose is to print the same part in 4 different quadrants on the print bed. The first layer must be aesthetically sound. This becomes a problem when the heads move to a different quadrant to do the next set of prints. It’s the first layer that gets all squirrelly and becomes unacceptable for the quality needed.
> This is accomplished by a purge and wipe prior to each print quadrant with no skirt. In theory it works, but not for a quality product. I can’t just walk away from this setup and expect to get the results needed.
> 
> This is why Marlins autolevel out performs Repetier. It can do the necessary corrections with ease as compared to Repetiers problematic bed matrix. Only problem is, Marlin doesn’t support more than 2 heads.
> You May also have an issue with sanity check in Marlin if you use three heads and try to ditto. If my memory serves me correctly.
> 
> This is why I have submitted this Feature. It will give Marlin a superior competitive advantage than Repetier in manufacturing.
> 
> Repetier has great clean simple code, but lacks documentation for understanding each procedure.
> 
> Any engineer who uses the word “Rotation” In autolevelig, without defining the parameters of that rotation as an application, might as well state the engineering description of screwed. Either one would be a rotation. Neither one would explain the process correctly for what is actually taking place.

Thanks, at the risk of going of topic I had same concern on bed leveling , my Z is done by four stepper each doing an independent level . but the bed need to be FLAT which should not be difficult to achieve every CNC machine in the world has a flat bed.
I lean towards a composite bed as I can eleiminate any warping  expansion problems during heating and cooling
It seems Marlin will do it and just because I more or less understand the code, Il wait it out or modify it myself.
I am just a bit paranoid about version control/mixing libraries and multiple programmers working on the same software.
Unfortunately I am old school and have not got the hang of GITHUB and disjointed development, will have to make some time

SlickNickeL, 2019-03-07:
I don’t deal with CNC setups, just 3D FF. 

As I said before, I don’t code. It’s not that I can’t learn, it’s I would rather use my brain power in other areas.

Consider this mechanical conundrum. 
You have a perfectly flat bed tilted and adjusted on each corner with a motor. Add 3-4 print heads that ALL have to be on the same plane and planed to each other. Now take that plane and stretch it across an axis by 12 inches with a head placement every 3-4 inches apart. 
Now consider the single movement of any of the leveling motors on any corner?
What will that one correction do across the 12 inches of extruder head axis?
What if auto level probe/bed matrix records a .12 mm dip in an area of 10mm across the bed?
How can any one head be level across 12 inches with this registered dip?
Just the probe will be level and the closest head. The other heads will be in danger of crashing.
The wider the plane, needed to be leveled, the more difficult the application. 
A single point, no problem. 
Two points close together, no problem, do an average, skew is low. 
Three point over a wide area? 
All bets are off.  
It’s all about the first layer. .18mm first layer, perfectly over 144 sq inches with a 4 point spread.

Roxy-3D, 2019-03-08:
> Thanks, at the risk of going of topic I had same concern on bed leveling , my Z is done by four stepper each doing an independent level . but the bed need to be FLAT which should not be difficult to achieve every CNC machine in the world has a flat bed.

Just as an FYI:     I haven't gotten to it yet...   But I wanted to adapt the Marlin Bed Leveling for IDEX machines.     The IDEX machines share the X Rail for both extruders.    The current bed leveling does a good job of correcting for the current extruder.

But when both extruders are active (in duplication mode) it isn't possible to correct the Z-Height independently for both extruders.   What I wanted to do was calculate the best Z-Height correction for each extruder and average those numbers.    Neither extruder would get everything it wanted.  But probably each extruder benefits from the 'partial' correction.

Smart3DCNC, 2019-03-08:
> Just as an FYI: I haven't gotten to it yet... But I wanted to adapt the Marlin Bed Leveling for IDEX machines. The IDEX machines share the X Rail for both extruders. The current bed leveling does a good job of correcting for the currently active extruder.
> 
> But when both extruders are active (in duplication mode) it isn't possible to correct the Z-Height independently for both extruders. What I wanted to do was calculate the best Z-Height correction for each extruder and average those numbers. Neither extruder would get everything it wanted. But probably each extruder benefits from the 'partial' correction.

I thought about it I dont think its necessary, The best would be to put the BedLevelSensor in the middle of the 3 or 4 extruders.  this will autimatically give you a |mechanical avarage| If one cannot build a machine with a bed flat enough for the avarage correction to work, one should probably not attempt to build a machine like this and expect the software to fix the problems.

SlickNickeL, 2019-03-08:
> > Thanks, at the risk of going of topic I had same concern on bed leveling , my Z is done by four stepper each doing an independent level . but the bed need to be FLAT which should not be difficult to achieve every CNC machine in the world has a flat bed.
> 
> Just as an FYI: I haven't gotten to it yet... But I wanted to adapt the Marlin Bed Leveling for IDEX machines. The IDEX machines share the X Rail for both extruders. The current bed leveling does a good job of correcting for the currently active extruder.
> 
> But when both extruders are active (in duplication mode) it isn't possible to correct the Z-Height independently for both extruders. What I wanted to do was calculate the best Z-Height correction for each extruder and average those numbers. Neither extruder would get everything it wanted. But probably each extruder benefits from the 'partial' correction.

That’s what I do currently in Repetier. The fewer points used over the largest area to make a single plane of reference. I rely on the principal that ALCA 5 is flat. Makes a big difference. Heat has no noticeable effect on the plate, but it does on each individual head. I use my own custom made socks for each of the heater blocks. This stabilizes head expansion. I didn’t engineer in any way to change a nozzle height due to confinement restrictions in such a small space. My three head design is fixed and used to produce one set of parts. 
I currently can’t change temps on each head which would be nice. (I just haven’t figured out the correct g or m code in repetier or S3D.) The problem arises when the colorant in the plastics changes the viscosity of the filament. Makes a big difference on first layer. Be nice to dial the white/ light colored filament down a couple of degrees. My understandig is Repetier clones all of the information from extruder-0 to extruder 1&2. This is something to consider if using different stepper motors. You could have a mess.

Roxy-3D, 2019-03-09:
If we were to enable duplication mode for ALL extruders...   I am currently out of town and have no hardware to test the changes.   Even if I had access to my printers, I only have an IDEX machine and the changes need to be put into a different section of the code to handle your case....

Can the two of you dedicate some 'test time' to check out the changes and provide feed back?

Smart3DCNC, 2019-03-09:
> If we were to enable duplication mode for ALL extruders... I am currently out of town and have no hardware to test the changes. Even if I had access to my printers, I only have an IDEX machine and [...]

Definately! thank you very much

SlickNickeL, 2019-03-09:
I have 20+ Cartesians that I can try it on. X-Y on top with Z bed that drops.
Easy to do, I just don’t code. I can edit and use and insert any and all G/M codes if I can understand their application.

Roxy-3D, 2019-03-09:
OK...   I'll start making the changes.   And we will see how fast we can make progress.

SlickNickeL, 2019-03-10:
> OK... I'll start making the changes. And we will see how fast we can make progress.

Let’s rethink this. Stacker S4 is using Repetier and they are doing a good job with it. I just did a peek into their software and it looks like mine.

My 3 head production run unit with Repetier is working well, and I see that STACKER has solved temp head issues. Don’t know about auto level, but I did go to the cantilevered auto level and it’s not bad at all. The axis is established centered between the fixed adjuster P2 and the Roll adjuster P3 with the P1 90 degrees off of this axis. MARLIN does not have this feature for a 3 point interlevel bed system. This may be the 800lb Gorrila in the room who has just decided to play nice!

Let the guys at Marlin work through this and get the LCD menus the way they should be and intigrated into the firmware. I say this because they may not even decide to make this work, and I think our time would best be served at tweaking something that is established and is showing great results. 

Repetier, is hands down better at this multhead duplication mode at the moment.  I think Marlin will be a problem in comparison. You will have to remove code for sanity check to work. This seems difficult for me. Repetier has documentation issues, but trial and error has solved two problems in just the last 24 hours for me. It’s on to temps and extrusion consistencies between filaments and flow vs size compensation through temps.

I will send you my complete code on Repetier, and some of the caveats that need to be addressed. It is a cleaner firmware. Compiles 10 times faster than Marlin. You will need to know exact specs on all of the physical motors and several other things,

Smart3DCNC, 2019-03-10:
This weekend I have designed a small machine to test the software, so I am still quite keen to test Marlin and Repetier to see how it work.
I am really keen on a very simple , mechanical stable machine, happy to design a machine per product I want to do. Dont need a universal adaptable design like the S4. Flexibility normally bring with it a lot of complexity.
My requirement is really as simple as running DITTO extruders, 1,2,3 max .
It would be a bonus if I could inhibit an extruder halfway if something goes wrong.
If the software to do that not available I will do it hardware intervention  just cut an extruder and stepper and simulate a HE temperature that the main control loop doesn't stop.

Im using our own electronics but its close to RAMPS will work with both platforms.

[image]

@Roxy-3D  I am not in a desperate rush will be unavailable for the next three weeks, but will give the testing proper attention when I return.

SlickNickeL, 2019-03-10:
> This weekend I have designed a small machine to test the software, so I am still quite keen to test Marlin and Repetier to see how it work.
> I am really keen on a very simple , mechanical stable [...]

You are doing the exact same thing that I have just finished. One machine for one single application. Mine is single part manufacturing.

If you use Repetier, and use the Zmax to zero function/autolevel you will be able to start and stop your print with “continue” at any time. This is one of the advantages, but I choose not to use this due to the length of time it takes for Z to go to max and then back to the bed. I’m not a big fan of that so, I use Z-min endstop with probe on Z-max. I can then fine tune with BED COATING-CUSTOM and adjust my 0 height here.
As far as a Ditto type selection, Repetier has this up to 4 heads that I have tried. Each head has its own configuration so feed can be adjusted per motor. Temp is set with M109 per head so viscosity can be temperature controlled per head and for any color. I am working on a specific set of test prints that will help determine temp and flow using just a little bit of filament, fast easy and fool proof so an employee can do the adjustments.
I’m at the point now, that my results are of the quality I need. I really don’t want to learn a new learning curve for a new firmware. It’s about time, efficiency and repeatability with the quality I need.

closed 2019-03-10 by Roxy-3D

Smart3DCNC, 2019-03-11:
@SlickNickeL  may I have a copy of a typical repetier firmware configuration?
my email is djcsys@iafrica.com

SlickNickeL, 2019-03-11:
Does this mean this Feature is going forward? If you are serious, I will dedicate a machine to the full testing of this function.

InsanityAutomation, 2019-03-11:
Yes, we will be working on it, and will tag this issue number when we have a PR ready to test. First step will be all extruders active, then gcode to set which ones are active, and finally menu items.

InsanityAutomation, 2019-03-11:
Theres something to look at and a decision to make. Had some time waiting on software to install and update today.