MarlinFirmware/Marlin #11703: Compile error on ANET board

sbaeder, 2018-09-01:
Arduino: 1.8.6 (Windows 10), Board: "Anet V1.0"
In file included from sketch\HAL.h:33:0,
                 from sketch\MarlinConfig.h:39,
                 from sketch\Marlin.h:35,
                 from sketch\temperature.cpp:27:
pins_ANET_10.h:134:33: error: pasting "/* comment*/" and "_DDR" does not give a valid preprocessing token

 #define HEATER_0_PIN       13   // comment
                                 ^
[...and so on...]

exit status 1
pasting "/* comment*/" and "_DDR" does not give a valid preprocessing token

Looks like there is a comment in the file (note: during debug, I replaced it just to be sure)....The two lines here  for the extruder and bed have comments.  Both cause the macro to fail.  SIMPLE fix...

Thanks for all the good work!!!

sbaeder, 2018-09-01:
Saw a later note about possible issues with 1.8.6 - and sure enough, with 1.8.5, the comment is parsed correctly.  Feel free to close and pass on to the upstream folks (or just remove the inline comments, since they aren't really necessary.

oakdesign, 2018-09-03:
it is a issue with the 1.8.6 gcc Compiler not an Marlin issue. 
https://github.com/arduino/Arduino/issues/7930#issuecomment-416079657

silycr, 2018-09-06:
Agreed with above. Compiles on 1.8.5, although I get an error from the name being too long. Instead of patching I thought I'd upgrade IDE. 
Same error "does not give a valid preprocessing token" in 1.8.6 & 1.9.0beta

thinkyhead, 2018-09-06:
Duplicate of #10820

closed 2018-09-06 by thinkyhead