MarlinFirmware/Marlin #11706: Delta Calibration Z Probe Endstop produces Wrong Z Probe Offset

davidmilunic, 2018-09-02:
**Description of the bug**

Set Z Probe Offset from Delta Calibration Menu gives wildly incorrect results.
I have an Anycubic Linear plus delta with removable probe.

### Steps to Reproduce
Under Control, Restore Failsafe and Store
Perform Delta AC
Store
Set Delta Height
Store
Remove z probe
Turn off Soft Endstop from Menu or with M211 S0 (tried both)
Probe Z Offset from Delta Calibration menu
Adjust until gap is correct from Bed 
Store

**Expected behavior:** 

Z Probe Offset will be adjusted by the amount determined under adjust Z Probe Offset]

**Actual behavior:**

Z Probe Offset is very large and wrong.
In my case Z Probe Offset was -15.9, adjust was -0.4 and final Z Probe Offset produced was -230.36

#### Additional Information
I have Anycubic Linear plus delta with removable probe.

I have tried  with Marlin 1.1.9, 1.1.x-bugfix or 2.0-bugfix and it happens with all three.

Turning off Soft Endstops from under the Prepare menu does not send anything to the log.
This log used M211 S0 instead.
Z Probe endstop Adjust from Delta Calibration issues a G33 P-1 in the Log, but returns wrong result with paper test.

Here is my Log;
08:55:33.358 : echo:Hardcoded Default Settings Loaded
08:55:35.480 : echo:Settings Stored (859 bytes; crc 3252)
08:56:05.852 : echo:enqueueing "G33"
08:56:05.872 : G33 Auto Calibrate
08:56:05.876 : Checking... AC
08:56:05.880 : .Height:303.53  Ex:-1.30  Ey:+0.00  Ez:-1.89  Radius:133.95
08:56:05.880 : .               Tx:+0.45  Ty:-0.67  Tz:+0.22
08:56:58.809 : Iteration : 01                                std dev:7.094
08:57:59.860 : Iteration : 02                                std dev:0.613
08:59:01.055 : Iteration : 03                                std dev:0.148
09:00:02.197 : Iteration : 04                                std dev:0.097
09:01:03.361 : Iteration : 05                                std dev:0.061
09:02:04.489 : Iteration : 06                                std dev:0.016
09:03:05.606 : Iteration : 07                                std dev:0.017
09:04:06.697 : Calibration OK                                rolling back.
09:04:06.702 : .Height:293.49  Ex:-1.24  Ey:+0.00  Ez:-11.92  Radius:133.84
09:04:06.704 : .               Tx:+0.43  Ty:-0.65  Tz:+0.22
09:04:06.704 : Save with M500 and/or copy to Configuration.h
09:08:17.807 : echo:Settings Stored (859 bytes; crc 17651)
09:08:36.303 : echo:Settings Stored (859 bytes; crc 17651)
09:08:52.028 : echo:enqueueing "G33 P1"
09:08:52.043 : G33 Auto Calibrate
09:08:52.043 : Checking... AC
09:08:52.045 : .Height:293.49                                Offset:-15.90
09:09:03.803 : Calibration OK                                std dev:0.000
09:09:03.804 : .Height:293.51                                Offset:-15.90
09:09:03.804 : Save with M500 and/or copy to Configuration.h
09:09:20.119 : echo:Settings Stored (859 bytes; crc 59619)
09:11:02.724 : echo:Soft endstops: Off  Min: X-110.00 Y-110.00 Z0.00  Max: X110.00 Y110.00 Z293.51
09:11:24.429 : echo:enqueueing "G33 P-1"
09:11:24.438 : G33 Auto Calibrate
09:11:24.438 : Checking... AC
09:11:24.441 : .Height:293.51                                Offset:-15.90
09:11:58.503 : Calibration OK                                std dev:0.000
09:11:58.505 : Height:293.51                                Offset:-230.36
09:11:58.505 : Save with M500 and/or copy to Configuration.h
09:12:09.429 : echo:endstops hit:  Z:0.00

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2343387/Configuration.zip)

brandstaetter, 2018-09-07:
As described in #11337, @grbd and myself also have issues with this feature - is it supposed to be run without the probe attached?

brandstaetter, 2018-09-07:
No, I don't think this will make any difference, the Pin reads the same regardless if the probe is connected or not.

I think this feature does not work for our probes.

As far as I understood it, it does the following:

 1) get the Z where the probe triggers
 2) does the half-manual paper test to get the Z=0 
 3) calculates the offset

It would work if there was a pause before the paper test that prompts us to remove the probe. I think.

davidmilunic, 2018-09-07:
Does the z-probe offset under Delta Calibration work for just Cartesian printers?
Do you get a really big offset when you run yours like I do?
Mine changes from -15.90 before probing to -230.36 after. (which is insane).
It's like it's grabbing the value from the z tower instead of the z height...
I really wish delta towers would be labeled "A", "B", "C" instead. It's too ambiguous.

I think the delta math is miffed.
From my configuration.h;
DELTA_DIAGONAL_ROD 271.50
DELTA_RADIUS 133.78
Using pythagorian theorem;
271.51^2 = 133.78^2 +z^2
z = 236.26

The effector drops down to 5 mm height when the z probe offset adjustment starts, so they probably subtracted the number from that to get -230.36 offset.

I have to find where they do that calculation in the source code...

davidmilunic, 2018-09-07:
Yup, I really think that this is what is happening...
They are setting the height from the z axis, like a Cartesian.

From marlin_main.cpp;

#elif ENABLED(DELTA)
    if (axis == Z_AXIS)
      current_position[axis] = delta_height;
    else
  #endif
  {
    current_position[axis] = base_home_pos(axis);
  }

  /**
   * Z Probe Z Homing? Account for the probe's Z offset.
   */
  #if HAS_BED_PROBE && Z_HOME_DIR < 0
    if (axis == Z_AXIS) {
      #if HOMING_Z_WITH_PROBE

        current_position[Z_AXIS] -= zprobe_zoffset;

davidmilunic, 2018-09-07:
I basically just avoid this feature under delta calibration.
It must be working for Cartesian, not delta and should not be there.

I'm just doing it the old fashioned way, set delta height, turn off soft endstop, move z axis and do paper test, note position, do math manually, update delta height, save to EEPROM...

I really wish they would fix it. It would be such a pleasure to do it through the LCD that way...
It can't be just Anycubic delta with this problem. It must be all deltas.

brandstaetter, 2018-09-07:
You won't have the problem if you have a BLTouch or other automatically retracting probe, I think.

davidmilunic, 2018-09-07:
Maybe they never really looked at "Fixed Mounted Probe" for delta and that is why there is a bug?
Maybe choose "Z_MIN_PROBE_ENDSTOP" instead of "FIX_MOUNTED_PROBE" fixes the problem.
I am in the middle of a print, maybe one more hour, then I can try.
Both options use the zmin endstop for the switch, so that should be the only change needed.

brandstaetter, 2018-09-07:
No, the  `Z_MIN_PROBE_ENDSTOP` just allows you to specify the pin explicitly instead of the default by Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

these are additional probe options, the available probe types are
PROBE_MANUALLY
FIX_MOUNTED_PROBE
Z_PROBE_SERVO_NR
BLTOUCH
SOLENOID_PROBE
Z_PROBE_SLED

where only FIX_MOUNTED_PROBE is applicable for us.

boelle, 2019-02-20:
@davidmilunic problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

davidmilunic, 2019-03-06:
I just don't use the feature any more.
It has been a while since I have updated Marlin, so maybe I will grab the
latest release and see what happens.

Thanks for following up!

 - David

boelle, 2019-03-06:
if you do then use bugfix 2.0

and a tool like winmerge comes in handy when porting over settings from old config file to new one, layout of those 2 have changed also :-D

boelle, 2019-03-13:
@davidmilunic had time to give latest bugfix 2.0 a go?

closed 2019-05-16 by thinkyhead