MarlinFirmware/Marlin #11720: [2.0] Lcd Fade height change + sd card printing = Print crash

studiodyne, 2018-09-04:
Sd printing , first layer 
Lcd change fade height from 10  to 1
Print dye

rmoravcik, 2018-09-06:
What electronics are you using? Did you tried to measure voltage of power supply? Is it stable? 

For example, I have Full Graphic Smart Controller and software cannot control it's contrast.

studiodyne, 2018-09-08:
It's a firmware bug , i used this feature before the new ' lcd menus configuration'

rmoravcik, 2018-09-08:
I'm sorry, I very quickly read your problem and misread it. So ignore my previous post...

I've just checked the code and menu "Fade Height" is still there under "Tune" menu that's visible during SD-print, but you have to enable it in Configuration.h

```
    //
    // Leveling Fade Height
    //
    #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT) && DISABLED(SLIM_LCD_MENUS)
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float3, MSG_Z_FADE_HEIGHT, &new_z_fade_height, 0, 100, _lcd_set_z_fade_height);
    #endif
```

studiodyne, 2018-09-09:
Hi thks to help me , but , because of bad translation , you don't understand my issue

When i begin a print ,  if i want to change  FADE HEIGHT with the lcd , the printing crash and , i have to restart a new print

Thks