MarlinFirmware/Marlin #11721: [2.0 Request] Mechanical/Motors menus missing on SDCARD printing

studiodyne, 2018-09-04:
Hi
The new configuration menus need more about mechanical and motors.
-Travel speed ' if too high , impossible to reduce it '
-Jerk ' printing is the only real moment when we adjust the jerk , but not possible now '

Please can you add , some essentials motors setting for sdprinting

Thks

marcio-ao, 2018-09-04:
See this comment:

https://github.com/MarlinFirmware/Marlin/pull/10360#issuecomment-417802100

hamster65, 2018-09-05:
Those values definitely were accessible during printing.
K-Factor for linear advance is missing as well.

studiodyne, 2018-09-08:
I've closed the old , because , i ask the community here , to use the sensor with some differences , no measurement , just used to detect a stopped filament , and it's perfect 

THks

closed 2018-09-08 by studiodyne