MarlinFirmware/Marlin #11722: [Feature Request] Retraction slack setting

ModMike, 2018-09-04:
### Description

Bowden retraction tuning retraction can be difficult because it is composed of 2 parts:

1) Filament slack in the tube
2) Absolute retraction limits of the heat break 

I have been struggling with tuning my E3DV6 because I have to compensate for slack in the Bowden tube, yet not exceed 2mm retraction in the E3DV6.

If we had a 2 parameter retraction setting we could set the tube retraction parameter and then the heartbreak retraction limit.

Calibration:

1) Calibrate slack by removing the tube at the extruder end
2) With the head cold, push the filament in as much as possible
3) Mark the filament at the tube
4) Release and measure how much it bounces back.
5) Repeat at all 4 corners and determine average
6) Set material (PLA, ABS, etc.): #DEFINE_PLA_SLACK
7) Set heartbreak maximum retraction: #DEFINE_MAXIMUM_HEATBREAK_RETRACTION

Actually sounds like more of a slicer thing but may be worth adding as sort of a sanity check for slicer.

thinkyhead, 2018-09-05:
It is more of a slicer thing, and slack should be accounted-for (if it matters) when setting the retraction length. And as long as the retract length is more than the slack length, everything should work fine.

ModMike, 2018-09-05:
After thinking about I think it really is an over complexification for Marlin.  If it has any value, it belongs in the slicer.

closed 2018-09-05 by ModMike