MarlinFirmware/Marlin #11727: [1.1.9] Still unable to compile with NO_LCD_MENUS and/or M73

Looxoor, 2018-09-04:
### Description
In response to previous issues #11600 and #11613 
There are still some kinks with NO_LCD MENUS (at least if you have an Anet).

Downloaded the latest fix, still throws the same errors.

First, with both NO_LCD_MENUS and LCD_SET_PROGRESS_MANUALLY enabled, it fails the sanity check.
I have an Anet with the ZONESTAR_LCD, and I can't see if ULTRA_LCD gets enabled anywhere for it if NO_LCD_MENUS is set (I think it gets defined if ULTIPANEL is defined), which apparently is a prerequisite for the progress bar.
SanityCheck.h:
```cpp
  #elif DISABLED(ULTRA_LCD)
    #error "LCD_PROGRESS_BAR requires a character LCD."
```
Next step, disable progress bar, it passes the sanity check and goes on to compiling files, works fine up until 
temperature.cpp:
```
sketch\temperature.cpp:2256:9: note: in expansion of macro 'HAL_START_ADC'

         HAL_START_ADC(ADC_KEYPAD_PIN);

         ^

exit status 1
'ADC_KEYPAD_PIN' was not declared in this scope
```
Which isn't so surprising seeing that PINS_ANET_10.h only defines the ADC_KEYPAD_PIN if ULTRA_LCD is enabled, which we've already concluded it isn't.

As a baseline, i changed to SLIM_LCD_MENUS and it works beautifully, even with progress bar.

Just for the kick of it, I added `#define ULTRA_LCD` to ZONESTAR_LCD, it threw ADC errors as well.

So, I don't know nearly enough to troubleshoot it further, thanks for all previous and continued work, I think it's awesome to be able to collaborate with the developers of such great software.

thinkyhead, 2018-09-05:
At this time, you can't use `NO_LCD_MENUS` in all situations. It is still a very crude hack. Check back with the `bugfix-x.x.x` branches periodically to see how it's progressing.

boelle, 2019-02-20:
@Looxoor problem solved?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead