MarlinFirmware/Marlin #11728: [ASK] Filament Zigzag and Extruder too close

nechveirst, 2018-09-04:
**Hi..** 

Before this, I have been try several 3D Printer, start from Cultivate, until now I am build m 3D Printer for myself, and I am doing some changing, the endstop Z I have been change using proximity metal sensor, LJ12A3-4-Z/BX, and it works well.

But I faced several problem that I have never been faced before.

1. The filament is filling out not linear, which mean the filament make a geometric like a zig-zag, (left and right. and I do not know how to solve this problem.

2. After several layers, the filament's geometric is nice, linear and order. But after 10th or 15th layer, the extruder is too close to the object, caused the object attracted from the top, and the object is moving. I have been read some article about increasing the Z-Stepper Feedrate and velocity, but it doesn't work. Because, after I changed the Z-Stepper until 1800, it still too close.

Could anyone help me to get out of these problems ?
The pictures:
https://intip.in/Problem1
https://intip.in/Problem21
https://intip.in/Problem22
https://intip.in/Problem23

Thank you.

thinkyhead, 2018-09-05:
Be sure to use `M502`, `M500` whenever you change the configuration and re-flash.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead