MarlinFirmware/Marlin #11729: [ASK][Marlin 1.1.9][Anet E12] EEPROM issues next to flash

vincentpfarr, 2018-09-05:
### Description
I just flashed the Anet E12 with the USBASP, and flashed to Marlin 1.1.9 with BlTouch. The printer is working well, Bltouch is working well, except that, using Pronterface, I can't use M500/M501 commands.

Printer is now online.
echo: External Reset
Marlin 1.1.9
echo: Last Updated: 2018-08-01 | Author: SPBB, Anet E12 BLTouch
echo:Compiled: Sep 4 2018
echo: Free Memory: 11556 PlannerBufferBytes: 1232
echo:EEPROM version mismatch (EEPROM=V47 Marlin=V55)
echo:Hardcoded Default Settings Loaded 

I tried to M502 --> M500 --> M501, no effects. When using the LCD to read/write config or reset to default, I can hear 2 bips. 

Settings have been well set in configuration.h. I have an Anet a8 and never had this issue. 

#define EEPROM_SETTINGS
#define DISABLE_M503   
#define EEPROM_CHITCHAT  

I tried reflash multiple time, change version, same issue. Config files attached.

Thanks very much. If you need more info, I'll do my best to help.

[E12 config.zip](https://github.com/MarlinFirmware/Marlin/files/2351648/E12.config.zip)

thinkyhead, 2018-09-06:
Un-comment the following line in `configuration_store.cpp`, re-flash, and see what the error output says on startup and with `M502`.
```cpp
// Check the integrity of data offsets.
// Can be disabled for production build.
//#define DEBUG_EEPROM_READWRITE
```

vincentpfarr, 2018-09-06:
Ok I'll do this this evening. Thanks a lot

vincentpfarr, 2018-09-06:
Hey,

Ok, with activating this option and enable the M503 command, it works... We were 2 to have this issue (Bastien ROSSI), and it seems that activating the M503 is resolving the issue.
ANyway thanks a lot

sebas76, 2018-11-30:
Hi,

I have the same issue with my Anet A6.
M500/501 not working when set DISABLE_M503.
I have modified configuration_store.cpp to check if it resolves the issue.

thinkyhead, 2018-11-30:
@sebas76 — Please switch to the `bugfix-2.0.x` branch and see if it works.

Cdrx3d, 2019-02-15:
Bonjour,
Ou je peux trouver marlin 1.19 pour l'anet E12 s'il vous plait ? Dans le fichier anet il y a les A6 et A8 mais pas la E12...

revilor, 2019-02-15:
@Cdrx3d there is an E12 config attached to the first post of this thread you could use as a starting point. Using the search engine of your choice you can also find https://www.thingiverse.com/thing:3039499

And a final tip: I still can read some French using what remained from school 20 years ago and with a little help from google translate. But in general you will get more responses here if you ask your question in English.

Cdrx3d, 2019-02-15:
@revilor 
Great, thank you
Indeed, the chrome translation option works well so I did not realize it. Thank you for the advice for next time I will write directly in English.

Thanks again for the info.

boelle, 2019-02-20:
@vincentpfarr problem solved?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead