MarlinFirmware/Marlin #11732: 24C512 EEprom addresses

boelle, 2018-09-05:
Hi

all arduino examples only show one address where all 3 address pins are tied to GND

i read the datasheet but got no wiser.

Are there a table somewhere that show all combinations and what address?

AnHardt, 2018-09-05:
In your datasheet - chapter "4.5 Device addressing"

http://ww1.microchip.com/downloads/en/DeviceDoc/doc1116.pdf
Figure 7 and 8-12.

boelle, 2018-09-05:
yep, i saw that and i did not figure it

all i know is that there are either 4 or 8 options

your datasheet links to another chip that has only 2 address select pins, mine has 3

boelle, 2018-09-05:
think i got it

0b1010000     0x50     E2=0 E1=0 E0=0
0b1010010     0x52     E2=0 E1=0 E0=1
0b1010100     0x54     E2=0 E1=1 E0=0
0b1011000     0x58     E2=1 E1=0 E0=0
0b1010110     0x56     E2=0 E1=1 E0=1
0b1011110     0x5E     E2=1 E1=1 E0=1
0b1011100     0x5C     E2=1 E1=1 E0=0
0b1011010     0x5A     E2=1 E1=0 E0=1